z88dk/z88dk #1479: (zsdcc) Anyone noticed increased compiler errors?

aralbrec, 2020-05-24:
I've been trying to trace some compile errors from a list of issues reported to me over the past six months and have noticed that the latest version of zsdcc has more errors than 1.99C zsdcc.  This includes the SO2 optimization level where it's only native sdcc rules that are applied and where 1.99C zsdcc doesn't have issues known to me at least.

Has anyone else noticed increased compile errors with the current zsdcc?  I may have chalked this up to new code generation patterns causing buggy SO3 rules to be invoked in new ways compared to 1.99C but the fact it's occuring even on SO2 levels gives pause for thought.

feilipu, 2020-05-24:
There is the thing that the zsdcc native (`-O2`) optimisation patches are not included in 2.0 zsdcc r11556 patch. These are still in the feature request to sdcc mainline, and are recorded [here](https://github.com/z88dk/z88dk/tree/master/src/zsdcc#patch-information), in the 10892 patch file.

The assumption was that only z88dk optimisation rules were being used. It sounds like that was a bad call.

I've not noticed (based on very limited work) any increased faults in zsdcc r11556, but I always use `-O3`. But that doesn't mean much.

aralbrec, 2020-05-24:
They are in fact essential for zsdcc :) Peephole optimization is done by sdcc using its peephole optimizer with levels selected by SO2 (sdcc rules only with some bugfixes and a few other things) and SO3

feilipu, 2020-05-24:
Yes typo. I meant -SO3.

aralbrec, 2020-05-25:
Mostly.  There are fixes in there for the code generation errors in mainline sdcc too.  I don't think all of those have been eliminated as I saw some fixes in some of the compiles I was trying.

But the misapplication of peephole rules is a major error in sdcc that hasn't shown up too much in mainline sdcc because the ruleset is small.  I think it might be even smaller now as I think they removed many of the rules in recent versions.

feilipu, 2020-05-25:
I had a quick look at a diff of 9958, and 10892, and 11556, and the differences are too great to follow. Particularly, as I don't understand SDCC.

Apologies that I broke this, and further that I can't really positively help to patch it up. 😞

Fabrizio-Caruso, 2020-06-10:
In #1503 I report a problem with SO1. So, I wonder whether this issue is already present with SO1.

Fabrizio-Caruso, 2020-06-11:
I am seeing a corrupted binary even with -SO0 in #1504

feilipu, 2020-06-17:
I've been through the sdcc source and all the patches and I am sure...
1. Every modification not in `peep.c` was carried over from r9958 to r11556.
2. Some of the patches in `peep.c` were carried into the mainline, but many were not.

So, I've been through the `peep.c` file and tried to make a sensible translation of 9958 into the 11556 code base.

And, this is the result: [sdcc-11556-peep-z88dk.patch](https://github.com/z88dk/z88dk/files/4792454/sdcc-11556-peep-z88dk.patch.txt)

@aralbrec @Fabrizio-Caruso offered here in the hope that you could test it to see if it helps.
I'll use it for a while to see what happens for me.

Fabrizio-Caruso, 2020-06-17:
@feilipu could you please provided a patched binary or give me the instructions on how to build one? I usually use the nightly builds.

feilipu, 2020-06-17:
@Fabrizio-Caruso there are standard instructions on the wiki. I wrote some too, [here](https://feilipu.me/2016/09/16/z80-c-code-development-with-eclipse-and-z88dk/) that you could try too.

feilipu, 2020-06-17:
PR in #1511, for review.

feilipu, 2020-06-18:
I’ve been trying to do test compiles using the internal sdcc peephole rules, but I haven’t found the correct command options to NOT automatically disable them and use the z88dk rules.

Does accessing the native peephole rules require NOT using `zcc` and its preprocessin, but rather accessing `zsdcc` directly?

suborb, 2020-07-06:
> Does accessing the native peephole rules require NOT using zcc and its preprocessing, but rather accessing zsdcc directly?

I think you have to invoke directly, it's these options:  `--no-peep --peep-file "/Users/dom/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/sdcc_peeph.2"` (etc) that switch in the z88dk rules.

So just do a `zcc ... -E` to do the preprocessing, rename the `.i` to a `.c` and then `zsdcc -S ...`

feilipu, 2020-07-06:
I copied an answer to my question into #1511 [here](https://github.com/z88dk/z88dk/pull/1511#issuecomment-646436693).

There's been quite a lot of independent work on fixing and improving the z80 peephole rules through to r11715, so the z88dk patch set doesn't work without improvement. And, although I've been through line by line to try to match the patch outcomes, I'm not certain of the effect the patch code implements in sdcc, so I don't know whether the outcome would be better than doing nothing?

The biggest question on my mind is whether there is still a need for the `argContPrec()` function? That is the core of the z88dk patch. If it is no longer needed because code elsewhere has improved, then z88dk patches are pretty minor in nature, and could be done line by line as patches to sdcc mainline.

aralbrec, 2020-07-24:
> The biggest question on my mind is whether there is still a need for the `argContPrec()` function? 

It depends on whether it's correctly fixed or not.  The old code did not properly identify registers affected in instructions and was the root of the misapplication of peephole rules in previous versions.

There are two large projects using zsdcc now and neither can get good builds out of the current zsdcc; they are instead using 1.99C.  So something somewhere is not right.  I do want to help out here so I will try to tie up other things in the next couple of weeks to have some time to come back to things.

One of the projects is the Magnetic Scrolls adventures for the zx next.  The problem file mentioned in connection to compile times https://github.com/z88dk/z88dk/issues/1474#issue-619069633 is in this project which is now open:  https://gitlab.com/strandgames/brahman/-/blob/master/zxnext_magnetic/src/emu.c .  The project will not produce a working binary under the current zsdcc (well last time it was checked some weeks ago).

feilipu, 2020-07-29:
Looking through the change log on SDCC there have been numerous and substantial changes to z80 by @basxto and @spth over the past weeks since r11722 (z88dk zsdcc). So now it no longer makes sense to us it as a base for comparison.

So suggest that we get a view as to a stable place to start, and then look at the improvements to the peepholes optimiser.

[r11762](https://sourceforge.net/p/sdcc/code/11762/) - bug fix
[r11761](https://sourceforge.net/p/sdcc/code/11761/) - z80n push nn fix
[r11755](https://sourceforge.net/p/sdcc/code/11755/) - read beyond buffer
[r11753](https://sourceforge.net/p/sdcc/code/11753/) - bug fix
[r11750](https://sourceforge.net/p/sdcc/code/11750/) - new z80 peephole rules
[r11747](https://sourceforge.net/p/sdcc/code/11747/) - improve use of `hl`
[r11739](https://sourceforge.net/p/sdcc/code/11739/) - bug fix ralloc2
[r11738](https://sourceforge.net/p/sdcc/code/11738/) - enhancement
[r11735](https://sourceforge.net/p/sdcc/code/11735/) - trampolines
[r11730](https://sourceforge.net/p/sdcc/code/11730/) - merge common z80 peephole rules
[r11728](https://sourceforge.net/p/sdcc/code/11728/) - banked functions
[r11726](https://sourceforge.net/p/sdcc/code/11726/) - bug fix

spth, 2020-07-29:
I would not recommend using current SDCC in z88dk.
For exactly the reason outlined above: There were lots of z80-affecting changes recently:

1) we might have introduced some regressions (one was found already: https://sourceforge.net/p/sdcc/bugs/3090/)
2) work will probably continue at similar intensity for a while

Additional testing of current SDCC is welcome. So is getting all non-z88dk-specific peephole optimizer improvements into SDCC.

But I wouldn't recommend using current SDCC as base for z88dk. Better wait until SDCC has stabilized a bit.

feilipu, 2020-07-29:
@spth given we're currently using r11722 as the ZSDCC base, but you've provided bug fixes in r11726, r11739, etc, would it be better for us just to cherry pick those and stay with r11722 plus patches?

There are some major projects (noted above) that would like to move forward on to a recent SDCC from r9958, but can't because it doesn't work for them. So it would be great to get your advice on where to "drop an anchor" , and try to resolve the reasons why the above projects don't work.

spth, 2020-07-29:
Some of the bug fixes might not be that relevant to z88dk: r11726 is about a bug regarding __itoa / _itoa, so mostly only relevant to users of that function.
And some of the later bug fixes are about bugs that essentially were only exposed by the improvements in register allocation  (mostly for gbz80).
Waiting for bug #3090 to be fixed (and any further regressions found until then) could be a reasonable choice.

The recent changes (https://sourceforge.net/p/sdcc/feature-requests/691/) in gbz80 register allocation have the potential to drastically improve the generated code (for some code I see huge improvements, for others, not much change). But that also means that quite different code is generated now for gbz80, so there might be new bugs lurking in some corner cases.

feilipu, 2020-07-29:
To play that back, there have been a lot of changes recently in SDCC for the better including code generation, so anchoring to an older iteration is not useful, particularly if there are other things repair. So, given there's going to be a few weeks of doing nothing anyway, it would make sense to wait until corner cases, including bug #3090, get fixed before finalising on a new ZSDCC version r number. Right?

spth, 2020-07-31:
SDCC[ bug #3090](https://sourceforge.net/p/sdcc/bugs/3090/) is fixed (turned out to be gbz80-specific). But it looks like there is one more regression: [Bug #3093](https://sourceforge.net/p/sdcc/bugs/3093/). I'll try to find time to look into and fix the latter this weekend.

It's not just me making changes now. There are 2 new SDCC developers that mostly work on z80-related ports.

basxto, 2020-07-31:
My latest commits for asgb and makebin, that shouldn’t affect z88dk.
My fixes for rgbds could’ve broken z80asm_mapping if there are no regression tests.

feilipu, 2020-08-05:
@basxto @spth the `peep.c` patches in question here are based around a `argContPrec()` function identifying exactly which registers are used. The function is [here](https://github.com/z88dk/z88dk/blob/master/src/zsdcc/sdcc-10892-z88dk-peep.patch#L82). Some of the outcomes were brought in previously, but I guess the real question is have the deficiencies `argContPrec()` was addressing been resolved?

If so, then the remaining [contents of the patch file](https://raw.githubusercontent.com/z88dk/z88dk/master/src/zsdcc/sdcc-10892-z88dk-peep.patch) become very selective, and in fact there may be other issues appearing rather than the missing patches.

spth, 2020-08-05:
The peep.c patch makes multiple changes:

1) argContPrec
2) Cosmetic changes we don't want (mostly "if (" vs. "if(" - see https://sourceforge.net/p/sdcc/wiki/SDCC%20coding%20style/
3) Some other cosmetic changes that don't really matter.

Clearly, 1) is what matters. Basically z88dk came up with argContPrec, when argCont wasn't good enough. But both are meant to do the same job. Since then, argCont has improved, but is still much simpler than argContPrec.
Today, I made a minor change meant to improve precision of argCont (https://sourceforge.net/p/sdcc/code/11794/), but it barely made a difference on the regression tests for z80, z80n (no difference for gbz80, didn't check further ports). In my experience, argCont is quite good these days. 

There area few remaining corner cases, where argCont could give false positives for hand-written asm, but usually the peephole optimizer is not used on hand-written asm.

If you have some sample C code, where argCont has insufficient precision, please file an SDCC feature request.

I think there is still a bit of lack of z80MightReadFlag() precision, but that is a different issue (and I'll look into it today).

basxto, 2020-08-05:
z80MightReadFlag() looses a lot if it punch because it treads `call` and `ret` always as read. And it’s only used on gbz80.

spth, 2020-08-05:
> z80MightReadFlag() looses a lot if it punch because it treads `call` and `ret` always as read. And it’s only used on gbz80.

It was only recently introduced to the z80-related ports, which explains that so far there are not many rules using it.

basxto, 2020-08-05:
I only enabled it for gbz80, so it should not be active on any other z80-related ports:
```
  // enable sp and flags only for GBZ80
  if(!isReg(what) && !isUReg(what) &&
     !(IS_GB && (!strcmp(what, "sp") || !strcmp(what+1, "f"))))
    return FALSE;
```
It’s not fit to handle z80, gbz80 and z80 were more incompatible than expected. I only partially prepared it for z80.

basxto, 2020-08-05:
> I think there is still a bit of lack of z80MightReadFlag() precision, but that is a different issue (and I'll look into it today).

That reminded me that I had some game breaking flag-related rules some time ago, which I disable and forgot.
It seems like I screwed up `jr`/`jp` heavily, I’m looking into this

spth, 2020-08-12:
Since z88dk uses --reserve-regs-iy a lot, you'll definitely want the fix for [Bug #3097](https://sourceforge.net/p/sdcc/bugs/3097/), that just went into upstream SDCC.

feilipu, 2020-09-26:
Added #1577 to test current.

feilipu, 2020-10-04:
#1504 result looks encouraging.

suborb, 2020-10-06:
Something is wrong  - md5 is now generated the incorrect result:

    zcc +test -compiler=sdcc md5sum.c

[md5sum.c.txt](https://github.com/z88dk/z88dk/files/5336265/md5sum.c.txt)

* 11891 - bad
* 11690 - bad
* 11311 - good

suborb, 2020-10-07:
I've managed to reduce this down to the following:

```
#include <stdio.h>
#include <stdint.h>

typedef uint32_t UINT4;

void func(UINT4 *buf, UINT4 *in)
{
  UINT4 a = buf[0], b = buf[1], c = buf[2], d = buf[3];

  printf("%lx %lx %lx %lx\n",a,b,c,d);

   {( a ) +=  ((( ( b ) ) & ( ( c ) )) | ((~ ( b ) ) & ( ( d ) )))  + ( in[ 0] ) + (UINT4)( 3614090360 ); ( a ) =  ((( ( a ) ) << ( ( 7 ) )) | (( ( a ) ) >> (32-( ( 7 ) )))) ; ( a ) += ( b ); } ;

  printf("%lx %lx %lx %lx\n",a,b,c,d);

   {( d ) +=  ((( ( a ) ) & ( ( b ) )) | ((~ ( a ) ) & ( ( c ) )))  + ( in[ 1] ) + (UINT4)( 3905402710 ); ( d ) =  ((( ( d ) ) << ( ( 12 ) )) | (( ( d ) ) >> (32-( ( 12 ) )))) ; ( d ) += ( a ); } ;
   printf("%lx %lx %lx %lx\n",a,b,c,d);
}

int main() {
   UINT4 buf[] = {0x67452301,0xefcdab89,0x98badcfe,0x10325476};
   UINT4 in[] = {0x3ec3,0x0,0x3dc3,0x0,0x3ac3};

   func(buf, in);
   return 0;
}
```

Expected output:

```
67452301 efcdab89 98badcfe 10325476
a53f48f4 efcdab89 98badcfe 10325476
a53f48f4 efcdab89 4ea7f82d 1fc4b45c
```

From failing sdcc versions:

```
67452301 efcdab89 98badcfe 10325476
a54ae174 efcdab89 98badcfe 10325476
a54ae174 efcdab89 37278631 d91ebce3
```

feilipu, 2020-10-12:
Noted that @spth has closed the report with a fix. I would like to generate a new PR, but AFK for a while.

feilipu, 2022-05-11:
Closed with #1969.

Will open a fresh issue if issues appear with v4.2.0.

closed 2022-05-11 by feilipu