z88dk/z88dk #1481: [info] Is parallel compilation for multiple targets possible?

Fabrizio-Caruso, 2020-05-30:
I compile my game for nearly all Z88DK 90 targets (and 60 non Z88DK targets)
In order to save hours, I have tried to use "make -j 8 ..." but it produces lots of errors that seem to be caused of the impossibility to remove zcc_opt.def.
Is there a solution to this problem that would allow parallel compilation? 
```
Could not open zcc_opt.def: File in use?
Cannot remove zcc_opt.def: File in use?
Cannot remove zcc_opt.def: File in use?
Cannot remove zcc_opt.def: File in use?
Cannot remove zcc_opt.def: File in use?
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:89: nascom_16k] Error 1
make: *** Waiting for unfinished jobs....
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:184: osca] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:99: nascom_32k] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:112: nc100] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:147: newbrain] Error 1
chase/end_screen.c:50:1: fatal error: Unexpected end of file
Compilation aborted
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:198: osborne1_chars] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:129: nc200] Error 1
```

suborb, 2020-05-30:
By a rather weird coincidence I wrote a wiki page about how to structure multi-target applications and created a makefile to make it all a bit easier: https://github.com/z88dk/z88dk/wiki/Compiling-Larger-Applications

However, it's not going to solve that nasty windows exclusive access problem you've got during the compilation stage. So I'm going to have to do a bit more work on that.

aralbrec, 2020-05-31:
I think if you compile from a different directory for each target then each compile will get its own zcc_opt.def to work with and its own destination directory for binary output files.

Fabrizio-Caruso, 2020-05-31:
@suborb in my cases I am not compiling every Z88DK with exactly the same files and options. I need a more general solution to just have "make -j 8" work on parallel compilation of even unrelated makefile targets.

Fabrizio-Caruso, 2020-05-31:
@aralbrec I have about 400 different targets in my Makefile(s). Creating 400 directories is not feasible.

suborb, 2020-05-31:
Yes. I realised that restriction when I tried to use it to compile the examples, so I made that possible, there’s an example here: https://github.com/z88dk/z88dk/blob/master/examples/console/Makefile Obviously it won’t fit in with how your project is structured. 

I think to achieve true parallelisation you might have to take some things from it - I think the output binary (before appmake) in your case is always the exact same filename so that will cause issues as well. Given that compilation speed is an issue I’d recommend moving over to incremental builds as well. 

I’m working on keeping zcc_opt.def in a temporary directory which should solve your original problem. As a result, the contents of it will be valid for one incantation of zcc. This will mean that single line compilations (like yours) will work and it will mandate use of  -pragma-include for multi-stage compilations.

Fabrizio-Caruso, 2020-05-31:
@suborb Thanks a lot!
What do I need to do to get some speed-up then?
I understand I need to avoid having the same output files. What do I need to do other than using the -o option?

How do I avoid the .o files to be overwritten from a parallel build?
Can I use your example to achieve that if if my targets do not necessarily use the same files and same options?

pauloscustodio, 2020-05-31:
z80asm has an option -Odir to put all the output files in the given directory, and you can pass it through zcc via -CaOdir. But I do not know if zcc can continue and call appmake from that directory.

suborb, 2020-05-31:
I want to test the above commit a bit more before merging.

---
In terms of speeding up the build, I think there's two parts to it:

Release builds:

The -j option which you're talking about. This will allow more builds to happen at the same time and take advantage of multi core machines.

Development cycle:

Thinking about an individual target: The way you've got things structured at the moment, the entire project is rebuilt when there's a small change to one of the .c files. It would be better to create .o files and then link them together. 

This can be taken further by tracking all the .h files included (we do this in the makefiles for the binaries), unfortunately at this point in time we haven't got support for an equivalent of the -MMD option

Fabrizio-Caruso, 2020-05-31:
@suborb, as usual you are right.
In my versy specific case (which is made of CrossLib, i.e., a HAL and a game code that uses it), I primarily need to understand how to speed the release builds.
As I develop mostly the abstraction later and much less the game(s) (which in principle does not change), I need to test multiple targets whenever I touch it.
So I need the release builds as abstraction later development cycle.

The development cycle is not a big deal for a second reason my game code can be reduced by some options that remove nearly all of the game code to produce a LIGHT or even TINY version of the game. This is enough to test that a new target works fine as I do not need to re-test the game code 200 times.

Fabrizio-Caruso, 2020-05-31:
@suborb, if you need small tests to test your new code to handle multiple zcc_opt, I have now split make huge Makefile in very small ones.
For example:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/tree/master/src/makefiles/targets/makefiles_z88dk

The smallest one is:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/master/src/makefiles/targets/makefiles_z88dk/Makefile_z88dk_w-z

which is built with: 
```make z88dk_targets_w-z```
and builds a dozen targets.

suborb, 2020-06-01:
I've merged it in, just have different output binary names is the easiest way to do it.

Fabrizio-Caruso, 2020-06-02:
@suborb I see no snaphot. Something went wrong.

Fabrizio-Caruso, 2020-06-02:
It does not seem to work... at least not under Cygwin
```
Unable to claim temporary filename C:\cygwin\tmp\zcc179B41
Unable to claim temporary filename C:\cygwin\tmp\zcc179B41
Unable to claim temporary filename C:\cygwin\tmp\zcc179B41
Unable to claim temporary filename C:\cygwin\tmp\zcc179B53
Unable to claim temporary filename C:\cygwin\tmp\zcc179B53
Unable to claim temporary filename C:\cygwin\tmp\zcc179B53
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:339: pencil2] Error 1
Unable to claim temporary filename C:\cygwin\tmp\zcc179B65
Unable to claim temporary filename C:\cygwin\tmp\zcc179B65
Unable to claim temporary filename C:\cygwin\tmp\zcc179B65
Unable to claim temporary filename C:\cygwin\tmp\zcc179B65
Unable to claim temporary filename C:\cygwin\tmp\zcc179B65
Unable to claim temporary filename C:\cygwin\tmp\zcc179B73
Unable to claim temporary filename C:\cygwin\tmp\zcc179B73
Unable to claim temporary filename C:\cygwin\tmp\zcc179B73
Unable to claim temporary filename C:\cygwin\tmp\zcc179B73
Unable to claim temporary filename C:\cygwin\tmp\zcc179B88
Unable to claim temporary filename C:\cygwin\tmp\zcc179B71
make: *** Waiting for unfinished jobs....e it is being used by another process.
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:90: nascom_16k] Error 1
rror 1lib/text/text_macros.c:10:20: fatal error: Unknown symbol: EXTRA_POINTS_IMAGE
431: pv2000] Error 1/targets/makefiles_z88dk/Makefile_z88dk_n-p:276: pasopia7] Error
make: ***
[makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:199: osborne1_chars] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:389: primo] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:408: pv1000] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:375: pps] Error 11
The process cannot access the file because it is being used by another process.
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:216: osborne1] Error 1
1ross_lib/text/text_macros.c:10:20: fatal error: Unknown symbol: EXTRA_POINTS_IMAGE
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:276: pasopia7] Error
 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:323: pc88] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:130: nc200] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:113: nc100] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:185: osca] Error 1
The process cannot access the file because it is being used by another process.
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:216: osborne1] Error 1
cross_lib/text/text_macros.c:10:20: fatal error: Unknown symbol: EXTRA_POINTS_IMAGE
Compilation aborted
Errors in source file cross_lib/display/display_macros.c:
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:454: px4] Error 1
chase/game_text.c:133:1: fatal error: Unexpected end of file
Compilation aborted
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:306: pc6001_32k_udg] Error 1
chase/main.c:620:1: fatal error: Unexpected end of file
Compilation aborted
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:474: px8] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:358: pmd85] Error 1
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_n-p:232: oz_tiny_buffered] Error 1
```

suborb, 2020-06-02:
It looks like there's some wonky code for handling tmp files on Windows that seeds the temporary name down to the nearest second. So that's the issue here - I'll sort it out this evening.

suborb, 2020-06-02:
I think it's now good - certainly a make -j4 on examples/console completes without errors under Cygwin and git-bash.

Let me know whether the build tonight works for you! @Fabrizio-Caruso

Fabrizio-Caruso, 2020-06-03:
@suborb It works wonderfully! I get about a  3 times speed-up with `-j 8` on a quad core with hyper-threading (10% less speed-up with `-j 4`).

closed 2020-06-03 by Fabrizio-Caruso

Fabrizio-Caruso, 2020-06-03:
@suborb THANKS a lot! This is making my tool-chain so much faster. I can now apply major changes to my abstraction layers and test it on more than 200 targets in reasonable times.