z88dk/z88dk #1482: [info] What causes Z88DK compilation times? Am I doing something wrong?

Fabrizio-Caruso, 2020-05-31:
First of all this I want to clarify that this is not a negative remark. I am amazed by Z88DK. I love it. It is way significantly better and bigger than any other dev-kit in terms of size, scope, support, dev effort and above all friendliness of the devs.
I am only trying to understand why it takes so much longer than similar compilers at least in my setup (which may be inefficient).
I do not expect anyone to make it faster. I am very happy with it as it is.
Is it slower than similar dev-kits because it is not single pass?
For example CC65 is single pass and it is extremely fast:  1.95 seconds vs 8:50 mins (Z88DK)
Maybe my Z88DK setup is wrong. I am using Windows/Cygwin for a dozen cross-compilers.

Some examples with Cross Chase:
```
- CC65 with MAX optimization -Or -Cl
time make c16_16k
real    0m1.950s
user    0m0.015s
sys     0m0.045s

- Z88DK (SCCZ80) with NO optimization -O0
time make spectrum_48k
real    8m50.479s
user    0m0.031s
sys     0m0.045s

- LCC1802 with max optimization:
time make full_comx
real    0m1.355s
user    0m0.135s
sys     0m0.318s

- ACK with max optimization:
make pc8086
real    0m5.362s
user    0m0.794s
sys     0m2.953s

- CMOC with max optimization:
make mo5
real    0m5.939s
user    0m2.378s
sys     0m2.267s

- native GCC without optimizations:
make gcc_buffered
real    0m4.759s
user    0m1.332s
sys     0m2.824s
```

pauloscustodio, 2020-05-31:
Have you tried with https://www.msys2.org/ ? This is what I normally use.

It is less user friendly than cygwin, you have to install packages with "packman -S" and search with "pacsearch".

Fabrizio-Caruso, 2020-05-31:
@pauloscustodio Thanks! I can give it a try. Do you think Cygwin could cause those very long compilation times? Have you tried to compile my game under your set-up?

pauloscustodio, 2020-05-31:
No, I did not. But I can, if you give me instructions.

suborb, 2020-05-31:
```
- Z88DK (SCCZ80) with NO optimization -O0
time make spectrum_48k
real    8m50.479s
user    0m0.031s
sys     0m0.045s
```

Are you sure this time is right? I'm seeing something like 10 seconds on my Windows VM vs 3ish on my Mac laptop.

Fabrizio-Caruso, 2020-05-31:
@pauloscustodio 
The repository is at: https://github.com/Fabrizio-Caruso/CROSS-CHASE
where you find some instructions but they may be incomplete.
- You just need to have the Makefile variable Z88DK_PATH point to the Z88DK binaries:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/master/src/makefiles/targets/Makefile.z88dk_targets#L5
- You can now use the `make` command to build any Z88DK target (for non-Z88DK targets you obviously need to install other dev-kits including a native GCC build)
Most of the targets are described in:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/tree/master/src/makefiles/targets/makefiles_z88dk
Examples:
`make spectrum_48k`
`make aquarius_exp_16k`
`make msx_32k_rom`
etc.

Fabrizio-Caruso, 2020-05-31:
@suborb, I am very sure about those times. It takes several minutes to build just one single Z88DK target. I do not have this problem with the other 12 dev-kits/compilers.

suborb, 2020-05-31:
To be honest, that's a ridiculous amount of time and I think we need to get to the bottom of it! I presume this is the same issue you reported a while back where at the start of April it was okay, and at the end it wasn't? 

Given it takes so long can you add a -v option and see where it seems to spend the most time? - A video of the terminal window would help as well because there may be more than one slow bit.

Fabrizio-Caruso, 2020-05-31:
@suborb THANKS a lot! Gonna do a video with OBS and post it here.

pauloscustodio, 2020-05-31:
@Fabrizio-Caruso , in which directory do I have to run "make spectrum_48k"?

Fabrizio-Caruso, 2020-05-31:
@pauloscustodio inside "src", which contains "Makefile":

Fabrizio-Caruso, 2020-05-31:
@suborb, I am recording the video: it is slow on every single file...

pauloscustodio, 2020-05-31:
Got it, thanks. Aproximately the same times:

real    0m8.057s
user    0m0.031s
sys     0m0.061s

Fabrizio-Caruso, 2020-05-31:
:-( So there is something very wrong in my Z88DK setup under Cygwin! :-(

suborb, 2020-05-31:
I've got comparable times using cygwin. So something is *very* unhappy on your machine. 

Whereabouts is z88dk in your PATH variable. If it's at the end, does it get any quicker if you put it at the start?

Fabrizio-Caruso, 2020-05-31:
@suborb, what do you mean at the start?
I define a variable at the top of the file:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/master/src/makefiles/targets/Makefile.z88dk_targets

In my case, I have a hierarchy of Makefile files:
`Makefile` is the root Makefile 
which includes
`Makefile.z88dk_targets` (where I define `Z88DK_PATH ?= `/cygdrive/c/z88dk/bin/`), which includes the files in the `makefiles_z88dk` directory:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/tree/master/src/makefiles/targets/makefiles_z88dk

Fabrizio-Caruso, 2020-05-31:
By the way I am NOT using a Cygwin Z88DK version. I am using the Windows Z88DK version. I have not built Z88DK from source under Cygwin. Are you guys doing the same as me?

pauloscustodio, 2020-05-31:
I call "./build.sh -k" at the top directory, and copy over the sdcc executables from the latest nightly build.
(-k because for some reason the tests fail on msys2, but not on Linux)

Fabrizio-Caruso, 2020-05-31:
@pauloscustodio so you cannot build all Z88DK binaries under Windows/Cygwin? 
Which sdcc are you referring to? From where do you take them?

@suborb, which Z88DK are you using under Cygwin? The Windows native build or a Cygwin build? 

What do I need to do to build under Cygwin?

pauloscustodio, 2020-05-31:
I cannot generate zsdcc.exe and zsdcpp.exe because I do not have subversion installed. These I copy from the latest nightly build. The others are generated by ./build.sh with msys2.

Fabrizio-Caruso, 2020-05-31:
@suborb here you are the video of the compilation:
https://youtu.be/YNjOHbnJlCY

pauloscustodio, 2020-05-31:
If you set the Z80ASM environment variable with -v, you can see the z80asm steps of the build:

    export Z80ASM=-v
    make

pauloscustodio, 2020-05-31:
This the total time it takes my machine to build z88dk (./build.sh -k) under msys2:

    real    85m56.224s
    user    4m33.255s
    sys     11m12.262s

@suborb, do you have very different figures?

Fabrizio-Caruso, 2020-05-31:
@pauloscustodio 
It looks Assembly steps are immediate while compilation is 50 times slower than in your case.
How do I recognize z80asm steps?
```
type "C:\cygwin\tmp\zcc314310.asm" >> "C:\cygwin\tmp\zcc314310.tmp"
type "C:\cygwin\tmp\zcc314310.tmp" >> "C:\cygwin\tmp\zcc314310.asm"
z80asm -I"c:/z88dk/\lib\target\zx\def" -D__SCCZ80  -s  -mz80 -I"c:/z88dk//lib" "C:\cygwin\tmp\zc
c314310.asm"
Predefined constant: __SCCZ80 = $0001
Library 'z80asm-z80-.lib' not found
Reading library 'c:/z88dk/lib/z80asm-z80-.lib'
Predefined constant: __CPU_Z80__ = $0001
Predefined constant: __CPU_ZILOG__ = $0001
Assembling 'C:/cygwin/tmp/zcc314310.asm' to 'C:/cygwin/tmp/zcc314310.o'
Reading 'C:/cygwin/tmp/zcc314310.asm' = 'C:/cygwin/tmp/zcc314310.asm'
Reading 'z80_crt0.hdr' = 'c:/z88dk/lib/z80_crt0.hdr'
Writing object file 'C:/cygwin/tmp/zcc314310.o'
Module 'chase_ghost_c' size: 260 bytes
```

pauloscustodio, 2020-05-31:
Just look for the phrases:
    Assembling .... (assembly)
    Reading library ... (readin z88dk libraries)
    Linking library module ... (linking the functions used by your code to the libraries)
    Code size ... (linking finished, start to build sections)
    Creating binary ... (building binary for each section)

Fabrizio-Caruso, 2020-05-31:
I do no see any problem in the assembling and linking phases which are immediate.
Only compilation is 50 times slower than it should

```
PROCESSING CRT
z80asm -I"c:/z88dk/\lib\target\zx\def" -D__SCCZ80  -s  -mz80 -I"c:/z88dk//lib" -I"c:/z88dk/\lib\
target\zx\classic"  "C:\cygwin\tmp\zcc314321.asm"
Predefined constant: __SCCZ80 = $0001
Library 'z80asm-z80-.lib' not found
Reading library 'c:/z88dk/lib/z80asm-z80-.lib'
Predefined constant: __CPU_Z80__ = $0001
Predefined constant: __CPU_ZILOG__ = $0001
Assembling 'C:/cygwin/tmp/zcc314321.asm' to 'C:/cygwin/tmp/zcc314321.o'
Reading 'C:/cygwin/tmp/zcc314321.asm' = 'C:/cygwin/tmp/zcc314321.asm'
Reading 'zcc_opt.def' = 'zcc_opt.def'
Reading 'crt/classic/crt_rules.inc' = 'c:/z88dk/lib/crt/classic/crt_rules.inc'
Reading 'crt/classic/crt_init_sp.asm' = 'c:/z88dk/lib/crt/classic/crt_init_sp.asm'
Reading 'crt/classic/crt_init_atexit.asm' = 'c:/z88dk/lib/crt/classic/crt_init_atexit.asm'
Reading 'crt/classic/crt_runtime_selection.asm' = 'c:/z88dk/lib/crt/classic/crt_runtime_selectio
n.asm'
Reading 'crt/classic/crt_section.asm' = 'c:/z88dk/lib/crt/classic/crt_section.asm'
Reading 'crt/classic/crt_section_standard.asm' = 'c:/z88dk/lib/crt/classic/crt_section_standard.
asm'
Writing object file 'C:/cygwin/tmp/zcc314321.o'
Module 'zx82_crt0' size: 203 bytes

z80asm -mz80 -b -d  -o"../build/FULL_spectrum_48k.prg" -m -s -L.   -D__SCCZ80 -L"c:/z88dk//lib/c
libs" -I"c:/z88dk//lib" -indos -izx_clib -i"z80_crt0"     "@C:\cygwin\tmp\zcc314322.lst"

Predefined constant: __SCCZ80 = $0001
Reading library 'c:/z88dk/lib/clibs/ndos.lib'
Reading library 'c:/z88dk/lib/clibs/zx_clib.lib'
Reading library 'c:/z88dk/lib/clibs/z80_crt0.lib'
Library 'z80asm-z80-.lib' not found
Reading library 'c:/z88dk/lib/z80asm-z80-.lib'
Predefined constant: __CPU_Z80__ = $0001
Predefined constant: __CPU_ZILOG__ = $0001
Linking library module 'fgetc_cons_inkey'
Linking library module 'in_GetKey_asm'
Linking library module 'in_Inkey'
Linking library module 'in_keytranstbl'
Linking library module 'in_GetKeyReset_asm'
Linking library module 'in_GetKeyVariables_asm'
Linking library module 'printf'
Linking library module 'fputc_callee_c'
Linking library module 'writebyte'
Linking library module 'fchkstd_c'
Linking library module '__printf_get_flags_impl'
Linking library module '__printf_handle_u'
Linking library module '__printf_number'
Linking library module '__printf_print_the_buffer'
Linking library module '__printf_get_buffer_address'
Linking library module '__printf_print_aligned'
Linking library module '__printf_doprint'
Linking library module 'asm_printf'
Linking library module 'get_16bit_ap_parameter'
Linking library module '__printf_print_to_buf'
Linking library module 'fputc_cons_generic_full'
Linking library module 'generic_console_attrs'
Linking library module 'generic_console_printc'
Linking library module 'generic_console_cls'
Linking library module 'console_vars'
Linking library module 'code_clib'
Linking library module 'console_vars'
Linking library module 'CRT_FONT'
Linking library module 'CRT_FONT_64'
Linking library module 'generic_console_scrollup'
Linking library module 'fputc_cons_generic_ioctl_c'
Linking library module 'generic_console_ioctl'
Linking library module 'console_coords'
[... 166 lines cut ...]
Section 'BANK_0' size: 0 bytes ($C000 to $BFFF)
Section 'BANK_1' size: 0 bytes ($C000 to $BFFF)
Section 'BANK_2' size: 0 bytes ($C000 to $BFFF)
Section 'BANK_3' size: 0 bytes ($C000 to $BFFF)
Section 'BANK_4' size: 0 bytes ($C000 to $BFFF)
Section 'BANK_5' size: 0 bytes ($C000 to $BFFF)
Section 'BANK_6' size: 0 bytes ($C000 to $BFFF)
Section 'BANK_7' size: 0 bytes ($C000 to $BFFF)
Creating binary '../build/FULL_spectrum_48k.prg'
Creating binary '../build/FULL_spectrum_48k_BANK_0.bin'
Creating binary '../build/FULL_spectrum_48k_BANK_1.bin'
Creating binary '../build/FULL_spectrum_48k_BANK_2.bin'
Creating binary '../build/FULL_spectrum_48k_BANK_3.bin'
Creating binary '../build/FULL_spectrum_48k_BANK_4.bin'
Creating binary '../build/FULL_spectrum_48k_BANK_5.bin'
Creating binary '../build/FULL_spectrum_48k_BANK_6.bin'
Creating binary '../build/FULL_spectrum_48k_BANK_7.bin'
appmake +zx  -b "../build/FULL_spectrum_48k.prg" -c "C:\cygwin\tmp\zcc314321"
rm ../build/FULL_spectrum_48k.prg
rm ../build/FULL_spectrum_48k_BANK_7.bin

real    10m38.224s
user    0m0.046s
sys     0m0.170s
```

pauloscustodio, 2020-05-31:
Can you add "-Ccv" to zcc, so that sccz80 gets -v and shows what it is doing?

Fabrizio-Caruso, 2020-05-31:
@pauloscustodio I am already ruuning zcc with -v as you can see in:
https://www.youtube.com/watch?v=O5vcTV0gxqw
Do I also need to add -Ccv?

pauloscustodio, 2020-05-31:
Yes, so that sccz80 gets also the -v option; I made a mistake, you should add -Cc-v:

     SCCZ80_OPTS ?= -O0 -v -Cc-v

Fabrizio-Caruso, 2020-05-31:
I do not see much more information with these options.

I have attached the output on stdout:
[slow_compile.txt](https://github.com/z88dk/z88dk/files/4708698/slow_compile.txt)

Fabrizio-Caruso, 2020-05-31:
This is a new video with the latest debug options:
https://youtu.be/7Ad4mU8fn-Y

pauloscustodio, 2020-06-01:
To understand where sccz80 is spending its time, you can:

1. Add profiling info to sccz80: edit src/sccz80/Makefile, and add "-pg" to both compile and link options:

    CFLAGS += -MMD -Wall -I../../ext/uthash/src/ -g -pedantic -std=gnu99 -pg
    $(CC) $(LDFLAGS) -o sccz80$(EXESUFFIX) $(OBJS) -lm -pg

2. rebuild it:

    make -C src/sccz80 clean
    ./build.sh

3. Make sure it has been updated in the directory where you run it from: in the CROSS-CHASE/src, do:

    which sccz80   --> in my case: /c/Users/paulo/git/z88dk/bin/sccz80
    ls -l /c/Users/paulo/git/z88dk/bin   -- change to your case, check file date

3. Now run your tool chain: "make spectrum_48k". There should now be a gmon.out file in your directory - this is created by the -pg option when sccz80 is run.

4. To analyze the data run:

    gprof /c/Users/paulo/git/z88dk/bin/sccz80.exe > gprof-report.txt   -- change paths to your case

pauloscustodio, 2020-06-01:
Here is my report: 
[gprof-report.txt](https://github.com/z88dk/z88dk/files/4708739/gprof-report.txt)

Fabrizio-Caruso, 2020-06-01:
The problem is SCCZ80 and not ZSDCC...
I have now run a simple test with the aquarius target:
ZSDCC:
`time make aquarius_exp_4k`
```
real    0m16.243s
user    0m0.076s
sys     0m0.182s
```

SCCZ80:
`time make aquarius_exp_4k` (without `-compiler=sdcc`)
```
real    2m36.489s
user    0m0.045s
sys     0m0.290s
```

suborb, 2020-06-01:
A couple of questions:

1. Can you time how long it takes to compile examples/console/world.c ? - i.e. is it a systemic problem or a complexity issue?
2. How many files to you have in your TEMP folder?

Fabrizio-Caruso, 2020-06-01:
Yes, I can test world.c. I suppose it is a systemic problem as I get it also on the tiny version of the game.
What is my TEMP folder? You mean my src directory and its sub-directories?

Fabrizio-Caruso, 2020-06-01:
Which options do I need to use to compile world.c?
$ time zcc +zx world.c
Error at file 'fputc_callee.c' line 139: symbol 'writebyte' not defined
Errors in source file c:/z88dk/\lib\target\zx\classic\spec_crt0.asm:
Error at file 'fputc_callee.c' line 139: symbol 'writebyte' not defined

real    0m2.804s
user    0m0.000s
sys     0m0.015s

-clib=ansi gives me the same problem

suborb, 2020-06-01:
The reason I'm asking is that sccz80 uses temporary files when calling functions. Your program has got quite a lot of function calls so will be using temporary files quite a lot. We know that windows file io isn't the fastest.

It looks like my TEMP folder is c:\Users\dom\AppData\Local\Temp\

Fabrizio-Caruso, 2020-06-01:
[image]

pauloscustodio, 2020-06-01:
Actually you should check the cygwin temp folder, i.e. the directory that /tmp points to.

Fabrizio-Caruso, 2020-06-01:
You may have found the issue... It is taking a lot of time to list the content....

Fabrizio-Caruso, 2020-06-01:
Guys! Problem SOLVED! Thanks A LOT! :-)
Now
SCCZ80:
```
time make aquarius_exp_4k (without -compiler=sdcc)
real    0m6.224s
user    0m0.105s
sys     0m0.258s
```
from 
```
real    2m36.489s
user    0m0.045s
sys     0m0.290s
```

The `/tmp` folder was causing the insane slow down.
The culprit is the LCC1802 cross compiler (for 1802-based CPUs) which pollutes the `/tmp` folder!

closed 2020-06-01 by Fabrizio-Caruso

suborb, 2020-06-01:
Does that mean you started to use lcc1802 at some point in April?

Fabrizio-Caruso, 2020-06-01:
Yes, I am working with Bill (LCC1802) and Marcel (Emma 02 dev) to make a real 1802-dev-kit out of the original bare-bone LCC1802 compiler. We are adding lib support for computers, consoles and arcade systems based on that architecture. I am mostly testing the libs and adding some headers files.
It can now be used like Z88DK, CMOC, CC65 to code for computers like COMX-35, Pecom 32/64, etc.

Fabrizio-Caruso, 2020-06-01:
@suborb I am not 100% LCC1802 is the only culprit. A modfified GCC for Amiga may have also abused the /tmp folder. Anyway the problem is the size of the /tmp directory.

While trying desperately to improve speed I tried to use ZSDCC to build some targets and I got lots of missing getk errrors. I will probably open a separate issue.