z88dk/z88dk #1485: Generating *.nex with Makefile problem

Dovgalyuk, 2020-06-01:
I'm using the makefile example from the following page (with zxn platform instead of zx):

https://github.com/z88dk/z88dk/wiki/Compiling-Larger-Applications

But when I add "-subtype=nex" option to program.nex compilation command line, it produces the 512-byte file (instead of 49k when I'm using the single compiler invocation command line).

suborb, 2020-06-01:
I can't reproduce that - I get a 16896 byte file when I add that option and compile world.c in examples/console. This is the same size/identical as a single compiler invocation command line.

```
make Q=
zcc +zxn     -clib=classic -subtype=nex -Izxn -o build/zxn/world.o -c world.c
zcc +zxn -m -o build/zxn/world.bin build/zxn/world.o      -clib=classic -subtype=nex  -lndos     -create-app
dom@heliotrope:~/z88dk/examples/console:982% ls -al build/zxn/
total 576
drwxr-xr-x    7 dom  staff     224  1 Jun 19:57 .
drwxr-xr-x  107 dom  staff    3424 31 May 18:33 ..
-rw-r--r--    1 dom  staff    4998  1 Jun 19:57 world.bin
-rw-r--r--    1 dom  staff  261031  1 Jun 19:57 world.map
-rw-r--r--    1 dom  staff   16896  1 Jun 19:57 world.nex
-rw-r--r--    1 dom  staff     363  1 Jun 19:57 world.o
-rw-r--r--    1 dom  staff       0  1 Jun 19:57 world_UNASSIGNED.bin
```

Dovgalyuk, 2020-06-03:
Here is my makefile for world.c

nex is empty, but map includes many names and addresses

```
CFILES = world.c
OFILES = $(CFILES:.c=.o)

PROGRAM = world.nex

all: $(PROGRAM)

world.nex: $(OFILES)
	zcc +zxn -m -o $@ $(OFILES) -clib=sdcc_iy -v -create-app -subtype=nex

%.o: %.c
	zcc +zxn -o $@ -c $^ -v -SO3 -clib=sdcc_iy --max-allocs-per-node200000 -subtype=nex -Izxn

clean:
	$(RM) *.o program.bin

```

suborb, 2020-06-03:
Ah, newlib that explains why I couldn't reproduce. For some reason with newlib, the -o option needs to be without the extension, so replace the appropriate line with this:

```
	zcc +zxn -m -o $(basename $@) $(OFILES) -clib=sdcc_iy -v -create-app -subtype=nex -Cl-v
```

Dovgalyuk, 2020-06-03:
This helps, thank you.
However, I think that wiki should be updated.

closed 2020-06-03 by Dovgalyuk