z88dk/z88dk #1487: [feature quest/info][appmake] Some output file names are capitalized. What is the rule?

Fabrizio-Caruso, 2020-06-02:
If I use appmake option "-create-app -o foo", I may get a small letter foo binary followed by capitalized FOO files (e.g., FOO.ROM).

Is there a general rule so that I can write scripts that clean-up and move the correct files?

zx70, 2020-06-02:
Fabrizio,  It is a deliberate choice. 
APPMAKE normally leaves the filename unchanged,  but for the extension. 
In some case,  in example for files to be copied into ancient operating systems,  the lowercase characters prevent the program to be found (by the original OS  or some useful tool or emulator).   Similarly in some case corrections can be applied for the name length,  the symbols,  etc.
Do you have a specific case/ conflict in mind?

Fabrizio-Caruso, 2020-06-02:
@z70, for instance the ones I have mentioned above (e.g., px4, px8, osca, etc.) but there are many more.
I am not at all sure this is an issue or a bad thing.
I am just wondering whether it was done on purpose on some targets to change the names to capital letters.

Fabrizio-Caruso, 2020-06-02:
I am closing this non-issue. Sorry for the inconvenience.

closed 2020-06-02 by Fabrizio-Caruso