z88dk/z88dk #1496: math32 - printing 4-byte floats with sccz80 is partially broken

feilipu, 2020-06-07:
Printing `math32` 4 byte floats is partially broken with sccz80.
Only the first variable is printed correctly. Something is not quite right.

suborb, 2020-06-07:
It’s the same code that is used for the other math libraries so I suspect there’s an issue in the inplementation. I had a kludge to make it work at one point in time - I think the sign was flipping at one point?

feilipu, 2020-06-07:
IIRC, the sign flipping was related to bad comparison code. Which is fixed.

What I see now is a 0 being printed for %f or some other incorrect small value for %f.

suborb, 2020-06-07:
It looks like this produces broken results for the `%e` part:

```
int main() {
        double e;

        for ( int i = 1; i < 200; i++ ) {
            e = i;
            e /= 100000000.;
            printf("%d %.18f %.10e\n",i,e,e);
        }
}
```

Using `%.9e` prints sensible numbers.

feilipu, 2020-06-08:
Yes. This looks like what I'm seeing, too.

If the print statement is broken into three statements, then it works.
```C
printf("%d ",i);
printf("%.18f ",e);
printf("%.10e\n",e);
```
The first floating point number is printed fine, but the second one is always broken.
The first `%.10e` works following a `%d` below, but the second `%.10e`  doesn't.
```C
printf("%d %.10e\n",i,e);
printf("%.18f %.10e\n",e,e);
```
I will have to look into what code the sccz80 `printf` statement calls that is implemented within the math library. I've forgotten.

feilipu, 2020-06-09:
So a little digging turns up the problem.
[This line](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdio/z80/asm_vfprintf_unlocked.asm#L475) causes the issue, by assuming 6 byte floats.

```asm
IF CLIB_OPT_PRINTF & $3fc00000

   ld hl,fcon_tbl_printf              ; float converters are independent of long spec
   call match_con_printf

   IF __SDCC | __SDCC_IX | __SDCC_IY
      jr c, printf_return_is_4
   ELSE
      jr c, printf_return_is_6
   ENDIF

ENDIF
```
Also present in the [_iy version](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdio/z80/asm_vfprintf_unlocked_iy.asm#L470) too.

Tested by changing `printf_return_is_6` to `printf_return_is_4`.
Problem located.

Not sure of the best long term way to solve it though...

suborb, 2020-06-09:
We were talking cross purposes. I was talking about classic not newlib. 

Classic detects the pragma at runtime:
https://github.com/z88dk/z88dk/blob/master/libsrc/stdio/__printf_handle_f.asm#L56

The alternative is to have a copy of the converter in each math library.

feilipu, 2020-06-09:
Oh... :facepalm: I didn't test with classic, so I didn't see the problem there too.
And there is the same (similar) problem too, I take it?

Well, anyway the problem with newlib is identified, but I'm not sure I want to tackle the issue of runtime float library identification. Above my paygrade, I'm afraid.

suborb, 2020-07-27:
This uses the same mechanism as classic now so is dynamically discovered at run time.

closed 2020-07-27 by suborb