z88dk/z88dk #1503: [zsdcc][bug] S01 option with ZSDCC breaks the binary

Fabrizio-Caruso, 2020-06-09:
Tested with the latest Z88DK snapshot (as of 09/06/2020).

If I compile:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/blob/master/src/makefiles/wip/Makefile.z88dk_wip_targets#L168
which uses -S01, then I get a corrupted game (it will just randomly show game over or crash or restart)

While it works fine with -S00 or by avoiding zsdcc with any optimization.

I have attached the -S0 and -S01 binaries.

[S00.zip](https://github.com/z88dk/z88dk/files/4754606/S00.zip)
[S01.zip](https://github.com/z88dk/z88dk/files/4754607/S01.zip)

feilipu, 2020-06-09:
Related to #1479, possibly.

A `--list -m --c-code-in-asm` zip file of each would be helpful.

Fabrizio-Caruso, 2020-06-10:
@feilipu Thanks for the suggestion.
I have created the list and map files.
[build_SO0.zip](https://github.com/z88dk/z88dk/files/4756914/build_SO0.zip)
[build_SO1.zip](https://github.com/z88dk/z88dk/files/4756915/build_SO1.zip)

feilipu, 2020-06-10:
Doing a diff of the two `.map` files shows that many labels are missing from the SO1 build.

[image]. See the labels present on the left, that aren't present on the right.

But, the `.lis` file set is not complete. Need to have the `.c.lis` files to see what happened to these labels, and to actually check what has been optimised.

The `.lis` files are identical, as they are generated by z88dk as the crt0.