z88dk/z88dk #1510: Compilation error at wrong line. How to debug?

Fabrizio-Caruso, 2020-06-15:
I am compiling with SCCZ80 -O0.
I am getting the error message below.
I am compiling with -v to see more details.
The error points at a totally unrelated line (that could even be commented).

```
PROCESSING chase/character.c
z88dk-ucpp -I"cross_lib" -I"cross_lib/sleep" -I"cross_lib/display" -I"cross_lib/display/graphics_data" -I"cross_lib/display/graphics_data/cc65" -I"cross_lib/display/graphic
s_data/z88dk" -I"cross_lib/display/graphics_mode" -I"cross_lib/display/alt_print" -I"cross_lib/display/image_settings" -I"cross_lib/display/redefine_characters" -I"cross_li
b/include" -I"cross_lib/sound" -I"cross_lib/sound/cc65/atmos" -I"cross_lib/sound/cc65/c264" -I"cross_lib/sound/cc65/pokey" -I"cross_lib/sound/cc65/sid" -I"cross_lib/sound/c
c65/vic20" -I"cross_lib/sound/z88dk/bit_bang" -I"cross_lib/sound/z88dk/psg" -I"cross_lib/sound/lcc1802/comx" -I"cross_lib/text" -I"cross_lib/input" -I"cross_lib/rand"  -iqu
:Error at file 'cross_lib/display/display_macros.c' line 43: symbol '_char_list' not defined
Error at file 'cross_lib/display/display_macros.c' line 108: symbol '_space_str' not defined
Errors in source file c:/z88dk/\lib\target\rex\classic\rex_crt0.asm:
Error at file 'cross_lib/display/display_macros.c' line 43: symbol '_char_list' not defined
                   ^ ----         #if !defined(__CIDELSA__)
Error at file 'cross_lib/display/display_macros.c' line 108: symbol '_space_str' not defined
                   ^ ----         PUTCH(0x41+xx);
```

suborb, 2020-06-15:
Have you got that in a branch somewhere? I can't reproduce with a `make rex`

I think things are getting confused because this is an error being reported by the assembler and it doesn't know which .c files have been included - I'm guessing there might be a missing `#include <>` of the file that defines the characters used on the Rex? 

 If you add `-cc` then that information will be passed across. I should look into adding the annotation for function definitions to give a little bit more of a chance of debugging this sort of thing.

suborb, 2020-06-28:
I've added some things to help so I'm going to close this one - reopen if it's still being problematic.

closed 2020-06-28 by suborb