z88dk/z88dk #1512: zsdcc — off-by-one miscompilation

imneme, 2020-06-20:
The following enclosed code, when run with `sccz80` produces the following (correct) output:
```
alloc_node(42,42) -> node 0
  a=0, nodes[a].lhs=42, nodes[a].rhs=42
  nodes=0x0FC5, nodes+a=0x0FC5, &.lhs=0x0FC5, &.rhs=0x0FC7
main <- node 0
  a=0, nodes[a].lhs=42, nodes[a].rhs=42
  nodes=0x0FC5, nodes+a=0x0FC5, &.lhs=0x0FC5, &.rhs=0x0FC7
```
(compiled with `zcc +cpm -clib=new -v bug.c -o bug  -create-app`)

However, if it's compiled with `zsdcc`, it gets miscompiled
```
alloc_node(42,42) -> node 0
  a=0, nodes[a].lhs=42, nodes[a].rhs=42
  nodes=0x0FFC, nodes+a=0x0FFD, &.lhs=0x0FFD, &.rhs=0x0FFF
main <- node 0
  a=0, nodes[a].lhs=10752, nodes[a].rhs=10752
  nodes=0x0FFC, nodes+a=0x0FFC, &.lhs=0x0FFC, &.rhs=0x0FFE
```
(compiled with `zcc +cpm -clib=sdcc_iy bug.c -o bug  -create-app`).

Notice that in `alloc_node`, the debugging output says `nodes=0x0FFC, nodes+a=0x0FFD`, but `a` is `0`, so this is **wrong**.

Commenting out the indicated line in the source causes the miscompilation not to be triggered.

---
```
// bug.c -- shows miscompilation by zsdcc

#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>
#include <assert.h>

typedef uint16_t atom;

struct node {
    atom lhs;
    atom rhs;
};

#define MAX_NODES 10

static struct node nodes[MAX_NODES];
uint16_t node_freelist = 0;

atom alloc_node(atom lhsval, atom rhsval)
{
    atom a = node_freelist;
    node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
    nodes[a].lhs = lhsval;
    nodes[a].rhs = rhsval;
    printf("alloc_node(%d,%d) -> node %d\n", lhsval, rhsval, a);
    printf("  a=%d, nodes[a].lhs=%d, nodes[a].rhs=%d\n", 
	   a, nodes[a].lhs, nodes[a].rhs);
    printf("  nodes=0x%p, nodes+a=0x%p, &.lhs=0x%p, &.rhs=0x%p\n",
	   nodes, nodes+a, &(nodes[a].lhs), &(nodes[a].rhs));
    return a;
}

int main()
{
    atom a = alloc_node(42,42);
    printf("main <- node %d\n", a);
    printf("  a=%d, nodes[a].lhs=%d, nodes[a].rhs=%d\n", 
	   a, nodes[a].lhs, nodes[a].rhs);
    printf("  nodes=0x%p, nodes+a=0x%p, &.lhs=0x%p, &.rhs=0x%p\n",
	   nodes, nodes+a, &(nodes[a].lhs), &(nodes[a].rhs));

    return 0;
}
```

imneme, 2020-06-21:
The issue does not exist in 1.99C, but does in 2.0 and nightlies.  (Thanks to Allen Albright for giving a plausible explanation and suggesting trying 1.99C.)

feilipu, 2020-06-21:
Interesting. I've played around with different optimisation on SDCC, and optimisation is not the issue #1147 

@spth From what I see, SDCC is forgetting that it has incremented into the second byte of `nodes[a].lhs`, when it tries to do the assignment. Optimisation only improves the incorrect code.

`zcc +cpm -clib=sdcc_iy -v -SO2 --list --c-code-in-asm offbyone.c -o bug -create-app`
SDCC Peephole Rules

```
555   0019              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
556   0019  E1          	pop	hl
557   001A  E5          	push	hl
558   001B  29          	add	hl, hl
559   001C  29          	add	hl, hl
560   001D              ;offbyone.c:31: nodes[a].lhs = lhsval;
561   001D  3E 00       	ld	a, +((_nodes) & 0xFF)
562   001F  85          	add	a, l
563   0020  5F          	ld	e, a
564   0021  3E 00       	ld	a, +((_nodes) / 256)
565   0023  8C          	adc	a, h
566   0024  57          	ld	d, a
567   0025              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
568   0025  1A          	ld	a,(de)
569   0026  21 00 00    	ld	hl,_node_freelist
570   0029  77          	ld	(hl), a
571   002A  13          	inc	de                          ; <- now pointing at second byte
572   002B  1A          	ld	a,(de)
573   002C  23          	inc	hl
574   002D  77          	ld	(hl), a
575   002E              ;offbyone.c:31: nodes[a].lhs = lhsval;
576   002E  6B          	ld	l, e                        ; <- now it should be pointing at the first byte, it is not
577   002F  62          	ld	h, d
578   0030  DD 7E 04    	ld	a,(ix+4)
579   0033  77          	ld	(hl), a
580   0034  23          	inc	hl
581   0035  DD 7E 05    	ld	a,(ix+5)
582   0038  77          	ld	(hl), a
```

`zcc +cpm -clib=sdcc_iy -v -SO3 --list --c-code-in-asm offbyone.c -o bug -create-app`
z88dk Peephole Rules

```
552   0016              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
553   0016  E1          	pop	hl
554   0017  E5          	push	hl
555   0018  29          	add	hl, hl
556   0019  29          	add	hl, hl
557   001A              ;offbyone.c:31: nodes[a].lhs = lhsval;
558   001A  11 00 00    	ld	de,_nodes
559   001D  19          	add	hl, de
560   001E  11 00 00    	ld	de,_node_freelist
561   0021  ED A0       	ldi                          ; <- now pointing at second byte
562   0023  7E          	ld	a,(hl)
563   0024  12          	ld	(de),a
564   0025              ;offbyone.c:31: nodes[a].lhs = lhsval;
565   0025  5D          	ld	e,l                  ; <- now it should be pointing at the first byte, it is not
566   0026  54          	ld	d,h
567   0027  DD 7E 04    	ld	a,(ix+4)
568   002A  77          	ld	(hl), a
569   002B  23          	inc	hl
570   002C  DD 7E 05    	ld	a,(ix+5)
571   002F  77          	ld	(hl), a
```

spth, 2020-06-21:
I haven't been able to reproduce the issue yet using sdcc 4.0.2 #11682, trying both -mz80 and -mgbz80.

feilipu, 2020-06-21:
It looks like peephole rule z88dk-247a is being fired for the `-SO3` case, even though `bc` was used. So there's a rule that has stepped in previously to change `bc` for `de`.

aralbrec, 2020-06-22:
@spth  It is a bug in sdcc as feilipu identified.  I don't have the current sdcc installed so I can't verify on sdcc itself tonight.

This is the z88dk version of sdcc 11556:

```
;bug.c:24: nodes[a].lhs = lhsval;
;	genPlus
	ld	a, +((_nodes) & 0xFF)
	add	a, l
	ld	e, a
	ld	a, +((_nodes) / 256)
	adc	a, h
	ld	d, a

;;;  ^^^ DE = & nodes[a].lhs

;bug.c:23: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
;	genPointerGet
;fetchPairLong
	ld	a,(de)
;fetchLitPair
	ld	hl,_node_freelist
	ld	(hl), a
	inc	de
	ld	a,(de)
;fetchLitPair
	inc	hl
	ld	(hl), a

;;; ^^^ node_freelist = nodes[a].lhs
;;;   DE = & nodes[a].lhs + 1 byte

;bug.c:24: nodes[a].lhs = lhsval;
;	genAssign (pointer)
;	AOP_STK for
;	isBitvar = 0
;fetchPairLong
	ld	l, e
	ld	h, d          ;;;;;    HL = & nodes[a].lhs + 1 byte
	ld	a,(ix+4)
	ld	(hl), a         ;;; stored into the wrong byte
	inc	hl
	ld	a,(ix+5)
	ld	(hl), a         ;;; stored into the wrong byte
```

Peephole rules are switched off.

From the z88dk version of sdcc 9958:

```
;bug.c:24: nodes[a].lhs = lhsval;
;	genPlus
	ld	a, +((_nodes) & 0xFF)
	add	a, l
	ld	c, a
	ld	a, +((_nodes) / 256)
	adc	a, h
	ld	b, a

;;; ^^^ BC = & nodes[a].lhs
;;; BC is now used to hold the address unchanged

;bug.c:23: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
;	genPointerGet
;fetchPairLong
	ld	e, c
	ld	d, b
	ld	a,(de)
;fetchLitPair
	ld	hl,_node_freelist
	ld	(hl), a
	inc	de
	ld	a,(de)
;fetchLitPair
	inc	hl
	ld	(hl), a

;;; ^^^ node_freelist = nodes[a].lhs
;;; DE = & nodes[a].lhs + 1 byte
;;; BC = & nodes[a].lhs

;bug.c:24: nodes[a].lhs = lhsval;
;	genAssign (pointer)
;	AOP_STK for
;	isBitvar = 0
;fetchPairLong
	ld	l, c
	ld	h, b                 ;;; here & nodes[a].lhs is retrieved from the correct value in BC
	ld	a,(ix+4)
	ld	(hl), a
	inc	hl
	ld	a,(ix+5)
	ld	(hl), a
```

Again peephole rules are switched off.

feilipu, 2020-06-22:
@aralbrec what I can't work out is why the zsdcc r11556 (current build) uses `bc` when called directly (correctly), but when called via zcc the same zsdcc uses `de` registers incorrectly.

I am guessing that there is a z88dk peephole rule that swapped register usage `de` for `bc` first, and then later on rule z88dk-247a did some further optimisation, and therefore there is a problem.

feilipu, 2020-06-22:
OK. (I think) I have a proof. The issue lies with the z88dk peephole rules.

Compiled with z88dk peephole rules disabled (I think), using this command line, there is no error.
`zcc +cpm -clib=sdcc_iy -v --list --c-code-in-asm --peep-file "" offbyone.c -o bug -create-app`
```
551   0011              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
552   0011  E1          	pop	hl
553   0012  E5          	push	hl
554   0013  29          	add	hl, hl
555   0014  29          	add	hl, hl
556   0015              ;offbyone.c:31: nodes[a].lhs = lhsval;
557   0015  3E 00       	ld	a, +((_nodes) & 0xFF)
558   0017  85          	add	a, l
559   0018  4F          	ld	c, a
560   0019  3E 00       	ld	a, +((_nodes) / 256)
561   001B  8C          	adc	a, h
562   001C  47          	ld	b, a
563   001D              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
564   001D  69          	ld	l, c
565   001E  60          	ld	h, b
566   001F  7E          	ld	a, (hl)
567   0020  32 00 00    	ld	(_node_freelist+0), a
568   0023  23          	inc	hl
569   0024  7E          	ld	a, (hl)
570   0025  32 01 00    	ld	(_node_freelist+1), a
571   0028              ;offbyone.c:31: nodes[a].lhs = lhsval;
572   0028  69          	ld	l, c
573   0029  60          	ld	h, b
574   002A  DD 7E 04    	ld	a,(ix+4)
575   002D  77          	ld	(hl), a
576   002E  23          	inc	hl
577   002F  DD 7E 05    	ld	a,(ix+5)
578   0032  77          	ld	(hl), a
```
But if the z88dk rules are NOT disabled, then with this command line
`zcc +cpm -clib=sdcc_iy -v --list --c-code-in-asm offbyone.c -o bug -create-app`
```555   0019              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
556   0019  E1          	pop	hl
557   001A  E5          	push	hl
558   001B  29          	add	hl, hl
559   001C  29          	add	hl, hl
560   001D              ;offbyone.c:31: nodes[a].lhs = lhsval;
561   001D  3E 00       	ld	a, +((_nodes) & 0xFF)
562   001F  85          	add	a, l
563   0020  5F          	ld	e, a
564   0021  3E 00       	ld	a, +((_nodes) / 256)
565   0023  8C          	adc	a, h
566   0024  57          	ld	d, a
567   0025              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
568   0025  1A          	ld	a,(de)
569   0026  21 00 00    	ld	hl,_node_freelist
570   0029  77          	ld	(hl), a
571   002A  13          	inc	de
572   002B  1A          	ld	a,(de)
573   002C  23          	inc	hl
574   002D  77          	ld	(hl), a
575   002E              ;offbyone.c:31: nodes[a].lhs = lhsval;
576   002E  6B          	ld	l, e
577   002F  62          	ld	h, d
578   0030  DD 7E 04    	ld	a,(ix+4)
579   0033  77          	ld	(hl), a
580   0034  23          	inc	hl
581   0035  DD 7E 05    	ld	a,(ix+5)
582   0038  77          	ld	(hl), a
```

aralbrec, 2020-06-22:
There were no peephole rules applied in the compiles above.  sdcc is a little bit indeterminate in the code it generates from compile to compile so you really have to look at the generated assembly with peephole applications in the listing.  You can do that with --fverbose-asm

```
zcc +cpm -clib=sdcc_iy -v -a --c-code-in-asm --fverbose-asm offbyone.c
```

By default zsdcc compiles use SO2 which are sdcc rules plus bugfixes and things like intrinsics added for z88dk.  SO3 is the aggressive peephole set.  If any rules are used they will be annotated in the generated asm file.

In general:

-SO0 = apply z88dk intrinsics only
-SO1 = apply z88dk intrinsics and sdcc peephole rules only
* -SO2 = apply z88dk intrinsics, sdcc peephole rules and bugfixes for sdcc's code generation errors
-SO3 = apply z88dk intrinsics, sdcc peephole rules, code generation bug fixes, z88dk aggressive optimization rules
-SO3 --opt-code-size = same as SO3 but with additional rules that try to fix up sdcc's handling of large integers (32-bit+)

* = default

The O levels control other processing.  O2 is the default:

-O0 = leave sdcc output in asz80 form, not usable in z88dk
-O1 = translate source and directives to z80asm form
* -O2 = O1 plus attach sdcc integer, floating point and built-ins to z88dk library code

* = default

You can be specific about the peephole processing:

```
zcc +cpm -v -a --fverbose-asm --c-code-in-asm -clib=sdcc_iy -SO2 offbyone.c

zcc +cpm -v -a --fverbose-asm --c-code-in-asm -clib=sdcc_iy -SO3 offbyone.c

zcc +cpm -v -a --fverbose-asm --c-code-in-asm -clib=sdcc_iy -SO3 --max-allocs-per-node200000 --opt-code-size offbyone.c
```

Anyway I saw no peephole rule substitution around the bad code.  You can maybe rerun with `--fverbose-asm` to double-check.

feilipu, 2020-06-22:
For contrast, zsdcc r9958 produces the similar code, but perhaps different enough that the z88dk peephole rules work differently.

Using __zsdcc r9958__.
`zcc +cpm -clib=sdcc_iy -v --list --c-code-in-asm --peep-file "" offbyone.c -o bug -create-app`
```551   0011              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
552   0011  E1          	pop	hl
553   0012  E5          	push	hl
554   0013  29          	add	hl, hl
555   0014  29          	add	hl, hl
556   0015              ;offbyone.c:31: nodes[a].lhs = lhsval;
557   0015  3E 00       	ld	a, +((_nodes) & 0xFF)
558   0017  85          	add	a, l
559   0018  4F          	ld	c, a
560   0019  3E 00       	ld	a, +((_nodes) / 256)
561   001B  8C          	adc	a, h
562   001C  47          	ld	b, a
563   001D              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
564   001D  69          	ld	l, c
565   001E  60          	ld	h, b
566   001F  7E          	ld	a, (hl)
567   0020  FD 21 00 00 	ld	iy,_node_freelist
568   0024  FD 77 00    	ld	(iy+0),a
569   0027  23          	inc	hl
570   0028  7E          	ld	a, (hl)
571   0029  FD 77 01    	ld	(iy+1),a
572   002C              ;offbyone.c:31: nodes[a].lhs = lhsval;
573   002C  69          	ld	l, c
574   002D  60          	ld	h, b
575   002E  DD 7E 04    	ld	a,(ix+4)
576   0031  77          	ld	(hl), a
577   0032  23          	inc	hl
578   0033  DD 7E 05    	ld	a,(ix+5)
579   0036  77          	ld	(hl), a
```

Using
`zcc +cpm -clib=sdcc_iy -v --list --c-code-in-asm --fverbose-asm offbyone.c -o bug -create-app`
```
579   0025              ;offbyone.c:30: node_freelist = nodes[a].lhs;  // <- comment out this line and all is good
580   0025              ;	genPointerGet
581   0025              ;fetchPairLong
582   0025  59          	ld	e, c
583   0026  50          	ld	d, b
584   0027  1A          	ld	a,(de)
585   0028              ;fetchLitPair
586   0028  21 00 00    	ld	hl,_node_freelist
587   002B  77          	ld	(hl), a
588   002C  13          	inc	de
589   002D  1A          	ld	a,(de)
590   002E              ;fetchLitPair
591   002E  23          	inc	hl
592   002F  77          	ld	(hl), a
593   0030              ;offbyone.c:31: nodes[a].lhs = lhsval;
594   0030              ;	genAssign (pointer)
595   0030              ;	AOP_STK for
596   0030              ;	isBitvar = 0
597   0030              ;fetchPairLong
598   0030  69          	ld	l, c
599   0031  60          	ld	h, b
600   0032  DD 7E 04    	ld	a,(ix+4)
601   0035  77          	ld	(hl), a
602   0036  23          	inc	hl
603   0037  DD 7E 05    	ld	a,(ix+5)
604   003A  77          	ld	(hl), a
```

aralbrec, 2020-06-22:
Both instances in 9958 are correct above but I think the difference is down to `--peep-file ""` not working properly.  The first compile is using the iy register but z88dk explicitly applies `--reserve-regs-iy` to instruct sdcc not to use iy.  So I think there might be some interference on the compile line caused by that peep-file option?

feilipu, 2020-06-22:
Yes. I'm not sure why there are no peephole rules applied in either of above, unless there was nothing to optimise?

Anyway, it seems that r11556 is picking the wrong registers to use. I'll see whether the current one has the same problem later today.

aralbrec, 2020-06-22:
They're both SO2 compiles which is just the sdcc rules.  There aren't many sdcc rules :)

r11556 may change how it's generating code (otherwise there is no improvement) so that's ok.  What is not ok is there seems to be an error in sdcc where it's not persisting the correct value of a pointer across statements like it is in 9958.

feilipu, 2020-06-22:
I'll fix the examples to use `-SO1` rather than the broken `--peep-file ""`.

And, try with the latest release later. If it is still there, then bug report.

feilipu, 2020-06-23:
Yes. The issue is still there with r11683.
Raised a bug report as [#3066](https://sourceforge.net/p/sdcc/bugs/3066/).

spth, 2020-06-24:
Fixed in sdcc now. The reason I couldn't reproduce it earlier was that I had forgotten z88dk uses --reserve-regs-iy.

feilipu, 2020-06-24:
Looks good.

Is post-decrementing the pointer robust? Would it still work if the `atom` were a `uint32_t`?
Answer to my own question... yes. it still works.

feilipu, 2020-06-24:
@imneme the fix has been merged. If you build zsdcc from source, please confirm. Otherwise please wait for the daily to confirm.

imneme, 2020-06-26:
@feilipu, @spth, here's what I see with the latest nightly (20200626):

```
unix% zcc +cpm -clib=sdcc_iy bug.c -o bug  -create-app
unix% mv BUG.COM bug.com
unix% tnylpo bug
alloc_node(42,42) -> node 0
  a=0, nodes[a].lhs=42, nodes[a].rhs=42
  nodes=0x0FFC, nodes+a=0x0FFD, &.lhs=0x0FFD, &.rhs=0x0FFF
main <- node 0
  a=0, nodes[a].lhs=10752, nodes[a].rhs=10752
  nodes=0x0FFC, nodes+a=0x0FFC, &.lhs=0x0FFC, &.rhs=0x0FFE
```

In other words, I still see the same bug.

feilipu, 2020-06-26:
@imneme  I’m reminded that Windows and Mac zsdcc binaries are done manually, and they haven’t been updated yet.

Check with zsdcc -v
It should show r11690 for the fixed version.

Sorry for missing info.

feilipu, 2020-06-29:
@imneme the binaries are updated, so you should be OK to test now.

imneme, 2020-06-29:
Yes, I can confirm that my app now builds on `zsdcc`, and has a ~45% reduction in execution time (i.e., almost twice as fast).

feilipu, 2020-06-29:
Thank you for reporting the issue, and for providing a very clear and easily reproducible test case.

closed 2020-06-29 by feilipu