z88dk/z88dk #1515: classic - configurable build for target libraries

feilipu, 2020-06-29:
I'm spending a long time (more than 20 minutes) watching over 100 target libraries compiling, each time I do a source rebuild.

Suddenly, it occurred that it would be nice to have a simple configuration file or tool somewhere to enable unnecessary targets to turned off. Primarily for classic, but also useful for newlib.

Does that exist today?

suborb, 2020-06-29:
For a specific target you can just do make cpm_clib.lib for example. 

For newlib I think it’s make TARGET=zx

feilipu, 2020-06-29:
For newlib it is `make job-target`. So for `math16` for example it is `make job-math16`.

But I was thinking of the inverse use case.
Is there a way to compile just the binaries, the standard libraries for classic, and just the machines I select, and not have every machine be built?

I mean, it is nice to have an environment supporting over 100 machines, but I would not to have to build for the 99 machines I don't have... for example.

suborb, 2020-06-29:
Not at the moment though, it's only a makefile so it's easy to knock into shape so I'll give it a go, make TARGETS="zx cpm" for example will build the crt0, math and the target specific libraries?

feilipu, 2020-06-29:
I don't know that my use case is everyone's use case, but as I'm mucking about in libraries and what-not rebuilding the core libraries, and then testing them on `+cpm` or `+zx` targets would be much faster if there weren't 98+ other targets that needed to be built each time too.

Perhaps (just thinking) the general makefile could read a target list in a file with (say) one target per line. If someone didn't want a specific target built they could comment out the unneeded target in their version of the file. For me, I would comment out 98 targets, for example. Someone else might want more or different targets, and that could be easy to adjust for their own preferences.

But, I would want the z88dk binaries, core (math, graphics, stdio, etc) libraries, built every time, because that's my interest (usually).

SDCC does it with a [command line option during configuration](https://github.com/z88dk/z88dk/blob/master/z88dk.Dockerfile#L31), but I think that would get annoying pretty quickly with over 100 targets to disqualify.

suborb, 2020-06-29:
End users aren't expected to build the libraries at all since all the configuration is at link time.

So, libsrc/Makefile can be made easier for us. My cycle is slightly different to yours but that's probably due to a couple of reasons: 1. If the binaries changes, then odds are that I made the change :) so I've already got the latest 2. I just build the library that I'm working on.  As a result, I very rarely use the top level makefile (which I guess you're using all the time?)

I think what we can do is to:

a) Support TARGETS=... from direct invocation of libsrc/Makefile
b) libsrc/Makefile includes `../Make.targets` (or similar) where you can specify which targets you want (by default it would be ?= all)

Which I think will work for you, the builds and for me?

feilipu, 2020-06-29:
> As a result, I very rarely use the top level makefile (which I guess you're using all the time?)

Yes. I'm using the `./buildsh` all the time, because I'm trying to balance what I'm doing across both classic and newlib. If I let one library get ahead of the other, then I tend to forget what I did.

Usually I just go out for a walk or coffee after hitting `./buildsh`, so it is not a big deal. But having a target + core option would be really helpful.

Working in newlib is easy though because of the ability to build just one target or library. So your proposal to replicate that process works well.

Thinking back, the only time it was a real PITA was when I was playing with the `fcntl` library, where I couldn't work out how to just get classic library cpm fcntl built by itself. And your proposal addresses that perfectly too.

suborb, 2020-07-03:
I've added a `-p [TARGET]`, for example `-p zx` option to build.sh to select the target(s) that you want to be built.

The same thing can be achieved by:

```
cd libsrc
export TARGETS=...
make
```

or

```
cd libsrc
make TARGETS=...
```

Which should be enough variants to keep everyone happy!

closed 2020-07-03 by suborb