z88dk/z88dk #1522: (zsdcc) Issue with __smallc again functions

suborb, 2020-07-04:
I've just had a chance to try this out and it segfaults whilst compiling hexdump.c from here ( http://specnext.com/forum/viewtopic.php?f=17&t=1667) in classic.

Compilation line:

zcc +zxn -clib=classic hexdump.c -o hexdump -subtype=dot -create-app -lesxdos -v -compiler=sdcc

It's fine for newlib, so it'll be a `__smallc` problem that's crept in again:

Stack trace:

```
Process 58188 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2)
    frame #0: 0x00000001000e2476 zsdcc`genZ80iCode [inlined] genIpush(ic=<unavailable>) at gen.c:4433:86
   4430
   4431	  if (size == 1 && smallc) /* The SmallC calling convention pushes 8-bit parameters as 16-bit values. */
   4432	    {
-> 4433	      if (AOP_TYPE (IC_LEFT (ic)) == AOP_REG && AOP (IC_LEFT (ic))->aopu.aop_reg[2]->rIdx == C_IDX)
   4434	        {
   4435	          emit2 ("push bc");
   4436	          regalloc_dry_run_cost++;
```

suborb, 2020-07-05:
Philipp has fixed this upstream now.

feilipu, 2020-07-05:
New zsdcc PR?

suborb, 2020-07-05:
Yes. We’ll need to head to the latest. I’m out for the day so I’ll do this this evening

suborb, 2020-07-05:
Patch applies cleanly so I've bumped the version in the makefile and rebuilt the binaries.

I'm bound to have forgotten something though...

closed 2020-07-05 by suborb