z88dk/z88dk #1527: Indirect function call -> spurious tab in assembly output -> confused z80instructionSize

imneme, 2020-07-08:
When compiled, the following code produces the warning:
```
buglet.c:14: info 218: z80instructionSize() failed to parse line node, assuming 999 bytes
'l_main_00117$:'
```

This appears to be because the generated code looks like this:
```asm
        ld      bc,l_main_00117
        push    bc
        ld      bc, (_ifunc)
        push    bc
        ret
        l_main_00117:
```
Notice that the `l_main_00117` is indented with a tab.  This issue only seems to occur if the indirect is to a `__z88dk_fastcall` function.

Code compiled with ` zcc +cpm -O3 -clib=sdcc_iy -S buglet.c` (sccz80 appears to handle this fine).

---
```c
short example(short x) __z88dk_fastcall
{
    return x+1;
}

short (*ifunc)(short arg) __z88dk_fastcall;

int main()
{
    int i, sum;
    ifunc = example;
    sum = 0;
    for (i = 0; i < 42; ++i)
	sum += (ifunc)(i);
    return sum;
}
```

feilipu, 2020-07-09:
Thanks. I've confirmed that this is an sdcc issue when used with `--reserve-regs-iy`.

Reported upstream as issue [#3081](https://sourceforge.net/p/sdcc/bugs/3081/).

feilipu, 2020-07-10:
Fixed in [sdcc r11722](https://sourceforge.net/p/sdcc/code/11722/).
Please confirm fix, and close if OK.

feilipu, 2020-07-17:
Presumed resolved. Please reopen if not.

closed 2020-07-17 by feilipu