z88dk/z88dk #1528: Compiling for 8080 - malloc() with -DAMALLOC not working

fcprindle, 2020-07-11:
I've been advised I could use z88dk to compile for an 8080 processor under CP/M using the following command line:
`zcc +cpm -clib=8080 -DAMALLOC pi.c -o pi -create-app`
and including "#include <malloc.h>" in the source code.
However, I tried this and I get the following errors:
```
Error at file 'C:/Users/Frank/Desktop/AltairDuino/z88dk/lib/../lib/crt/classic/crt_init_amalloc.asm' line 45: illegal identifier
Error at file 'C:/Users/Frank/Desktop/AltairDuino/z88dk/lib/../lib/crt/classic/crt_init_amalloc.asm' line 46: illegal identifier
Error at file 'C:/Users/Frank/Desktop/AltairDuino/z88dk/lib/../lib/crt/classic/crt_init_amalloc.asm' line 47: illegal identifier
Error at file 'C:/Users/Frank/Desktop/AltairDuino/z88dk/lib/../lib/crt/classic/crt_init_amalloc.asm' line 48: illegal identifier
Errors in source file C:\Users\Frank\Desktop\AltairDuino\z88dk\lib\config\..\..\\lib\target\cpm\classic\cpm_crt0.asm:
Error at file 'C:/Users/Frank/Desktop/AltairDuino/z88dk/lib/../lib/crt/classic/crt_init_amalloc.asm' line 45: illegal identifier
                   ^ ----       srl d
Error at file 'C:/Users/Frank/Desktop/AltairDuino/z88dk/lib/../lib/crt/classic/crt_init_amalloc.asm' line 46: illegal identifier
                   ^ ----       rr e
Error at file 'C:/Users/Frank/Desktop/AltairDuino/z88dk/lib/../lib/crt/classic/crt_init_amalloc.asm' line 47: illegal identifier
                   ^ ----       srl d
Error at file 'C:/Users/Frank/Desktop/AltairDuino/z88dk/lib/../lib/crt/classic/crt_init_amalloc.asm' line 48: illegal identifier
                   ^ ----       rr e
```
My naive interpretation of these errors is that the library that supports malloc() contains some z80 code. Is there anything that can be done about this?
Source code for pi.c is attached here: [PI.zip](https://github.com/z88dk/z88dk/files/4907024/PI.zip)

suborb, 2020-07-11:
Oh, that is silly of me - I made sure the library was 8080 compatible, but not the setup code.

I'll fix it this weekend and drop a note here when it's done.

fcprindle, 2020-07-12:
suborb, thanks for looking into this!

fcprindle, 2020-07-12:
Thanks suborb - seems to work now. Program now compiles without error and runs, but generates some minor garbage to stdout on exit that seems to leave CP/M console I/O slightly corrupted in some way, It does run significantly slower than when compiled for Z80 (about 1/3 the speed) but that was expected.

feilipu, 2020-07-12:
Thanks Frank. Drop a note to the Altairduino team, and let them know.

suborb, 2020-07-13:
I'll take a look at the performance issue - it looks like the CP/M dependencies are pretty small so I should be able to profile it in ticks.

Looking at the code, I suspect the bottle neck is the 32 bit division which probably isn't an optimal implementation.

feilipu, 2020-07-13:
Sorry, I'll leave this open then.

fcprindle, 2020-07-13:
Yeah, that division is literally at the crux of the algorithm (I have no understanding of the algorithm, but I do know that's where the magic takes place) and d cannot be 16 bits. I have another algorithm that also requires a long, but it may still compute PI (out to fewer correct digits) if it is changed to 16 bits - I'll look at that when I get a chance.

But isn't this "performance" issue off-topic for the original issue? I really didn't mean to open a can of worms, just wanted to get past the malloc() problem.

suborb, 2020-07-13:
It is a bit OT- so I've opened #1530 instead.

closed 2020-07-13 by suborb