z88dk/z88dk #1529: [bug] Recent snapshot breaks Cross Chase/Shoot compilation with a pointer addition error

Fabrizio-Caruso, 2020-07-11:
Only on very recent snapshots, I am now getting this error which I do not understand. Also I don't see where the "unsigned char" comes from.  Moreover "uint8_tspritesp-1]..." is displayed without a separating space in the error message.
It looks like a bug in the compiler.
If it is a bug in my code, I would like to have a more precise error message to help me figure it out.

```
cross_lib/display/display_macros.c:157:46: fatal error: Pointer addition: unsigned char uint8_tsprites[-1]+unsigned ch
ar uint8_t* is invalid
Compilation aborted
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_a-k:287: c128_z80_80col] Error 1
```
This happens on some code that computes a pointer to software sprite in an array that contains sprite data:
```
    void _draw_ch(uint8_t x, uint8_t y, uint8_t ch)
    {
        __DELETE(x,y);
        putsprite(spr_or,x*(SPRITE_X_STEP),y*(SPRITE_Y_STEP),sprites + (uint8_t *) ((ch-32)*(2+SPRITE_Y_SIZE))); \
    }
```
where
```
#define __DRAW(x,y,image) \
{ \
	putsprite(spr_or,x*(SPRITE_X_STEP),y*(SPRITE_Y_STEP),sprites + (uint8_t *) (image->_imageData*(2+SPRITE_Y_SIZE))); \
}

	
#define __DELETE(x,y) \
{ \
	putsprite(spr_and,x*(SPRITE_X_STEP),y*(SPRITE_Y_STEP),sprites + (uint8_t *) ((_SPACE)*(2+SPRITE_Y_SIZE)) ); \
}
```
and where sprites is provided in a .h file 

```
__asm
	._sprites

	defb    8,8
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"
	defb %"--------"
	
	defb    8,8
	defb    @00011000
	defb    @00100100
	defb    @00000000
	defb    @01100110
	defb    @10011001
	defb    @00000000
	defb    @00100100
	defb    @01100110

	........................................

__endasm;

suborb, 2020-07-12:
Boiling this down we have this code:

```
typedef unsigned char uint8_t;

extern uint8_t arr[];

uint8_t *func(int offset) {
        return arr + (uint8_t *)offset;
}
```

Which yields:

```
Issue_1529_array_addition.c:7:32: fatal error: Pointer addition: unsigned char uint8_tarr[-1]+unsigned char uint8_t* is invalid
```

So we can see you're effectively adding two pointers. A relevant bit of K&R is:

> The valid pointer operations are assignment of pointers of the same type, adding and subtracting a pointer and an integer, subtracting or comparing two pointers to the members of the same array, and assigning or comparing to zero. All other pointer arithmetic is illegal. It is not legal to add two pointers

For reference SDCC's error is:

```
 Issue_1529_array_addition.c:7: error 42: pointer + pointer invalid
```

And clang's is:

```
Issue_1529_array_addition.c:7:13: error: invalid operands to binary expression
      ('uint8_t []' and 'uint8_t *' (aka 'unsigned char *'))
        return arr + (uint8_t *)offset;
```

I'll see what I can do about the printing - it looks like it's dumping out typedefs, hence the `unsigned char uint8_t`. The [-1] indicates an unsized array (I should be able to drop the -1), and the missing space is a just unfortunate.

This diagnostic went in on 2020-06-10.

As to a fix, just drop the `(uint8_t *)` cast you've added.

Fabrizio-Caruso, 2020-07-12:
@suborb sorry for reporting an improvement as a bug...

closed 2020-07-12 by Fabrizio-Caruso

suborb, 2020-07-12:
No worries, sccz80 used to (and probably still does) accept a fair few invalid expressions, I'm just fixing up diagnostics as I find the cases.