z88dk/z88dk #1530: (8080) Performance of long division is (probably) poor

suborb, 2020-07-13:
From #1528 - pi on the 8080 performs pretty badly.

The suspicion is that it's the 32 bit division code.

fcprindle, 2020-07-13:
I have another PI program that also has one 32-bit **long int**, and it is also involved in a division as well as other arithmetic. But unlike the original PI program, I can change the **long int** to an **int** and the program will still work, but instead of being able to compute PI accurately to 22904 decimal places, changing **long** to **int** reduces the number of correct decimal places to just 249. But that's adequate to benchmark the relative timing...

With the **long int** in there, it runs about 2.5 times slower than the program when compiled for Z80; with the **long int** replaced by **int**, it runs nearly as fast as the Z80 code (only about 14% slower)! So the **long** arithmetic is the culprit, the division being the most likely operation to be slowing it down. That is not to say that I think division of **long**s should be as efficient in 8080 code as in z80 code - it may just not be possible, but I'll leave that to the experts.

suborb, 2020-07-13:
Yes, I ran a hotspot test earlier and the clear winner in the original pi.c is the shift loop in the division algorithm.

Unrolling the inner loops will help a little, but I think the main gain will be from using leading 0 detection (as `-lfastmath` does) which will of course take up more memory which I'm loathed to do by default.

feilipu, 2020-07-13:
IIRC, the Microsoft FP mantissa multiply calculation did [leading 0 Byte detection](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/rc2014_NascomBasic32k_HexLoadr/bas32k.asm#L3254). Perhaps it is enough to test for 0 Byte and jump to a lesser iterator?

suborb, 2020-07-14:
Some baseline numbers for pi:

* +test (z80) -> 1_754_696_046 ticks
* +test -clib=8080 -> 5_613_993_242 ticks

Unrolling the shift loop at the cost of 12 bytes, 8080 decreases to: 5_010_478_106.

Yes, that's my next plan to cut down the number of iterations which should make a massive difference even if the cost of each iteration remains the same.

suborb, 2020-07-16:
This is now down to ~4_581_000_000

* I tried calling the 16x16 multiply when appropriate but it didn't really save much in this test case. 
* Changing the compare/sub to sized based on the divisor slowed things down (to 4_803...)

So, without a fundamental algorithm change this isn't going to get any faster so I'm going to close this issue.

closed 2020-07-16 by suborb

fcprindle, 2020-07-17:
Thanks suborb! Can't reasonably eliminate the long div from either program, but this optimization should make them significantly faster than Aztec C on the 8080. And of course it's already outstanding on the Z80. I'll download the Z88DK build tomorrow, recompile my 2(PIs)x2(CPUs) test cases, and pass them along to the the retro-comp and Altair-Duino forums when I get a chance.

suborb, 2020-07-17:
If modifying pi.c is an option then using ldiv() would be the obvious optimisation.  It may not exist for the 8080 but should be easy to add.