z88dk/z88dk #1531: Bug in 8080 compilation for CP/M - extern variable expression evaluation fails

fcprindle, 2020-07-13:
When I compile the attached source program **bug.c** using the zcc command line:
_zcc +cpm -clib=8080 -DAMALLOC bug.c -o bug -create-app_
I get a strange warning that references a line number that doesn't exist:
_Warning at file 'bug.c' line 34: interpreting indirect value as immediate_

When I move the bug.com file to an 8080-based CP/M system and run it, it prints:
_It is not zero_
But the expression evaluates to zero, so it should print:
_It is zero_

I noticed that if the 3 variables are local to main() and not global (extern), no warning occurs and the expression is evaluated correctly.

Here is the source program: [bug.zip](https://github.com/z88dk/z88dk/files/4914303/bug.zip)

suborb, 2020-07-13:
Thank you, there's an incorrectly qualified rule that was inserting ld de,(xxx). z80asm then catches this and prints up the warning you see. 

The warning should have been at the right C line, but that's a fairly recent sccz80 change so you might not have it.

closed 2020-07-13 by suborb

fcprindle, 2020-07-13:
Ok, just to make sure I get the latest and greatest for Windows, I figured I'd wait until tomorrow and then pull the latest nightly build for Windows. Buf for some strange reason, the nightly builds only go up to 7/7/2020. Why is that?

suborb, 2020-07-13:
Let's just say that I commented out copying of the kits when I was switching the windows build to 64 bit.

I've copied the build from last night into place now.

fcprindle, 2020-07-13:
I guess it'll be tonight's build I'll be wanting - this bug is still there in last night's.
But the line number is better - it's only off by 1 line now... close enough for government work.

suborb, 2020-07-13:
Yes it'll be tonight's - I didn't trigger another one - the change is pretty small so just grab [z80rules.1](https://github.com/z88dk/z88dk/blob/master/lib/z80rules.1) and drop it in.

fcprindle, 2020-07-13:
Yep, that did it. Thanks!

fcprindle, 2020-07-13:
Well, suborb, I spoke too soon. I got your latest nightly build and put in the new z80rules.1. It then compiled for 8080 fine, but when I tried to compile the PI programs for Z80, it gives the following error:

**m4.exe - System Error
The code execution cannot proceed because msys-2.0.dll was not found. Reinstalling the program may fix this problem.**

Any clues what's causing this?

suborb, 2020-07-13:
I upgraded m4 to 64 bit at the same time and it looks like it has a dependency on that dll.

I've added it into kit for tomorrow/later this morning so lets see how it goes. If you're stuck, just switch out the m4.exe from the version you were previously using.

fcprindle, 2020-07-14:
Well, I better wait for tomorrow's build because swapping in the older m4.exe (from the 6/14/2020 build) gets a similar error but now the missing dll is **regex2.dll**.

suborb, 2020-07-14:
It's like a whack-a-dependency!

I've now rebuilt m4.exe myself so it's dependency-less which should make life a bit easier going forwards.

commit 827505da (2020-07-13): Fix #1531: Kill opt rule kicking in for 8080 when it shouldn't