z88dk/z88dk #1532: Branches - Cleanup

PaintedBlck, 2020-07-19:
Dear Development team,

when looking at the branches and the dates of last commits, I see the list below. Would it be feasible to delete for instance all branches with a last commit of 18 months or more ago (except the release branches of course), since they are probably out-of-date anyway? Just asking...

Kind regards,
PaintedBlck

```
2020-07-19 20:37:01 +1000 3 hours ago 	origin/master
2020-07-19 01:20:08 +0100 13 hours ago 	origin/z80asm2
2020-07-03 22:19:22 +0100 2 weeks ago 	origin/feature/libsrc_makefile_v2
2020-06-18 08:44:03 +1000 5 weeks ago 	origin/zsdcc-11556-peep-patch
2020-06-04 20:33:57 +0100 6 weeks ago 	origin/sccz80/start_rename
2020-05-06 14:53:31 +0100 2 months ago 	origin/z80asm_asmpp
2020-04-15 22:44:42 +0100 3 months ago 	origin/feature/zx_colour_gfx
2020-03-31 17:58:05 +1100 4 months ago 	origin/release200
2020-02-18 11:43:05 +0100 5 months ago 	origin/feature/m100
2020-01-02 14:55:34 +0000 7 months ago 	origin/fix/ticks
2020-01-02 14:54:46 +0000 7 months ago 	origin/feature/z80asm_pseudo_opcodes
2019-10-29 02:59:43 +0000 9 months ago 	origin/z80asm_pseudo_opcodes
2019-08-15 21:44:13 +0100 11 months ago 	origin/z80asm_macros
2019-08-01 22:33:51 +0100 12 months ago 	origin/feature/gencon_cursor
2019-01-27 01:05:38 +0000 1 year, 6 months ago 	origin/issue_1077
2019-01-15 21:46:41 +0000 1 year, 6 months ago 	origin/framepointer_fixups
2018-12-02 18:20:57 +0000 1 year, 8 months ago 	origin/z80asm_cxx_port
2018-09-29 16:48:12 +0100 1 year, 10 months ago 	origin/z80asm_z280
2018-08-15 15:32:10 +0100 1 year, 11 months ago 	origin/z80asm_macros3
2017-05-05 20:38:09 +0100 3 years, 2 months ago 	origin/feature/debuild
2017-05-03 08:46:04 -0600 3 years, 3 months ago 	origin/dev/aralbrec
2017-03-28 01:13:09 -0600 3 years, 4 months ago 	origin/Benchmark
2017-03-02 14:17:02 -0700 3 years, 5 months ago 	origin/newclib/IncorporateFASE
2017-01-12 00:36:12 +0000 3 years, 6 months ago 	origin/release/1.99B
2017-01-12 00:36:12 +0000 3 years, 6 months ago 	origin/origin
2002-04-07 12:31:55 +0000 18 years ago 	origin/rc-v1_4
2000-07-04 15:33:25 +0000 20 years ago 	origin/z88dk
```

pauloscustodio, 2020-07-19:
Good point. 0

pauloscustodio, 2020-07-19:
Good point. I will have a look at the z80asm ones, save what I need and delete the rest. Most are probably changes that were started and later abandoned.

pauloscustodio, 2020-07-19:
z80asm's old branches removed.

suborb, 2020-07-20:
Thanks @pauloscustodio - I've cleaned a few as well so I think we're down to a sensible number now.

closed 2020-07-20 by suborb