z88dk/z88dk #1534: (Classic) ZX +3 fcntl not working

suborb, 2020-07-20:
Investigation to follow

DarwinNE, 2020-07-20:
Hello,
   if it can be useful for you, here is a possible test program (called `plus3file.c`) I obtained from some existing code of mine:

~~~~
#include <stdio.h>
#include <string.h>

#define BUFFERSIZE 128

void wtr(const char *s)
{
    for(; *s!='\0';++s)
        fputc(*s,stdout);
}

int main(void)
{
    FILE *f;
    char buffer[BUFFERSIZE];

    f=fopen("test.dat","w");

    if(f==NULL) {
        wtr("Can not open file ");
        return 1;
    }
    fputs("SAVEDAWS2.1\n",f);

    fputs("this is a test\n",f);
    fclose(f);
    f=fopen("test.dat","r");
    fgets(buffer, BUFFERSIZE, f);
    if(f==NULL) {
        wtr("Can not open file ");
        return 1;
    }
    if(strcmp(buffer, "SAVEDAWS2.1\n")!=0) {
        wtr("Incorrect format ");
        fclose(f);
        return 2;
    }
    wtr("Everything was OK.\n");
    fclose(f);
    return 0;
}
~~~~

The makefile is as follows:

~~~~
all:
	zcc +zx -clib=ansi -lndos -create-app plus3file.c -subtype=plus3 -o plus3file
~~~~

I compile with no issues, but the file write fails when executing the program on Fuse:

[image]

I hope this helps,
D.

suborb, 2020-07-20:
I can't spot anything obviously wrong at a first glance, so this one is on me!

closed 2020-07-20 by suborb

DarwinNE, 2020-07-20:
Thank you very much, that makes sense to me!!!