z88dk/z88dk #1535: 48-bit genmath library crashes Z180

jblang, 2020-07-21:
The 48-bit genmath library crashes on Z180 because it uses undocumented instructions to access ixl/ixh registers. These are not supported by the Z180 (nor are any undocumented instructions) and cause a TRAP interrupt which vectors to 0H.  When running under CP/M this manifests as the program unexpectedly exiting with no explanation as to why.

IXL/IXH is used in float.asm and ldbchl.asm. A potential workaround is to load the values into some other register pair, push it and pop it back into IX or IY.  If the register is holding something important already it will need to be pushed before being modified and popped after IX is popped.   

https://github.com/z88dk/z88dk/blob/master/libsrc/math/genmath/c/sccz80/float.asm#L38

feilipu, 2020-07-21:
@jblang pending getting it fixed in next opportunity, try either math48 `-lm` or math32 `--math32` as an alternative.

jblang, 2020-07-21:
Already tried -lmath48 and it works.

jblang, 2020-07-21:
I tried math32 and it works too. I see the docs say it supports the Z180. Is that out of the box or is there an additional flag I have to use? Is the Z180 autodetected so that the same binary will work on both Z80 and Z180?

feilipu, 2020-07-21:
Support for z180 and z80n (Spectrum Next hardware), works out of the box. You just need to compile with the appropriate machine type included, `-mz180` or `-mz80n` respectively. The machine type is always associated with the target specification.

The only place where you'll need to specify the machine type is where you're using classic CP/M target `+cpm`,  the newlib CP/M target `+cpm -clib=new`, or the newlib HIBIOS target `+hbios`, as they all will default to z80.

You can check the respective [.cfg file associated with the target](https://github.com/z88dk/z88dk/tree/master/lib/config) you're using, to see whether a different machine type needs to be added.

suborb, 2020-07-21:
genmath using ixh/ixl quite extensively - I can count about 70 uses so I don't think fixing it up is viable.

Given that math48 has the same characteristics it might be sensible to switch it in as the default maths library for classic, retaining genmath for targets where using the alternate set causes issues.

feilipu, 2020-07-21:
Perhaps "won't fix", and change the default?

closed 2020-08-21 by feilipu