z88dk/z88dk #1536: (math32/mbf32/math16?/cpcmath?) Floor/ceil implementations incorrect

suborb, 2020-07-21:
At present these convert to a 32 bit integer and then back into a float to remove the fractional part.

This works well unless the integer value would take more than 4 bytes to describe. In this scenario we end up with floor(x) being significantly different to (x) with the result that, for example, ftoa() and ftoe() which end up printing garbage since they're not dividing by factors of 10.

Spotted during the the dai32math work.

feilipu, 2020-07-22:
Also noting from your commits that there are several missing intrinsic functions in math16 and math32 (for `codegen.c`)
I'll add them once the dai32math branch is merged.

feilipu, 2020-07-22:
Reading math48, the algorithm to get integer is:
- Check whether all mantissa is integer by exp too big.
- Check whether no mantissa is integer by exp too small.
- Build mask by shifting by carry exp times.
- And mask and mantissa to get integer.

Frac, floor, ceil, all derivatives of this.

feilipu, 2020-07-23:
Added the math16 routines, and streamlined the math32 routines. :+1:

closed 2020-07-24 by suborb