z88dk/z88dk #1539: CP/M argv not parsed correctly with z88dk v2.0

durgadas311, 2020-07-26:
Using the standard argv dump program:
```
#include <stdio.h>

int main(int argc, char **argv) {
        int x;
        for (x = 0; x < argc; ++x) {
                printf("argv[%d] = \"%s\"\n", x, argv[x]);
        }
        return 0;
}
```
and compiled using `zcc +cpm -o argv.com argv.c` (with or without `-set-r2l-by-default`).

It appears that any CP/M options strings (possibly special characters) at the beginning of the command cause the argument to be skipped:
```
B>argv foo [c] bar
argv[0] = ""
argv[1] = "FOO"
argv[2] = "[C]"
argv[3] = "BAR"

B>argv foo bar [c]
argv[0] = ""
argv[1] = "FOO"
argv[2] = "BAR"
argv[3] = "[C]"

B>argv [c] foo bar
argv[0] = ""
argv[1] = "FOO"
argv[2] = "BAR"

B>
```
The above was run on CP/M 3, although it should not make a difference.

suborb, 2020-07-26:
I can't reproduce this, I've tried with:

- zxcc
- ticks (uses the same code)
- On CP/M 2.2 on the SMC-777 (I can't find the right key for [ though)

[image]

The only thing that's parsed out of command line arguments is `<` and `>` for redirection.

durgadas311, 2020-07-26:
I dug up a version of CP/M 2.2, and there the problem did not reproduce. I'll study the CRT0 code and see if I can figure out why CP/M 3 shows the problem. I'd be surprised if CP/M 3 is suppressing the argument, as this C code was (supposedly) working on CP/M 3 back in the 80s, but will check that also.

suborb, 2020-07-26:
Thank you, I don't think I have any CP/M 3 images to try with. What we're doing is fairly simple:

```
        ld      hl,$80
        ld      a,(hl)
        ld      b,0
        and     a
        jr      z,argv_done
        ld      c,a
        add     hl,bc   ;now points to the end of the command line
```

And then we work backwards splitting on spaces.

durgadas311, 2020-07-26:
I ran a quick test on CP/M 3, and the CCP does not alter the commandline at 0x0080 at all, even if it begins with '['. So, I'll hunt for something that might be different between CP/M 3 and CP/M 2.2 w.r.t. CRT0.

durgadas311, 2020-07-26:
Ah, did find one subtle difference. CP/M 3 eliminates the blank space at the beginning of 0x0080 - but *only* if the commandline starts with '['. (the following program just dumps the contents of 0x0080):
```
B>cmdlin foo bar [c]
" FOO BAR [C]"

B>cmdlin [c] foo bar
"[C] FOO BAR"

B>
```

suborb, 2020-07-26:
That's really odd behaviour - any idea what the logic behind it is?

durgadas311, 2020-07-26:
Looking at the CCP3.ASM code, it searches the commandline for '[' and then stores an indicator if there are options in the command. It would seem that if the option is first in the commandline, it resets the string. Arguably a bug, but it's probably a "day 1" bug. I'm guessing the z88dk commandline parser needs to understand that it has a valid arg when the count reaches zero, instead of assuming there was nothing.

suborb, 2020-07-26:
I'm not sure I've fixed this one correctly...

durgadas311, 2020-07-26:
OK, I was trying to rebuild v2.0 with these recent fixes, but am running into issues with submodules. I assumed I could just cherry-pick the commits and run, but that did not work well.

closed 2020-07-26 by suborb

durgadas311, 2020-07-26:
Tried out the latest patches, this is working as expected now on CP/M 3. Thanks!

commit 649cd49e (2020-07-26): Fix #1539: Try to handle CP/M arguments with a [ on CPM/3

In this case the command line doesn't begin with a space so we
have to include the first character.
commit 341a8027 (2020-07-26): Fix #1539: Try again with the command line fix.

This will strip whitespace from the start.

Update ticks to pass through command line arguments to the CP/M
emulator.