z88dk/z88dk #1541: Having trouble with AMALLOC in +cpm

durgadas311, 2020-07-26:
I was trying to get the malloc facility to auto-initialize, and used -DAMALLOC and -DAMALLOC3. The program that originally showed a problem was large (46748 bytes) and it seemed that even the first call to malloc would fail. Running a simple test program:
```
#include <stdio.h>
#include <malloc.h>

char fill[42000] = { 8 };

int main(int argc, char **argv) {
        int len = 128;
        char *foo;
        foo = malloc(len);
        printf("malloc(%d) = %04x\n", len, foo);
        return 0;
}
```
I see that the first malloc performed uses a heap that thousands of byte past the end of the program. Running without the "fill[]" array, the program ends at 0x1910 but the malloc() returned 0xb693 (loss of nearly 40K). Adding the "fill[]" array caused the end of the program to be 0xbc20 and the malloc() returned 0xdf97 (BDOS starts at 0xec00) - which is a loss of over 8K.

Traditionally, and even with the latest Linux GCC, C generates a symbol "_end" in the final output which points to the end of the program (.bss), in other words it is the first unused byte in the prospective heap. I can find no evidence of such a symbol with zcc, but perhaps I am not looking in the right places. It would seem that the malloc auto-init would use the end of the BSS as the start of the heap, but that does not seem to be the case.

Since my program never frees memory, I scrapped malloc and created my own replacement. I carefully placed an unitialized (BSS) variable at the end of the last module linked, and then started the heap 256 bytes past the address of the variable (just to be safe). There does not seem to much, if any, BSS used by the libraries. But, I'd rather be able to use malloc.

I also notice that the BSS is included in the program binary, which is unfortunate - at least in some respects - because the COM file becomes larger than it needs to be (of course, ensuring BSS is zero avoids common bugs).

durgadas311, 2020-07-26:
Expanding the test a little, I see that this version of malloc is actually allocating backwards in the heap. I guess that is then "working as designed". The problem is that the stack is also allocating downward, and so we are forced to setup fixed-size heap and stack. The traditional model is for the stack to grown downwards from "top of memory", and the heap to grow upwards from end of BSS. This way, tuning (pre-allocating) of heap and stack size is generally not needed.

suborb, 2020-07-26:
Lots of things bundled up here.

There is an assembler symbol `__BSS_END_tail` which defines the end of the BSS memory, conversely `__BSS_head` is the start of BSS memory.  For RAM models we've chosen to include the BSS section in the binary - it makes it much more obvious when you've run out of RAM (thinking about targets other than CP/M here).

`AMALLOC` is a convenience way to make things work simply, it assumes that the stack address is > bss and just subdivides the space inside. You can manually do things yourself using `sbrk()` which is useful when you've got a segmented memory map.

It's easy to switch out the malloc library - just define the malloc/calloc/free/realloc functions yourself and you're free to do whatever you like - we used to have one that allocated in the "expected" direction. I'm guessing it was done in reverse to make searching for free blocks quicker - i.e. the heap pointer (usually) points to a free block initially rather than a used block. Certainly design of the heap structures has a _huge_ effect on benchmarks see: https://github.com/z88dk/z88dk/wiki/Benchmarks#binary-trees

durgadas311, 2020-07-27:
Thanks. For this particular application, I'll go with the simpler alloc-only scheme and use &_BSS_END_tail as the start of the heap. Now that I understand this malloc implementation better, I'll know how/when to use it.

closed 2020-07-27 by durgadas311