z88dk/z88dk #1542: Build problems on Linux Ubuntu 18.04.4 x86_64

durgadas311, 2020-07-26:
I'm getting the following failure when running "build.sh":
```
cc -MMD -I. -Ilib -It -g -Wall -std=gnu11 -O3 -I../common -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -g -O2 -c -o ../common/objfile.o ../common/objfile.c
../common/objfile.c: In function ‘file_rename_sections’:
../common/objfile.c:1038:8: error: ‘REG_OKAY’ undeclared (first use in this function); did you mean ‘REG_EEND’?
     == REG_OKAY)
        ^~~~~~~~
        REG_EEND
../common/objfile.c:1038:8: note: each undeclared identifier is reported only once for each function it appears in
../common/objfile.c: In function ‘file_add_symbol_prefix’:
../common/objfile.c:1128:74: error: ‘REG_OKAY’ undeclared (first use in this function); did you mean ‘REG_EEND’?
      if ((reti = regexec(&regex, str_data(symbol->name), 0, NULL, 0)) == REG_OKAY) { // match
                                                                          ^~~~~~~~
                                                                          REG_EEND
../common/objfile.c: In function ‘file_change_symbols_scope’:
../common/objfile.c:1237:74: error: ‘REG_OKAY’ undeclared (first use in this function); did you mean ‘REG_EEND’?
      if ((reti = regexec(&regex, str_data(symbol->name), 0, NULL, 0)) == REG_OKAY) { // match
                                                                          ^~~~~~~~
                                                                          REG_EEND
Makefile:71: recipe for target '../common/objfile.o' failed
```
It does not matter if I use the "v2.0" branch or master. I seem to have had this same issue back in May, but have no record of how I got past it. Any ideas of what needs to be tweaked? I see that all the REG_* defines exist in the regex/regex.h header file (and objfile.c includes that), so can't explain why it is complaining about this.

suborb, 2020-07-26:
I've seen this when the system regex library is used rather than the version in the submodules - did you clone them?

durgadas311, 2020-07-26:
It appears that, for some reason, the `#include "regex.h"` in objfile.c is picking up the Linux/GNU /usr/include/regex.h instead of ../../ext/regex/regex.h. I see that "regex.h" is not in any of the include path dirs, it is in src/copt/regex/regex.h.

durgadas311, 2020-07-26:
This is a git clone of the repository.

durgadas311, 2020-07-26:
oh, right, there are submodules. something must have been snarled in my repo. How do I get the submodules setup?

durgadas311, 2020-07-26:
ok, re-reading README.md for bits about submodules...

durgadas311, 2020-07-26:
So, originally I was trying to do an incremental build after applying a couple minor patches (commits from master). What is the proper procedure for doing that? Is there a simple way to rebuild, or do I need to scratch everything and build all over?

suborb, 2020-07-26:
There's been quite a lot of changes from 2.0 so just patching a couple of files isn't guaranteed to work as you found.

Easiest is to run `./build.sh` and grab a cup of something. That will build all 100+ libraries which is a bit painful, so if you just want CP/M go with `./build.sh -p cpm`

closed 2020-07-26 by durgadas311