z88dk/z88dk #1543: newlib - maths library using Am9511A

feilipu, 2020-07-26:
Feeling inspired by this whole thing Am9511A maths thing for the DAI, I laid out this Am9511A Module for the RC2014 today.
[image]

The Module supports using interrupt mode, with the Z80 NMI. But since the NMI jump address `0x66` clashes with CP/M it is not really useful.

The library will use the same bindings as math32, but the back end will be supported by either "demand wait" or "polling" mode access to the Am9511A. Bindings to IEEE-754 32-bit floating point will be provided, so that the use of the APU is transparent.

suborb, 2020-07-27:
Nice idea hiding it behind math32 format - if we can create 8080 variants of the converters/utility routines we should be able to get it running on the dai as well.

feilipu, 2020-07-27:
Stretch goal.
As it is modular, multi-APU config should be possible (but totally impractical).

suborb, 2020-07-27:
FWIW, I've added an AM9511a emulator taken from an Enterprise128 emulator.

I've not hooked it in, it fudges the clock cycles but it should make testing this library a bit easier when it's done.

feilipu, 2020-07-27:
> FWIW, I've added an AM9511a emulator into ticks.

Looks very useful.

The emulator uses a [fixed multiplier](https://github.com/z88dk/z88dk/blob/master/src/ticks/am9511.c#L612) for APU:CPU ticks. It would be a useful extension to make the ticks multiplier configurable, depending on the F_CPU and implementation. The APU Module above has a 1:3 multiplier off a 7.3MHz CPU. I guess the DAI is 1:1 off a 2MHz CPU. The YAZ180 is 1:16 FWIW.

I'll need to read the command register for the result flags (zero, overflow, NaN, etc) as part of the result unloading and proper conversion to IEEE-754. That was omitted from the [current routines](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/z80/f32_fam9511.asm), because of being lazy. And the `_APU_F_BUSY` would be important to have available to break out of a "demand wait" or "polling" loop. But, I guess that for the emulator adding the correct number of ticks and returning not busy flag would be nearly the same as implementing a demand wait? Predicting what the CPU Could be doing during 7,000 ticks of polling is anyone's guess.

feilipu, 2020-07-28:
[Reference for MBF32 format](https://www.wikiwand.com/en/Microsoft_Binary_Format#/Technical_details), to potentially integrate as back end to Microsoft Basic v4.7.

feilipu, 2020-08-10:
Synchronicity. Literally push the last commit on #1547, and the PCB package shows up. :tada: 
Out with the iron and scope.

closed 2020-08-11 by feilipu

feilipu, 2020-08-11:
This is version 1 of the hardware.
[image]

feilipu, 2020-08-14:
@suborb The Am9511 doesn't provide a remainder in  its divide function. Is it useful to provide a 'l_long_div' that only provides a quotient result?

suborb, 2020-08-14:
On 8080 there’s l_long_mod that returns the remainder. l_long_div on 8080 just returns the quotient. 

But on z80 the remainder is expected to be in the alt register set.

feilipu, 2020-08-14:
Yes, that's what I thought. I can't think of a simple algorithm to generate a remainder (except subtracting the result multiplied), so perhaps it is not worth doing? But perhaps even subtracting the result multiplied would be faster than doing it in software?

I'll need to check into whether and how the remainder is actually used in z80 `l_long_div`.

suborb, 2020-08-15:
I was coming to the same conclusion that a separate function might be the answer. 

Take a look at ldiv as well - that might need some changes to cope with the core division routine not returning the remainder.

feilipu, 2020-08-15:
I've done library functions for SDCC [`__mullong` `__divslong` `__modslong`](https://github.com/z88dk/z88dk/commit/d90c558abc5ede4481603bd66d7a0c81ddffb658) and I'm working on the [`__divulong` `__modulong`](https://github.com/z88dk/z88dk/commit/9f82d026ce53bda5ccdf3392017d07a405bdd8dd) functions now. I think the unsigned versions just need to reset the most significant bit of the divisor, if I'm not mistaken. Or perhaps do nothing.

I'll add the `__XXXsint` versions along the way too.

I'll also need to generate a new `ldiv()` too, by piping the quotient out early from the modulus routine.

feilipu, 2020-08-15:
Ran spectral-norm benchmark (yawn)... with the integer & long support integrated. 
Only the integer multiply is activated by the code.
RC2014 Z80 @ 7.3728Mhz

__math32__
_28 minutes 57 seconds_
1.2742190

__am9511__
_6 min 35 seconds_
1.2742140

Admittedly math32 doesn't do this particular benchmark very well, never-the-less, pretty happy with the outcome.