z88dk/z88dk #1544: [not an issue/just need for help] How can

Fabrizio-Caruso, 2020-07-27:
First of all, sorry for this non-issue. Is there a better place (forum?) where I should post questions on how understand error messages and on how to get some help and advise?
I think I am doing something wrong.

On my local working copy I am getting an error I cannot really understand:
```
z88dk_sprites_graphics.h: line 14: macro '__DRAW' redefined unidentically
        included from cross_lib/display/graphics_settings.h:13
        included from cross_lib/cross_lib.h:115
        included from cross_lib/display/display_macros.h:28
        included from cross_lib/display/alt_print/memory_mapped_alt_print.c:71
z88dk_sprites_graphics.h: line 17: macro '__DELETE' redefined unidentically
        included from cross_lib/display/graphics_settings.h:13
        included from cross_lib/cross_lib.h:115
        included from cross_lib/display/display_macros.h:28
        included from cross_lib/display/alt_print/memory_mapped_alt_print.c:71
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_a-k:287: c128_z80_80col] Error 1
```
Where is __DRAW, __DELETE defined?
The only file where __DRAW, __DELETE are supposed to be defined is only `z88dk_sprites_graphics.h`.
Is there a way to debug this kind of bugs?

suborb, 2020-07-27:
Looking at the 80column 128, I think `SEPARATE_SPRITE_DEFINITIONS` is defined. For some reason I've not got/created `z88dk_sprites_definitions.h` - does that contain a definition for `__DRAW` and `__DELETE`?

Fabrizio-Caruso, 2020-07-27:
Yes, you are right (as usual). I was not including `z88dk_sprites_definitions.h`
If I do, I get
```
Error at module 'games_chase_character_c': symbol '_sprites' already defined in module 'cross_lib_display_display_m
acros_c'
Errors in source file c:/z88dk/\lib\target\c128\classic\c128_crt0.asm:
Error at module 'games_chase_character_c': symbol '_sprites' already defined in module 'cross_lib_display_display_m
acros_c'
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_a-k:303: c128_z80_80col] Error 1
```

Fabrizio-Caruso, 2020-07-27:
I have pushed the changes into:
https://github.com/Fabrizio-Caruso/CROSS-CHASE/tree/broken_sprites

suborb, 2020-07-27:
So running this:

```
zcc +c128 -E -Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/psg -Icross_lib/sound/lcc1802/comx -Icross_lib/text -Icross_lib/input -Icross_lib/rand -Icross_lib/display/redefine_characters -Icross_lib/display/tiles -Igames/chase/generated_assets -Igames/chase/ -DBOMB_DRAW_SKIP=7 -DWALL_DRAW_SKIP=15 -DGHOST_DRAW_SKIP=1  -lndos -subtype=disk cross_lib/display/alt_print/memory_mapped_alt_print.c -D__C128_Z80__ -DZ88DK_SPRITES -DANIMATE_PLAYER -DFULL_GAME -DEND_SCREEN -DLESS_TEXT -DSOUNDS -DNO_BLINKING -DFORCE_BIT_BANG -DZ88DK_JOYSTICK -DSIMPLE_STRATEGY -DFORCE_GHOSTS_NUMBER=8 -DALT_PRINT -DNO_COLOR -E
```

Generates an error, but turning off `Z88DK_SPRITES` makes the error go away. So a couple of searches:

```
find . -type f  | xargs grep __DELETE
./cross_lib/display/display_macros.h:    __DELETE(X_OFFSET+(x),(y))
./cross_lib/display/display_macros.h:    __DELETE(X_OFFSET+(x),Y_OFFSET+(y))
./cross_lib/display/display_macros.c:        __DELETE(x,y);
./cross_lib/display/graphics_mode/buffered_graphics.h:#define __DELETE(x,y) \
./cross_lib/display/graphics_mode/cpc_cpcrslib_graphics.h:#define __DELETE(x,y) _DRAW_AUX(CPC_BLACK,space_str,(x),(y))
./cross_lib/display/graphics_mode/tgi_graphics.h:#define __DELETE(x,y) \
./cross_lib/display/graphics_mode/bit_mapped_4_graphics.h:    #define __DELETE(x,y) \
./cross_lib/display/graphics_mode/memory_mapped_graphics.h:#define __DELETE(x,y) DISPLAY_POKE(loc(x,y), _SPACE)
./cross_lib/display/graphics_mode/rex_graphics.h:#define __DELETE(x,y) DsPrintf((x),(y),16,space_str)
./cross_lib/display/graphics_mode/no_graphics.h:#define __DELETE(x,y) 
./cross_lib/display/graphics_mode/z88dk_sprites_graphics.h:#define __DELETE(x,y) \
./cross_lib/display/graphics_mode/bitmapped_graphics_laser500.h:#define __DELETE(x,y)
./cross_lib/display/graphics_mode/vectorial_graphics.h:#define __DELETE(x,y) 
./cross_lib/display/graphics_mode/conio_graphics.h:#define __DELETE(x,y) \
./cross_lib/display/graphics_mode/bit_mapped_graphics.h:	#define __DELETE(x,y) \
./cross_lib/display/graphics_mode/bit_mapped_graphics.h:	#define __DELETE(x,y) \
```

And:

```
grep ^# cross_lib/display/alt_print/memory_mapped_alt_print.c.i 
#line 1 "cross_lib/display/alt_print/memory_mapped_alt_print.c"
#line 1 "cross_lib/display/graphics_mode/memory_mapped_graphics.h"
#line 1 "cross_lib/include/standard_libs.h"
#line 1 "/home/dom/z88dk/lib/config/../..//include/sys/compiler.h"
#line 12 "/home/dom/z88dk/lib/config/../..//include/stdlib.h"
#line 1 "/home/dom/z88dk/lib/config/../..//include/sys/types.h"
#line 13 "/home/dom/z88dk/lib/config/../..//include/stdlib.h"
#line 1 "/home/dom/z88dk/lib/config/../..//include/stdint.h"
#line 14 "/home/dom/z88dk/lib/config/../..//include/stdlib.h"
#line 1 "/home/dom/z88dk/lib/config/../..//include/malloc.h"
#line 108 "/home/dom/z88dk/lib/config/../..//include/stdlib.h"
#line 17 "cross_lib/include/standard_libs.h"
#line 18 "cross_lib/include/standard_libs.h"
#line 1 "/home/dom/z88dk/lib/config/../..//include/fcntl.h"
#line 75 "/home/dom/z88dk/lib/config/../..//include/stdio.h"
#line 19 "cross_lib/include/standard_libs.h"
#line 5 "cross_lib/display/graphics_mode/memory_mapped_graphics.h"
#line 3 "cross_lib/display/alt_print/memory_mapped_alt_print.c"
#line 1 "cross_lib/display/alt_print/memory_mapped_alt_print.h"
#line 1 "cross_lib/include/standard_libs.h"
#line 5 "cross_lib/display/alt_print/memory_mapped_alt_print.h"
#line 4 "cross_lib/display/alt_print/memory_mapped_alt_print.c"
#line 1 "cross_lib/include/standard_libs.h"
#line 6 "cross_lib/display/alt_print/memory_mapped_alt_print.c"
```
I think shows that the `__DRAW` and `__DELETE` come from `memory_mapped_graphics.h` which for a c128 in z80 mode seems wrong - I think all screen data has to be passed over to the 6502?

Fabrizio-Caruso, 2020-07-27:
What do you mean by __DRAW and __DELETE being wrong? They have worked so far. They have stopped working after some re-factoring. Now I don't understand the error messages I get...
In the new "broken_sprites" branch, if I run "make attache" I get
```
z88dk_sprites_graphics.h: line 14: macro '__DRAW' redefined unidentically
        included from cross_lib/display/graphics_settings.h:13
        included from cross_lib/cross_lib.h:115
        included from cross_lib/display/display_macros.h:28
        included from cross_lib/display/alt_print/memory_mapped_alt_print.c:71
z88dk_sprites_graphics.h: line 17: macro '__DELETE' redefined unidentically
        included from cross_lib/display/graphics_settings.h:13
        included from cross_lib/cross_lib.h:115
        included from cross_lib/display/display_macros.h:28
        included from cross_lib/display/alt_print/memory_mapped_alt_print.c:71
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_a-k:227: attache] Error 1
```
If I run "make debug_attache" (which just avoids `memory_mapped_alt_print.c`, I get this error:
```
z80asm -I"C:\cygwin\tmp\zcc00001EE4BA7D18" -D__SCCZ80  -s  -mz80 -I"c:/z88dk//lib" -I"c:/z88dk/\lib\target\cpm\clas
sic"  "C:\cygwin\tmp\zcc00001EE4BA7D19.asm"

z80asm -mz80 -b -d  -o"../build/FULL_attache.bin" -m -s -L.   -I"C:\cygwin\tmp\zcc00001EE4BA7D18" -D__SCCZ80 -L"c:/
z88dk//lib/clibs" -I"c:/z88dk//lib" -iattache -icpm_clib -i"z80_crt0"     "@C:\cygwin\tmp\zcc00001EE4BA7D20.lst"
Error at module 'games_chase_character_c': symbol '_sprites' already defined in module 'cross_lib_display_display_m
acros_c'
Errors in source file c:/z88dk/\lib\target\cpm\classic\cpm_crt0.asm:
Error at module 'games_chase_character_c': symbol '_sprites' already defined in module 'cross_lib_display_display_m
acros_c'
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_a-k:245: debug_attache] Error 1
```

I am not sure I understand your question about passing screen data to the 6502.
c128_z80 is the C128 target compiled by Z88DK (which uses the Z80), whereas c128 is the C128 target compiled by CC65.

All graphics assets (shapes of tiles/sprites/udg, etc.) are shared among all targets (not just Zilog 80 and MOS 6502 targets but also Motorola 6809, COSMAC, RCA 1802, etc. targets). This cross-dev-kit part is handled by Python scripts that generateds the .s/.asm/.h files in src/games/<game name>/generated_assets, which any target uses.

suborb, 2020-07-27:
I was indicating that you're getting `__DRAW` from `memory_mapped_graphics.h` and also from `z88dk_sprites_graphics.h`.

I was just musing as to which one is correct one for this target.

Fabrizio-Caruso, 2020-07-27:
Thanks! Sorry for not understanding. 
Remark: `memory_mapped_alt_print` is a misnomer as it now includes code for nearly all graphics modes. I need to rename this to something more generic.

If you mean that _DRAW is defined in `memory_mapped_graphics.h` then there is a problem.

Indeed the C128_z80_80col target in CrossLib (my abstraction layer) is a target that uses Z88DK software sprites and not pokes/vpokes into video memory to draw objects (as for memory_mapped_graphics).
If _DRAW is also defined in memory_mapped_graphics, then I am wrongly including memory_mapped_graphics which I should not.

Fabrizio-Caruso, 2020-07-27:
@suborb you are right about a wrong   ` #include "memory_mapped_graphics.h"` in `memory_mapped_alt_print` when used for Z88DK_SPRITES.

If I avoid that I get:
```
Error at module 'games_chase_character_c': symbol '_sprites' already defined in module 'cross_lib_display_display_m
acros_c'
Errors in source file c:/z88dk/\lib\target\c128\classic\c128_crt0.asm:
Error at module 'games_chase_character_c': symbol '_sprites' already defined in module 'cross_lib_display_display_m
acros_c'
make: *** [makefiles/targets/makefiles_z88dk/Makefile_z88dk_a-k:303: c128_z80_80col] Error 1
```

suborb, 2020-07-27:
I think that's because `display_macros.c` and `games/chase/character.c` both include `z88dk_sprites_definition.h`

Fabrizio-Caruso, 2020-07-27:
You are right. I have now managed to get it to compile but the sprite data is not read correctly. This maybe a totally different issue that I have to figure out before I bother you again.

suborb, 2020-07-27:
Okay.  I’ll close this one then.

closed 2020-07-27 by suborb

Fabrizio-Caruso, 2020-07-27:
@suborb Thanks a lot for your help!
I have pushed the changes that get everything to compile into the broken_sprites branch.

The problem I face now is that sprites do not get the right data:
[image]

Should I post this as a new issue? Can I post it here? Should I rather post it on a forum?

Fabrizio-Caruso, 2020-07-27:
It seems to get better if I add  `extern uint8_t sprites[];`
Maybe I need this on each file that refers to `sprites`. Don't I?

Fabrizio-Caruso, 2020-07-27:
It seems to work now! Testing it...