z88dk/z88dk #1545: Nightly compile errors out with "make[1]: *** [Makefile:37: Far_Pointer_Call_ieee.opt] Fehler 1"

bastetfurry, 2020-07-28:
Tried to compile z88dk according to the manual over here https://github.com/z88dk/z88dk/wiki/installation and got a failing test, Far_Pointer_Call_ieee.c seems to be the culprit.

The relevant output of the build process:
```
zcc +test -fp-mode=ieee -vn -a Far_Pointer_Call_ieee.c -o tmp1.opt
diff -w tmp2.opt results/Far_Pointer_Call_ieee.opt
23c23
< 	ld	hl,2
---
> 	ld	hl,6
31c31
< 	ld	a,1
---
> 	ld	a,3
36d35
< 	call	l_gchar
38a38
> 	call	l_gchar
55c55
< 	ld	hl,2
---
> 	ld	hl,6
63c63
< 	ld	a,1
---
> 	ld	a,3
68,69d67
< 	call	lp_gchar
< 	call	l_sxt
71a70,71
> 	call	lp_gchar
> 	call	l_sxt
```

My system is a Debian Bullseye running on an AMD Ryzen 5 1600, if that matters.

suborb, 2020-07-28:
It does look there's another (older) sccz80 somewhere in the path.

If you run `zcc +test -fp-mode=ieee -vn -a Far_Pointer_Call_ieee.c -o tmp1.opt` in testsuite and take a look at tmp1.opt does the Version line at the top of the file match what you checked out?

bastetfurry, 2020-07-28:
Nope, no zcc in my path:

```
bastetfurry@katzenrechner:~/Compile/z88dk/testsuite$ zcc +test -fp-mode=ieee -vn -a Far_Pointer_Call_ieee.c -o tmp1.opt

Command 'zcc' not found, did you mean:
[...]
```

suborb, 2020-07-28:
Just a hunch, can you supply the output of `locale`?

bastetfurry, 2020-07-28:
Hmm, yes, this is a German install. :sweat_smile: 

```
bastetfurry@katzenrechner:~/Compile/z88dk/testsuite$ locale
LANG=de_DE.UTF-8
LANGUAGE=
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=
```

suborb, 2020-07-28:
It looks like sccz80 gets the `"C"` locale by default so your settings shouldn't have anything to do with it. Forcing the locale to be `de_DE.UTF-8` in sccz80 gets a completely different set of errors so that eliminates any locale settings.

I have managed to reproduce the exact output as you by changing the `4` on `src/sccz80/main.c:250` to `0`. This is where `4` indicates the size of variables of that type. If that's being compiled incorrectly then all bets are off regarding what might be wrong elsewhere.

I've not got Bullseye environment so I think I'm going to have to set up a VM to see what's happening.

suborb, 2020-07-28:
I've pushed a fix for this issue - I'm not sure exactly why it was happening (it seems to be a gcc9 optimiser issue). If you get the 20200729 nightly tomorrow then you should building success.

bastetfurry, 2020-07-29:
Thanks, ill try this evening. :D

bastetfurry, 2020-07-29:
Now i get the next one. :(

```
make[1]: Verzeichnis „/home/bastetfurry/Compile/z88dk/testsuite“ wird betreten
zcc +rcmx000 -vn -a rabbit/rabbit_optimisations.c -o tmp1.opt
copt: can't open patterns file
make[1]: *** [Makefile:29: rabbit/rabbit_optimisations.opt] Fehler 1
make[1]: Verzeichnis „/home/bastetfurry/Compile/z88dk/testsuite“ wird verlassen
make: *** [Makefile:168: testsuite] Fehler 2
```

And the compiler thinks this is a Windows machine somehow when i try to run this manually. o.o
```
bastetfurry@katzenrechner:~/Compile/z88dk/testsuite$ ../bin/zcc +rcmx000 -vn -a rabbit/rabbit_optimisations.c -o tmp1.opt
Can't open config file c:/z88dk///lib/config/rcmx000.cfg
```

suborb, 2020-07-29:
That's erm, "impressive".

I think `export ZCCCFG=/home/bastetfurry/Compile/z88dk/lib/config` should solve it.

I've no idea how you've managed to get a default path for windows *unless* you grabbed the windows kit by accident.

bastetfurry, 2020-07-29:
> ... you grabbed the windows kit by accident ...
*looks into download directory* *headdesk*
I just pressed END on that page and downloaded the latest package without even realizing that this was for Windows instead of just grabbing "z88dk-latest.tgz".
It compiled now and works. <3

suborb, 2020-07-30:
Oops! That's great that you're up and running.

closed 2020-07-30 by suborb