z88dk/z88dk #1553: sccz80 - l_xxx_mod support

feilipu, 2020-08-15:
Would it be possible (at some stage) to get SCCZ80 to generate `l_xxx_mod` specifically?
To replicate SDCC `__mod_long` and `__mod_int`.

Currently SCCZ80 generates `l_xxx_div` `exx` instructions, assuming that the remainder will be found in `dehl'` registers.

In some situations it would be useful to provide a specific modulus function. #1545

suborb, 2020-08-15:
Working on this, assume that it exists - it'll go in tomorrow at some point.

I'm just doing longs since 16 bit division probably won't benefit too much from the APU.

closed 2020-08-16 by suborb

feilipu, 2020-08-16:
Works as anticipated. :+1:

feilipu, 2020-08-16:
Done some benchmarking, and good call to stop at `long`.

Adding `__mulint` or `l_mult`respectively using the APU slows things down appreciably. At least for a 7MHz z80 platform.
It adds 25 to 30 seconds onto each of the below times, although the binary is 200 Bytes smaller.

spectral-norm with am9511
1.2742140

__sdcc / newlib__
6 min 10 seconds - fast integer math

__sccz80 / newlib__
5 min 48 seconds - fast integer math
5 min 28 seconds - fast integer math & `--opt-code-speed=inlineints`