z88dk/z88dk #1556: classic - warnings in CP/M Library

feilipu, 2020-08-19:
Noted some warnings appearing when building classic CP/M.
```
--- Building CP/M Library ---

make[1]: Nothing to be done for 'all'.
make[2]: Nothing to be done for 'all'.
make[2]: Nothing to be done for 'all'.
Makefile:27: *** mixed implicit and normal rules: deprecated syntax
dir_get_entry_size.c:18:43: warning: Assigning 'szptr', type: unsigned char * szptr from unsigned int  [-Wincompatible-pointer-types]
stat.c:85:9: warning: Assigning 'sfc', type: struct sfcb * sfc from struct fcb * fc [-Wincompatible-pointer-types]
dir_get_entry_size.c:18:43: warning: Assigning 'szptr', type: unsigned char * szptr from unsigned int  [-Wincompatible-pointer-types]
parsefcb.c:42:19: warning: Assigning 'ptr', type: unsigned char * ptr from char *  [-Wincompatible-pointer-types]
stat.c:85:9: warning: Assigning 'sfc', type: struct sfcb * sfc from struct fcb * fc [-Wincompatible-pointer-types]
dir_get_entry_size.c:18:43: warning: Assigning 'szptr', type: unsigned char * szptr from unsigned int  [-Wincompatible-pointer-types]
stat.c:85:9: warning: Assigning 'sfc', type: struct sfcb * sfc from struct fcb * fc [-Wincompatible-pointer-types]
dir_get_entry_size.c:18:43: warning: Assigning 'szptr', type: unsigned char * szptr from unsigned int  [-Wincompatible-pointer-types]
parsefcb.c:42:19: warning: Assigning 'ptr', type: unsigned char * ptr from char *  [-Wincompatible-pointer-types]
stat.c:85:9: warning: Assigning 'sfc', type: struct sfcb * sfc from struct fcb * fc [-Wincompatible-pointer-types]
TARGET=cpm TYPE=z80 DEVICE=nodevice z80asm -d -I/home/phillip/Z80/z88dk/lib/config//../ -DSTANDARDESCAPECHARS -x/home/phillip/Z80/z88dk/libsrc//cpm_clib @./cpm.lst

```

suborb, 2020-08-19:
I think they've been there for a while, I'll stick a cast in since I think it's just going from external to internal representation of the same structure.

The Makefile one is a bit funky so I'll look at that too.

suborb, 2020-08-29:
I've casted away the warnings. I can't reproduce the makefile complaint but I'll keep an eye open for similar issues in the future.

closed 2020-08-29 by suborb