z88dk/z88dk #1558: Z88DK nightly fails to build.

KrzysztofJeszke, 2020-08-26:
`
svn checkout -r 11722 https://svn.code.sf.net/p/sdcc/code/trunk/sdcc -q /tmp/sdcc
svn: E175012: Connection time is up
make: *** [Makefile:74: bin/zsdcc] Error 1
`

suborb, 2020-08-29:
I haven’t seen that one lately but yes sometimes the sdcc svn server is unreliable. 

I want to stop building sdcc all the time but I’ve not had the chance yet unfortunately. 

In terms of the build I think it’s only used for one of the z80asm tests so there’s not a hard dependency so we could remove it.

pauloscustodio, 2020-08-30:
I'll have a look at the tests to see if I can skip it if sdcc is not compiled.

Another way is maybe to add the sdcc extraction to the .cache

commit 91112642 (2020-09-03): Merge pull request #1562 from z88dk/issue_1558

Fix #1558 - build fails due to svn timeout
commit b62f20db (2020-09-03): Fix #1558 - build fails due to svn timeout

Make the build of sdcc optional (invoke with make BUILD_SDCC=1 to build).
Skip the z80asm tests that depend on sdcc, if zsdcc not found.
closed 2020-09-03 by pauloscustodio