z88dk/z88dk #1560: z88dk-ticks - profiling basic applications

feilipu, 2020-08-30:
Noting that `z88dk-ticks` has been enhanced to support both an Am9511A emulation function, and loading of MS Basic from RC2014.

Is there a way to use `z88dk-ticks` to profile a Basic application loaded into the code (MS Basic) running in `ticks`?

This would imply `z88dk-ticks` running in an interactive emulator mode, as the Basic program would have to be interpreted.

Does this work?

suborb, 2020-08-30:
There’s no real way to load programs into the environment at the moment - there’s a serial port but it’s hooked up to the console - it will need to be repointed to a pipe/socket in the same way that mame serial ports can be. 

Assuming that’s done you set a breakpoint on the run command. Turn on hotspots and continue. On exit a hotspot file will be created with instruction counts and cycles. 

It might be best to convert the basic source to z80asm - it’s only a few replacements so that you don’t need to mangle the map file.

feilipu, 2020-08-31:
I'll have a go at converting the source for `bas32k.asm` soon. Now that everything is working "perfectly", it is the time to start breaking and reworking. 😄

I keep archive copies of the original MS Basic, and Searle's modifications in my repo too, so that it is easy to diff out the Z80 op code optimisation and other changes as and if needed.

If `ticks` has a serial port, and an interactive mode, then yes `HLOAD` should do the trick. That is really how to see how to optimise Basic using eg. `bubblesort.bas` or `mandelbrot.bas`. I hope to see what's slowing things down.

feilipu, 2020-09-04:
Instructions as followed.
```
% ./z88dk-ticks -x rc2014/bas32k.map -w 30 -debug -pc 0x250 rc2014/bas32k.bin

debugger> break RESET
debugger> hotspot on
debugger> cont

[Paste in program]

RUN

[Wait]

RESET

debugger> quit

```
The file hotspots will be generated. To sort by cycles:
`% sort -nrk2 hotspots > hotspot-cycles`

To sort by iterations:
`% sort -nrk1 hotspots > hotspot-iterations`

closed 2020-09-04 by feilipu