z88dk/z88dk #1561: ticks or stdlib no longer detecting EOF?

pauloscustodio, 2020-09-03:
When compiling enigma.c with the command line:
`zcc +test -oenigma.bin examples/console/enigma.c`

and running with ticks:
`z88dk-ticks enigma.bin`

I get the prompt from Enigma, but typing the text string and sending an EOF (Ctrl-D) is not recognized - the program apparently hangs.

Detected with a z80asm test failing, that just runs `z88dk-ticks enigma.bin <infile >outfile'

I will try a git bisect to check where the problem started.

suborb, 2020-09-03:
I never realised that ^D ever worked that way with +test/ticks!

The usual way with enigma to end the encoding is to enter a "."

pauloscustodio, 2020-09-03:
I meant: type "HELLO.", newline and Ctrl-D.

Anyway the problem in Win32-specific. Just tried on OSX without issues.

pauloscustodio, 2020-09-04:
Found the "culprit" - replacement of getchar() by getch() in ticks:
`
766237b2b6f16341ddca15017039b4b52bbf9cba is the first bad commit
commit 766237b2b6f16341ddca15017039b4b52bbf9cba
Author: suborb <dom@suborbital.org.uk>
Date:   Fri Aug 28 21:29:32 2020 +0100
`

getch() (in Win32) does not read from a file redirected to standard input, causing the z80asm test to fail.

Will fix by calling getchar() if redirecting input.

commit fed56f88 (2020-09-04): Fix #1561

Call getchar() in cmd_readkey() if input is redirected from a file.
closed 2020-09-04 by pauloscustodio