z88dk/z88dk #1564: Hardware debugger for zx?

desertkun, 2020-09-07:
I am new to this project so I might be stepping over something that has been discussed before.

That being said, I want to discuss an idea of hardware debugger device and debug library, once linked with, the program should become debuggable on real hardware.

1. A debugger device with custom ROM that would have routines required to operate a debugging session: to set/remove a breakpoint, do a single step, evaluate expression or/and at least obtain memory values and registers. I imagine such device could be attached to spectrum via user port and could communicate with a PC/mac via USB. An STM32F1 implementation comes to mind.

2. Because we have `--c-code-in-asm` we could translate source lines to machine code addresses. The program should also have an allocated block of breakpoints somewhere in memory as part of debugging library the program is linked with. 
3. Once breakpoint is set, the debugger issues a `NMI` routine, which should replace the first byte of the instruction that should break, with e.g. `RST 8`. A new breakpoint structure is also placed on the list I have mentioned before, that should also contain the original instruction that `RST 8` has corrupted (or replaced).
4. The custom ROM should contain a breakpoint-hit-handling routine which could be as follows:
  a. Look on the stack to see what breakpoint has been hit.
  b. Report to the debugging device that it's being hit.
  c. `EI` && Fix stack's return address to be the next instruction && `HALT` && `RET`
  d. Once `continue` is issued, the debugger should issue an interrupt routine that should execute the original instruction via `Interrupt Mode 0`.  Since `PC` should remain intact, it should fall through into the `RET` that goes after `HALT` that should pop us to the next instruction.

What do you think? Is there future to this idea? I am thinking of implementing this for `FUSE` emulator first and then see how it goes.

feilipu, 2020-09-09:
There is a [somewhat related issue](https://github.com/RC2014Z80/RC2014/issues/16) from Philipp (@spth, an author of SDCC), at the RC2014 github. Where he is looking for a way to execute the SDCC tests on real Z80 hardware in an automated way.

One option I presented was to use [an extension of MS Basic](https://github.com/feilipu/NASCOM_BASIC_4.7/tree/master/rc2014_NascomBasic32k) that I've been working on, including new keywords `HLOAD`, and `RESET`, which would permit automated tests.

This extension to MS Basic also includes a [RAM Jump table for the `RST` locations](https://github.com/feilipu/NASCOM_BASIC_4.7/blob/master/includes/z80intr.asm#L74), which can be rewritten at run time. Note that `RST08` is occupied by the Tx routine. The `RST30` is free, and could be used for debugging purposes. `RST38` is the ZSID/SID default, but that is used by the ACIA in IM1.

This issue's proposal would extend the support to allow placing breakpoints into any address, and also to allow recognition of the NMI (on the RC2014 Extended Bus). Setting and handling of breakpoints could be done from Basic using `PEEK`, `POKE`, `DEEK`, `DOKE`, and memory for the break-point handling could be placed before the default z88dk origin of `0x9000`.

Have a go in Basic... If you get functionality you like, we can add it in assembly inside Basic.

aralbrec, 2021-02-10:
There is debugging hardware in the zx spectrum next with debuggers working via uart connected to pc.  The uart connection can be wired (through the joystick) or wireless (via esp8266 but none are using that atm afaik).  These debuggers are using hardware derived from either the divmmc or the multiface.

The divmmc in the zx next is similar to the original peripheral that causes itself to page its own memory into the bottom 16K when instructions are fetched from specific addresses (this is for emulating tape on sd card and the beta disk system on sd card) and when instructions are fetched from some rst$xx entry points.  These fetches are normally trapped "late" where the first byte is fetched from normal memory before the change in mapping is applied.  This has a certain advantage when emulating the mentioned devices in that the software can tell if the entry point is being reentered while the divmmc memory is paged in.  But it's not so good for debugging as it requires the byte fetched in normal memory to be single opcode byte or known, which it is since the entry points are all in the spectrum rom on the original machine.  In a general zx next or z80 program that is not necessarily the case.

In the zx next, the divmmc is changed to be more suitable for debugging.

Entry points include all the rst$xx locations and the same memory address traps but these are made individually programmable to be on/off or immediate/late.  The former so you can turn individual entry points off and the latter to allow trapping of instruction fetches before bytes are read from main memory.  This allows traps to be executed with rst$xx instructions or by pressing the nmi button which generates an instruction fetch from 0x66.  The debugger can temporarily write rst$xx instructions into the running code to embed breakpoints.  Code inside the divmmc memory, which is separate from the z80's normal memory space, communicates with the pc via uart.  The zx next has the additional advantage of allowing the z80's memory space to be paged into any 8k location so it is still possible to read the running program's memory even if it is underneath the bottom 16k which the divmmc memory overlaps.  It operates in a similar way to the rc2014's mapper except it operates on 8K pages rather than 16K pages.

An nmi button activation is useful for the user to get the debugger's attention (this generates an effective rst$66) but on systems with physical z80s, this actually pollutes the running program's state by pushing a return address there.  This can cause problems for programs that use the stack pointer for computations or as a memory pointer.  The zx next contains a synthesized z80 so it has been modified to allow a stackless nmi option where the return address is written to an internal register rather than the memory and the first `retn` instruction afterward reads the return address from that register instead of memory.  This eliminates the possibility of corrupting the program when the nmi button is activated but it is not something you can do for a physical z80 unfortunately.  I did the hardware for this feature and I did try to think of a way to do it without modifying the z80 but there wasn't a reliable way to do it that distinguished between nmi response, a call to 0x66 or a plain memory read/write to 0x66.  Any externally applied logic would be imperfect.  Or maybe I didn't look hard enough as the modification to the z80 was so simple it was preferred :)

The next's operating system (nextzxos) actually uses a second nmi originating from multiface hardware to enter its debugger and only uses the divmmc to trap some rst$xx and the tape entry points.  The hardware also allows the generation of nmi interrupts from software by writing a specific register.  An nmi can be generated for the divmmc nmi button or the multiface nmi button and that can be used as a means to reenter the debugger in place of embedding rst$xx instructions.  The register bits are rising edge activated and the register continues to hold the bits set after the fact so that the nmi routine can read those bits to find out if the nmi was caused by user button (bits clear) or software (bits set).  To arm the bits, they have to be reset to zero by the nmi routine.

The current debuggers are using the divmmc to run for various reasons but a truly invisible debugger can only be done via the multiface hardware.

1. z80 has a stackless nmi mode
2. software can generate nmi via register write for breakpoints
3. user can generate nmi via button to enter debugger
4. software can distinguish between button or software interrupt
5. multiface is invisible except when activated via nmi
6. multiface hw only pages into the bottom 16k on nmi to service the interrupt
7. the z80 system can still see the bottom 16k overlaid by the multiface by allowing its memory to be paged in elsewhere

feilipu, 2022-05-11:
Is this now resolved with #1843 and following PRs?

desertkun, 2022-05-11:
Correct. The hardware debugger (aka gdbserver) itself is here https://speccytools.org/

feilipu, 2022-05-11:
Ok to close then?

spth, 2022-05-11:
> Correct. The hardware debugger (aka gdbserver) itself is here https://speccytools.org/

To really use that to debug C code, I guess ELF/DWARF output in SDCC will be needed?

feilipu, 2022-05-12:
Ok, one more done. +1

closed 2022-05-12 by feilipu