z88dk/z88dk #1565: 8080 embedded target

GwenNelson, 2020-09-08:
Hi all

Is this possible?

Basically, I want the equivalent to the +z80 target, but with only 8080 code. Currently i'm doing this:
zcc +z80 --list -m8080  -v -set-r2l-by-default  -startup=0 main.c printf.asm -o ctest.bin -create-app

Where printf.asm is my own implementation of printf.

I also use the following CRT pragmas:
#pragma output CRT_ORG_CODE = 0
#pragma output CRT_ORG_VECTOR_TABLE = 0
#pragma output CRT_ENABLE_RST = 0
#pragma output CRT_ENABLE_NMI = 0
#pragma output CRT_INCLUDE_PREAMBLE = 1

my crt_preamble.asm simply sets up the stack pointer and then does a call to _main:
```
        GLOBAL _start
        EXTERN _main

_start:
        ; reset UART
        mvi a,03h
        out 10h

        mvi a,15h
        out 10h

        ; setup SP
        lxi h,_stack_top
        sphl

        call _main
_loop:  nop
        jmp _loop
        ret

SECTION bss
        _stack_bottom:  ds 128
        _stack_top:
```

The final result will then run on my altairduino, but there must be a more standard or official way to do this

suborb, 2020-09-08:
Newlib doesn't officially support 8080 machines - so I suspect things are only working through chance and not using many library routines.

Classic does support the 8080, so the best path (IMO) will be to setup a target for barebones 8080/altair there - it'll be easiest to start with +test and rip out the bits not needed and add the serial console routines.

suborb, 2020-09-29:
@GwenNelson I was procrastinating on +ondra and quickly added an Altair8800 port in this commit: https://github.com/z88dk/z88dk/commit/f0a9acd676f37d86c98b847a70434a08d4f4d94e

Hopefully this will be very similar to what you need to do.

GwenNelson, 2020-09-30:
@suborb that looks like exactly what I want, thanks!

suborb, 2020-10-01:
Perfect, I'll close this one now then!

closed 2020-10-01 by suborb