z88dk/z88dk #1573: z80asm: -O doesn't affect location of .def, .o, or .map files

hpax, 2020-09-24:
The -O option does not change the output location of .def, .o, or .map files; they continue to be written to the location where the source module (first source module for .def) is located.

pauloscustodio, 2020-09-24:
The -O option changes the output location of those files. Can please you provide an example command line that fails?

hpax, 2020-09-24:
[test1573.zip](https://github.com/z88dk/z88dk/files/5277954/test1573.zip)

Here are three tests.
Note the following:
1. The module-specific output files (.o .lis) are written into output/input/ not output/ . This is a big problem, because one would expect the output directory to be independent of path, as one major use case is to be able to build into a different tree than the source files live in. In particular in the presence of .. paths this is a major problem.
2. The -o option overrides the names of .bin and .reloc, but *not* .def and .map, which are linker outputs, not per-module outputs.
3. [this is fine, and probably preferable, but may want to be clarified]: the -o option overrides the -O option; if both are specified the -O directory has to be replicated in the -o option if that behavior is desired.

Try the "b" test running the scripts in the "objtree"directory; the "srctree" directory hierarchy should *NOT* be written into.

[test1573b.zip](https://github.com/z88dk/z88dk/files/5278069/test1573b.zip)

hpax, 2020-09-24:
Correction: two archives with three test cases each.

hpax, 2020-09-24:
Another issue in the same vein:

The -o option overrides -O for the main .bin file, and *all* .reloc files regardless of if they refer to a split section, but *not* any split-section bin files...

hpax, 2020-09-24:
In case you're wondering what the heck I'm doing...

https://git.zytor.com/abc80/abc80-1m-sram.git/tree/sw/baspp.in
https://git.zytor.com/abc80/abc80-1m-sram.git/tree/sw/baspp.asm

... to enable fairly transparent mixed-language, modular programming like ...

https://git.zytor.com/abc80/abc80-1m-sram.git/tree/sw/startram.bx
https://git.zytor.com/abc80/abc80-1m-sram.git/tree/sw/detect.asm
https://git.zytor.com/abc80/abc80-1m-sram.git/tree/sw/startram.asm

pauloscustodio, 2020-09-24:
Nice!

pauloscustodio, 2022-01-09:
1. test1.sh:
The -O directory is prepended to the source path; to achieve the expected result, do:
```
cd input; z88dk-z80asm -O../output -b -l -g -m -reloc-info file1.asm file2.asm
```
2. test2.sh:
Fixed in z80asm; now the -o also changes the name of the map and def files, and the -O directory is prepended to the -o file name.

3. Consequence of the fix; now -O directory is prepended to -o path

commit aa539a07 (2022-01-09): Merge pull request #1941 from z88dk/issue_1573

Fix #1573: now map and def files use same basename as bin file
commit c71bddf9 (2022-01-09): Fix #1573: now map and def files use same basename as bin file
output bin file (-o option) is stored in output directory (-O option)
Impact in zcc - no longer need to copy map and def files, arguments to appmake .
closed 2022-01-09 by pauloscustodio