z88dk/z88dk #1578: z88dk-link: proposal for new linker

pauloscustodio, 2020-09-25:
There are several issues/feature request open that are difficult to implement in the current implementation of z80asm, due to the single-pass nature of the assembler. Changes to the assembler impact the linker and vice versa.

References:
- #1576 z80asm: request more control over the placement of sections
- #1290 (z80asm) Linker support to banked memory Iter 2 
- #1287 (classic) Complete gameboy support
- #860 z80asm: building scalable memory maps
- #832 (z80asm) Leaving a hole in section padding
- #762 z80asm: align directive not working in newlib memory map
- #716 z80asm: new directive DISP
- #692 (z80asm/classic) Pulling in unnecessary library code
- #391 Add z88dk-map to toolset that can translate symbol files for different tools
- #346 z80asm: generating an empty binary for last defined section
- #342 z80asm: new section model supporting overlays
- #333 z80asm: map file (symbol info) needs some more information on memory banks and symbol types
- #243 REL format linking
- #19 z80asm: memory map composition using sections (problem)
- #13 z80asm: .reloc for runtime binary relocation

The objective is to build a command line linking tool that can replace the linker in z80asm and that allows the above mentioned issues to be solved and allows for easier evolution.

hpax, 2020-09-28:
Perhaps a stupid idea, but have you considered the possibility of simply using GNU binutils for the linker, using ELF as the object file format? z80-elf is already a supported binutils target. If there are z88dk-specific configuration options (suitable default linker script[s], additional z80 instructions etc.) I'm sure the binutils team would happily accept them. The upside is leveraging a *very* complete framework for linking as well as object file management in general, plus a widely supported file format (ELF) with very good debug support (DWARF); the downside of course is that it is a heavyweight external dependency with addtional dependencies of its own.

I have verified that it does, indeed, support both low byte and high byte relocations:

	.section ".text","ax"
hello:
	ld hl,hello_str
	ld bc,hello_len
	call 0x05
	ret

world:
	ld h,hello_str >> 8
	ld l,hello_str
	ld bc,hello_len
	call 0x05
	ret
	
	.section ".rodata","a"
hello_str:
	defm "Hello, World!\r\n"
hello_len = . - hello_str

Relocation section '.rela.text' at offset 0x110 contains 3 entries:
 Offset     Info    Type            Sym.Value  Sym. Name + Addend
00000001  00000804 R_Z80_16          00000000   .rodata + 0
0000000b  00000808 R_Z80_BYTE1       00000000   .rodata + 0
0000000d  00000801 R_Z80_8           00000000   .rodata + 0

After using ld to link to address 0x1ff0:

Disassembly of section .text:

00001ff0 <hello>:
    1ff0:       21 05 20        ld hl,0x2005
    1ff3:       01 11 00        ld bc,0x0011
    1ff6:       cd 05 00        call 0x0005
    1ff9:       c9              ret

00001ffa <world>:
    1ffa:       26 20           ld h,0x20
    1ffc:       2e 05           ld l,0x05
    1ffe:       01 11 00        ld bc,0x0011
    2001:       cd 05 00        call 0x0005
    2004:       c9              ret

The best case, of course, would be if gas could be modified to accept z80asm syntax; I don't know how much work it would be, but I have done binutils hacking before and found it pretty straightforward to handle target-specific syntax in gas -- but that would probably depend on how the z80 version is currently implemented. A list of differences/missing features would be a good place to start.

hpax, 2020-09-28:
For what it's worth, there is a "-sdcc" option to gas already; the full set of architecture-specific options are:

CPU model options:
  -march=CPU[+EXT...][-EXT...]
                          generate code for CPU, where CPU is one of:
  z80                     Zilog Z80
  ez80                    Zilog eZ80
  gbz80                   GameBoy Z80
  r800                    Ascii R800
  z180                    Zilog Z180
  z80n                    Z80 Next
And EXT is combination (+EXT - add, -EXT - remove) of:
  full                    assemble all known instructions
  adl                     eZ80 ADL mode by default
  xyhl                    instructions with halves of index registers
  infc                    instruction IN F,(C)
  outc0                   instruction OUT (C),0
  sli                     instruction known as SLI, SLL, or SL1
  xdcb                    instructions like RL (IX+d),R (DD/FD CB dd oo)

Compatibility options:
  -local-prefix=TEXT      treat labels prefixed by TEXT as local
  -colonless              permit colonless labels
  -sdcc                   accept SDCC specific instruction syntax
  -fp-s=FORMAT            set single precision FP numbers format
  -fp-d=FORMAT            set double precision FP numbers format
Where FORMAT one of:
  ieee754                 IEEE754 compatible (depends on directive)
  half                    IEEE754 half precision (16 bit)
  single                  IEEE754 single precision (32 bit)
  double                  IEEE754 double precision (64 bit)
  zeda32                  Zeda z80float library 32 bit format
  math48                  48 bit format from Math48 library

hpax, 2020-09-28:
The set of relocations supported seem to be:

R_Z80_8: 8-bit absolute
R_Z80_8_DIS: 8-bit index register displacement
R_z80_8_PCREL: 8-bit relative jump (JR, DJNZ...)
R_Z80_16: 16-bit absolute
R_Z80_24: 24-bit absolute
R_Z80_32: 32-bit absolute
R_Z80_BYTE0: first byte
R_Z80_BYTE1: second byte
R_Z80_BYTE2: third byte
R_Z80_BYTE3: fourth byte
R_Z80_WORD0: low word (byte 1:0)
R_Z80_WORD1: high word (byte 3:2)
R_Z80_16_BE: 16-bit absolute address in *bigendian* format

suborb, 2020-09-28:
One of the problems I had in the past regarding switching linkers was that some of the library had complex expressions as relocations. 

There will be a handful still in classic (which aren’t needed) but I’m mindful of how many are in users code. 

The above was certainly a problem with vasm/vlink when I was playing around with it years ago. I’ve not played with gas in even longer to know whether it’s an issue there.

Getting all the debugging stuff mostly for free is + point for me

The z80asm syntax additions are hopefully not too onerous to add either.

pauloscustodio, 2020-09-28:
I am drawn back by the heavy weight of the GNU binutils for our uses. I have looked into ELF as an object module format in the past, and it just seems an overkill. Also, as mentioned by @suborb, the relocation expressions are much simpler than what we are using now, where all expressions are computed at link time. I do not know how much of the user code would break by forcing relocations to be just base+offset.

Regarding debugging, I do not expect that the Z80 emulators out there can understand DWARF (but please correct my if I'm wrong). The one I usually use, Eightyone, is very simple and assembly-language level only.

My plan for the moment is separating the assembly and link phase of z80asm (as they are now one unit) while porting the linker to C++; this is a simple step that will allow several shortcomings of z80asm to be solved.

hpax, 2020-09-28:
On 2020-09-28 11:29, suborb wrote:
> One of the problems I had in the past regarding switching linkers was
> that some of the library had complex expressions as relocations.
>
> There will be a handful still in classic (which aren’t needed) [...]

ELF generally cannot support arbitrary relocations, and I don't believe
gas can, either. Then again... a relocating loader would have to pretty
darn advanced to do that too.

ELF *can* support symbol addition and/or subtraction, but the linker
needs to be appropriately enabled.

It *could* be possible, perhaps, to put an expression in as the name of
a symbol, and then make ld calculate it at runtime as if it had been
given in the linker script. No idea how much work that would be.

Again, all of this becomes a HUGE headache for a relocating loader. At
the very least having complex expression in library code is very
undesirable.

I did just double-check, and the binutils support includes gdb. This is
a pretty major win.

hpax, 2020-09-28:
On 2020-09-28 14:58, Paulo Custodio wrote:
> I am drawn back by the heavy weight of the GNU binutils for our uses. I
> have looked into ELF as an object module format in the past, and it just
> seems an overkill. Also, as mentioned by @suborb [...]

The Z80 simulator would need to support the gdb RPC protocol, then gdb
can run as a standalone process. No need to burden the emulator with it.
This is how just about every higher-level CPU does it.

You could argue that ELF is "overkill", but in so many other ways it is
an extremely simple format, and there are a lot of very excellent tools
for manipulating it. From my perspective, it would be way simpler to use
it than the existing object format for exactly that reason: I can know
exactly what to expect.

	-hpa

hpax, 2020-09-29:
I think I came up with an idea for how to deal with that, with the same precaution that it is likely to fall pancake if the user wants to relocate their binary, unless the loader is so complex it allows arbitrary expressions.

The idea is to have a separate (non-manifest) section (called, for example, .expr), which is a table of expressions encoded in some format that would let the expression itself contain relocations. Once the final link is done and addresses are assigned everywhere, but without stripping the relocations yet, compute the values of these expressions and fix up any relocations pointing into the .expr section. This could eventually be integrated in the linker, but can also be done entirely in a wrapper... and yes, In know how to write such a wrapper and it isn't even particularly hard; I have written enough ELF manipulation code that I'm pretty sure I could do it in a couple of hours at most.

suborb, 2020-09-29:
For reference, I've just had a look through one of the libraries, and here are a couple of the expressions used:

```
    E Cw $0036 $0037: ((CONSOLE_COLUMNS)*(CONSOLE_ROWS-1)) (section code_clib) (file target/ace/stdio/generic_console.asm:79)
    E Cw $0081 $0082: rowtbl-224 (section code_clib) (file target/ace/input/in_Inkey.asm:103)
```
The second one I hope shouldn't cause any issues - taking an offset from a label is a fairly standard thing to do with hand written assembler.

The first one refers to constants defined in other modules (actually in the crt), I'm sure it could be rewritten so that calculation takes place in the crt/other module, it's just knowing where and not having the assembler resolve it down to 0 (which I saw with vasm!)

To combine the two together, I suspect there's cases where we take `(label + calc) % 256` but I can't remember where at the moment. 

Personally I'm not adverse to leveraging binutils, I know sdcc is heading that way, and source level debugging has been a big missing feature for many years (I'm really going to enjoy adding dwarf support to sccz80). My concern is that we've written code and behaviour to exploit the features within z80asm and the way it does certain things is very different to some other assembler/linkers.

Being realistic, I barely have enough time to do what I currently do (hence the +ondra work has sat for a couple of weeks untouched) so I'm not going to pick up a binutils spike anytime soon but I am interested in whether it could be made to work and cover the current linker behaviour and satisfy the requirements of the issues listed above.

pauloscustodio, 2020-09-29:
> Being realistic, I barely have enough time to do what I currently do

Same here. I've started a rewrite of z80asm a zillion times (exactly to be able to rip the assembler and linker apart), but never had time to complete it. Therefore I prefer baby steps.

spth, 2020-10-23:
If there is demand for it, SDCC could add support for DWARF for z80 and related backends.
SDCC already supports DWARF for hc08, s08 and stm8, it shouldn't be hard to port that to z80 and related.

zx70, 2020-12-31:
..I'm adding something a bit off-topic, but possibly related to a linker enhancement.
https://z88dk.org/forum/viewtopic.php?f=2&t=11435

hpax, 2021-09-06:
FYI: gdb for z80 is now in the development branch of the binutils repository.

spth, 2021-09-06:
Here is a patch that enables ELF output for z80 and related in SDCC:
[z80-elf.patch.gz](https://github.com/z88dk/z88dk/files/7113737/z80-elf.patch.gz)

I hasn't been tested much: the sdldz80 linker doesn't support the -f option, so linking fails for me with the patch applied.

hpax, 2022-10-05:
Well, well... I just found something rather astounding.

It turns out that GNU binutils already has support for arbitrary expressions in ELF! They are encoded as special relocations; It is enabled in all the tools, except that the assembler (gas) needs to be configured to emit them. This is literally a matter of turning on an option:

[diff.txt](https://github.com/z88dk/z88dk/files/9717735/diff.txt)

With that enabled in the assembler, the rest of binutils doesn't even have to be rebuilt; it already works. The following source code (sorry for the .txt extension, github is being a pain):

... produces the following output ...

[expreloc.readelf.txt](https://github.com/z88dk/z88dk/files/9717773/expreloc.readelf.txt)

... which then links to:
00000100 <test>:
 100:   21 30 07        ld hl,0x0730
 103:   fd 7e 84        ld a,(iy-124)

... for the following symbol values:

hpax, 2022-10-05:
There is an obvious downside to using expression relocations of this sort, which is that any segment base is (rather inherently) lost. This becomes a problem if one wants to postprocess the ELF output to generate a target-specific relocatable output, as all those symbols will be considered absolute by the linker; the postprocessor would have to be able to process the relocation expression symbol and determine if it is something that it can validly handle. I am guessing that the same is true for other linkers, however.

ld *could* add the capability to determine if the expression is of the form <base symbol> + <absolute expression> after resolving all the expression symbols and mark it relative to the appropriate section, but at this point it simply doesn't.

hpax, 2026-08-16:
The main reason to adopt ELF is that it would provide compatibility with GNU binutils and gdb, which have had support for z80-ELF for quite a while now.

hpax, 2026-08-16:
Personally, I have migrated most of my personal projects to binutils already, so it is not a huge deal for me, either way.

pauloscustodio, 2026-08-27:
I think the question you are asking is not if z80asm outputs ELF, but it is if z80asm outputs a binary image plus a set of relocations, including R_Z80_16, R_Z80_8, R_Z80_LO8, R_Z80_HI8, ...

Yes, the current assembler does output the list of relocations with the -reloc-info option, but they only cover 16-bit relocations and only one loadable address change possibility, so it does not cover the ABC800 needs as you describe.
 
For example the code:
```
section code
	org 0x100
	
	ld hl, string
	call print_string
	ret
	
section data
string:	defm "hello", 0

section code
print_string:
	ld a, (hl)
	and a
	ret z
	rst $10
	inc hl
	jp print_string
```

assembled with ```z88dk-z80asm -v -b -m -reloc-info test``` generates test_code.bin:
```
00000000  21 0f 01 cd 07 01 c9 7e  a7 c8 d7 23 c3 07 01 68  |!......~...#...h|
00000010  65 6c 6c 6f 00                                    |ello.|
```
and test_code.reloc:
```
00000000  01 00 04 00 0d 00                                 |......|
```

The ABC800 example suggests a need for richer relocation metadata rather than necessarily ELF output. The current -reloc-info output only reports locations of 16-bit relocations assuming a single relocation base. A possible future evolution would be to extend the relocation format to record relocation type (R_Z80_16, R_Z80_LO8, R_Z80_HI8, etc.) and relocation domain/section, allowing startup code or loaders to apply multiple independent relocations. This would address the use case you describe while remaining independent of any particular object-file format such as ELF.

hpax, 2026-08-29:
As I explicitly said, ELF is *one* way to solve these problems. The main advantage is that it already exists.

hpax, 2026-08-30:
The other nice thing with ELF, as said, is that it brings compatibility with the entire GNU ecosystem without the need to implement BFD for a new object format (and BFD is not particularly nice to muck with.) The biggest deal with that is gdb.

spth, 2026-09-01:
SDCC already supports ELF/DWARF output for the hc08, s08 and stm8 ports. So it shouldn't be too much effort to also enable ELF/DWARF output for z80, z80n and z180 ports.

zx70, 2026-11-07:
Paulo, I'm working on a nice but probably too pretentious toolset to manage the rel files.
The idea is to expand the original bit-compacted structure to a special intermediate file describing the rel content in simple byte-based structure.
The structure is (often) reversible to bits, which makes the rel files editable.

I've put the three tools in z88dk/support/rel.

I tried to add a feature to create z80asm object files out from a rel one I think I got close but it doesn't still work.  Now I'm beginning to understand that using the common code in z80asm etc.  Could be a better approach.

Will z80asm be ready to support the relocs in the object files structure soon?
Would you like to take the three tools over and evolve then or integrate them the z80asm realm?

pauloscustodio, 2026-11-08:
Stefano, I appreciate your contributions and I think we can integrate the REL-file reading in z80asm so that it processes them and present to the linker as a standard object file. I've been writing the evolution of z80asm, but the new code did not yet reach the linker.

Let's continue the discussion in #243