z88dk/z88dk #1580: newlib - differentiate between r2k and r2ka

feilipu, 2020-10-02:
Noting that SDCC is now differentiating between rabbit 2k and the bugfix revisions of the 2k hardware (r2ka).

```
2020-09-27 Philipp Klaus Krause <pkk AT spth.de>
    * r2ka port to better support Rabbit 2000A, 2000B, 2000C, 3000. RFE #709.
```

Is there iterest to continue that differentiation here?

suborb, 2020-10-03:
For reference: ftp://ftp1.digi.com/support/documentation/html/manuals/Rabbit2000/UsersManual/brevs.htm

spth, 2020-11-02:
The original Rabbit 2000, i.e. the most buggy variant hasn't been available since early 2002, when it was replaced by the Rabbit 2000A.
But for about 2 years and 3 months, it was the only Rabbit made, so I guess it can still be found in some older devices.

feilipu, 2021-08-23:
closed by [this commit](https://github.com/z88dk/z88dk/commit/602ba59116bb3e93816e3f2105ae5914715ebdd0).

closed 2021-08-23 by feilipu