z88dk/z88dk #1581: [mc1000] MC1000 8k binary freezes on real machine without memory expansion but runs with expansion

Fabrizio-Caruso, 2020-10-03:
Some people are testing my game(s) on real MC-1000 with and without memory expansion.
Maybe this is for @zx70.

An 8k binary (see attachments) freezes after loading on the unexpanded MC1000 (which should have 14k of available, probably contiguous memory for code and data).
The very same binary works fine on Mame without expansion. So we should not rely on Mame/Mess for tests.
On the other hand it DOES work on a real MC1000 with a 64k memory expansion:
- unexpanded Mame: OK
- unexpanded real hardware: KO
- expanded real hardware: OK

Tests are performed by Claudio Henrique Picolo and reported in:
https://www.facebook.com/groups/263599397024920/user/1494520522/

I suspect that the problem may come for UDG graphics. Are UDG data stored too high in memory? Can I configure UDG graphics so that it uses memory that is available on unexpanded models?
This is how I compile the game:
```
mc1000: $(ASSETS_PATH)/z88dk_xchase.asm 
	$(Z88DK_PATH)$(MYZ88DK) +mc1000 $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-subtype=gaming -DSLOW_DOWN=700 \
	-DZ88DK_JOYSTICK -DFULL_GAME -DANIMATE_PLAYER -DUDG_GRAPHICS \
	-D__MC1000__ -DSOUNDS -DEND_SCREEN -DBETWEEN_LEVEL -DCONIO \
	-vn  -lndos -Cz--audio \
	-create-app -o mc1000.bin \
	-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(FULL_FILES)
```

[mc1000.zip](https://github.com/z88dk/z88dk/files/5321310/mc1000.zip)
[FULL_Xbomber_mc1000_joystick.zip](https://github.com/z88dk/z88dk/files/5321311/FULL_Xbomber_mc1000_joystick.zip)

suborb, 2020-10-03:
Looking at the crt and the report I have a hunch that the problem is the stack. 

The default crt tries to detect how much memory the machine has - it could fail quite easily depending on the behaviour of the real machine when no memory is present. 

Try compiling with -pragma-define:REGISTER_SP=0x3fff to bypass the memory detection.

Fabrizio-Caruso, 2020-10-03:
@suborb, with `-pragma-define:REGISTER_SP=0x3ff` the game loads but it is broken (broken graphics) under Mame/Mess

suborb, 2020-10-03:
Argh, my typo! It should be 0x3fff (16383)

Fabrizio-Caruso, 2020-10-03:
Thanks! It seems OK under Mess. I will ask Claudio to run some tests on the real hardware.

Fabrizio-Caruso, 2020-10-04:
@suborb indeed the stack at 0x3fff fixes the issue.

closed 2020-10-04 by Fabrizio-Caruso