z88dk/z88dk #1582: [sccz80][bug][code generator][all targets] Corrupted binaries with SCCZ80 but OK with ZSDCC

Fabrizio-Caruso, 2020-10-04:
I am getting a corrupted generated binary if I compile with SCCZ80 but I do get the right result if I use ZSDCC.
This was working fine some time ago with both SCCZ80 and ZSDCC.

This seems to concern all Z88DK targets.

The following code is supposed to display a spiral but if compiled with SCCZ80 in a recent Z88DK version, then it keeps on displaying the very same tile only horizontally and loops forever:
```
void spiral(register Character *characterPtr)
{
    uint8_t i;
    uint8_t j;
    characterPtr->_x = XSize/2;
    characterPtr->_y = YSize/2;
    for(i=0;i<2*MIN_SIZE-18;++i)
    {
        for(j=0;j<i/2;++j)
            {
                displayCharacter(characterPtr);        
                if(i&2)
                {
                    ++(*((uint8_t *) characterPtr + (i&1)));
                }
                else
                {
                    --(*((uint8_t *) characterPtr + (i&1)));                
                }
                #if SLOW_DOWN>0
                    DO_SLOW_DOWN(55);
                #endif
            }
    }
}
```

Moreover a warning when I compile with SCCZ80 produces a probably wrong warning:
```
games/chase/end_screen.c:92: warning 158: overflow in implicit constant conversion
```
which is about 
```
				for(ind=0;ind<254;++ind){};
```
where ind is defined as `uint8_t`.
This should be OK as ind can have a 255 value before overflowing.

suborb, 2020-10-04:
The problem was caused by an optimisation rule that wasn't sufficiently qualified - I've remove it for now.

The warning looks like it's come from sdcc from than sccz80 - you raised it previously here: #1129

Fabrizio-Caruso, 2020-10-04:
@suborb By the way, nightly builds are stuck at the 28th of September. Something is broken in the tool-chain.

Fabrizio-Caruso, 2020-10-04:
@suborb you are right about the warning. Sorry for confusing SCCZ80 and ZSDCC on the warning issue.

suborb, 2020-10-04:
> @suborb By the way, nightly builds are stuck at the 28th of September. Something is broken in the tool-chain.

Hopefully will be fixed tonight <fingers crossed>

suborb, 2020-10-05:
@Fabrizio-Caruso  all good?

Fabrizio-Caruso, 2020-10-05:
I confirm that the problem is solved. Thanks a lot for fixing it.

closed 2020-10-05 by Fabrizio-Caruso