z88dk/z88dk #1583: [gb][udg/sprite/conio] Support for standard Z88DK interfaces? If not, how to set different initial tiles?

Fabrizio-Caruso, 2020-10-04:
I see that the gameboy target is up and running and has stdio support and gbdk support.
I guess that there is no support for VT52 control codes and CONIO.
What is meant by lowres graphics support in Z88DK in https://github.com/z88dk/z88dk/wiki/Platform---Gameboy ?

It would be nice to code it as the rest of the Z88DK targets. I suppose more support may come at some point but for the moment I may try to use it with some minimal gbdk. 
I see that the Z88DK comes with some default fonts loaded in order to have stdio working.
Is there a way to select a different set of tiles?
I could do this in gbdk with `set_bkg_data` but I suppose Z88DK would generate some useless binary to define default tiles that I may not want to use.

suborb, 2020-10-04:
I've just added the feature list to the gb page since it was missing.

You should be able to broadly code it the same as the other platforms: VT52 is there, it supports custom fonts, UDGs. Font is selected via the usual `-pragma-redirect:CRT_FONT=....`

In terms of input there's `joystick()` which will be preferable to the virtual keyboard that pops up when you call fgetc().

Fabrizio-Caruso, 2020-10-04:
Thanks a lot!

closed 2020-10-04 by suborb

Fabrizio-Caruso, 2020-10-04:
@suborb Indeed I can get gb to work with VT52. But I am not able to get UDG to work. I only get a white screen.

For targets with UDG I just link a .asm file. This seems not to work. I have to double-check my Makefile.
In the .asm file I am assuming that the tiles are 8x8 and monochromatic. Do I have to do different assumptions for the GB?
If I have to assume 4 color (or more) tiles. What is the format I should use?

```
gb_gfx: $(ASSETS_PATH)/z88dk_xchase.asm
	$(Z88DK_PATH)$(MYZ88DK) +gb $(SCCZ80_ROM_OPTS) $(INCLUDE_OPTS) -pragma-redirect:fputc_cons=fputc_cons_generic \
...
	-lndos -DSLOW_DOWN=900 \
	-DUDG_GRAPHICS $(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(FULL_FILES) \
	-create-app -o$(BUILD_PATH)/FULL_gb.bin
```

suborb, 2020-10-04:
It looks like you may have more success in mode 1:

[image]

I'll look at mode 0 and see why the ioctls don't seem to work.

Fabrizio-Caruso, 2020-10-05:
@suborb  Where can I find the doc on how to change screen mode?
I am trying `set_mode`.

suborb, 2020-10-05:
It's the usual:

    int mode = 1;
    console_ioctl(IOCTL_GENCON_SET_MODE,&mode)

Fabrizio-Caruso, 2020-10-05:
Thanks! Is there a page with examples and some documentation?
What do I need to include to get it to work?
I get:
symbol '_IOCTL_GENCON_SET_MODE' not defined

Fabrizio-Caruso, 2020-10-05:
```
#include <sys/ioctl.h>

void INIT_GRAPHICS(void)
{
    int mode = 1;
    console_ioctl(IOCTL_GENCON_SET_MODE,&mode);
}
```
fails with:
c:/z88dk//include/sys/ioctl.h:7:43: fatal error: Cannot parse function argument
'uint16_t' - unknown type, but not K&R syntax
Compilation aborted
make: *** [makefiles.chase/targets/makefiles_z88dk/Makefile_z88dk_a-k:618: gb] Error 1

Fabrizio-Caruso, 2020-10-05:
I got it to compile with 
#include <stdint.h>

Probably sys/ioctl.h should include it.

Fabrizio-Caruso, 2020-10-05:
I am getting wrong tile shapes. Is mode_1 interpreting tile data differently? Is it still 8x8 tiles?

[image]