z88dk/z88dk #1584: [question/bug?][gameboy] Setting a font with console_ioctl

Fabrizio-Caruso, 2020-10-05:
I am trying to use UDGs on the gameboy target as I do on all other udg-enabled targets, i.e., I use a .asm file with the characters/tiles definition and I export a symbol to have an array in C that points to the tile data. 

I use the same code I already use (shown below)

```
#include <graphics.h>
#include "display_macros.h"
#include <sys/ioctl.h>
extern uint8_t udgs[];
void INIT_GRAPHICS(void)
{
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
	#if defined(__MC1000__) || ... || defined(__GB__)
	{
		#if defined(FORCE_SCREEN_MODE)
			#define SCREEN_MODE FORCE_SCREEN_MODE
		#elif defined(__MC1000__) || defined(__GB__)
			#define SCREEN_MODE 1
		#elif defined(__SPC1000__)
			#define SCREEN_MODE 2
		#else
			#define SCREEN_MODE 2
		#endif
		int mode = SCREEN_MODE;
		console_ioctl(IOCTL_GENCON_SET_MODE, &mode);
	}
...
        #endif
}
```

I get wrong tiles/characters (see picture below).

The .asm file with the tiles definitions that I am passing to the linker is the very same I use on all other targets where I assume 8x8 monochromatic tiles/characters.

I am compiling with:
```
gb_gfx: $(ASSETS_PATH)/z88dk_xchase.asm
	$(Z88DK_PATH)$(MYZ88DK) +gb $(SCCZ80_ROM_OPTS) $(INCLUDE_OPTS) -pragma-redirect:fputc_cons=fputc_cons_generic \
...
	-lndos \
	-DUDG_GRAPHICS $(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(FULL_FILES) \
	-create-app -o$(BUILD_PATH)/FULL_gb.bin
```

What am I doing wrong?

[image]

Fabrizio-Caruso, 2020-10-05:
@suborb I can see that 	`console_ioctl(IOCTL_GENCON_SET_FONT32, &param);` has some effect because I see different characters than the default ones but they are both wrong and are displayed wrongly by one or two bytes (wrong offset).

suborb, 2020-10-05:
I've got lost in your repo! But I think you are using IOCTL_GENCON_SET_FONT32?

If so, then I added some docs for you here: https://github.com/z88dk/z88dk/wiki/Platform---Gameboy#custom-fonts

Fabrizio-Caruso, 2020-10-05:
Thanks a lot.
Indeed I am using in `IOCTL_GENCON_SET_FONT32`
https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/cross_lib/display/init_graphics/z88dk/udg/udg_init_graphics.c

But from the doc for the gameboy case only I should use `console_ioctl(IOCTL_GENCON_SET_UDGS,&param)` else and link a different file (no header).

For all other udf-enabled targets I use a file generated (by my scripts) from this template:
https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/assets/templates/8x8/z88dk_xchase.asm.template

Fabrizio-Caruso, 2020-10-05:
@suborb 
What is the header mentioned in the doc?
Should I remove the SECTION lines?
```
SECTION rodata_font
SECTION rodata_font_8x8
```
and leave
```
PUBLIC _udgs
PUBLIC _udgs_end
```

suborb, 2020-10-05:
Sorry about the difference here - trying to maintain compatibility with gbdk, its virtual keyboard and z88dk proved a bit too difficult.

What you should do is:

```
SECTION rodata_user
PUBLIC _udgs

_udgs:

; Paste in the contents of lower.asm here

; Include your usual font file here

```

Fabrizio-Caruso, 2020-10-05:
@suborb I think you have done a great job. I understand this is a compromise with GBDK. It is a very good one as I can use the same code (or nearly the same) and reuse the asset data.

off-topic: is this kind of compromise now also available for other oddball targets like SMS?

suborb, 2020-10-05:
I've not touched the SMS port much so the library is a little bit underfeatured.

 I could probably look to see how much of the TMS99x8 code can be reused, if it can be we'll get a load of features for free.

Fabrizio-Caruso, 2020-10-05:
sms did not have an generic console when I used it last time. I need to give it a try again.

suborb, 2020-10-05:
It's missing some things and when they're added it doesn't work. I'll have a play around!

Fabrizio-Caruso, 2020-10-05:
@suborb I got GB graphics working!

Fabrizio-Caruso, 2020-10-05:
@suborb, I followed your instructions and it works now.
You just need to pull/clone the latest version of my repo and do:
`make clean && make gb`

Fabrizio-Caruso, 2020-10-05:
@suborb I guess you are talking about sms...

suborb, 2020-10-05:
> @suborb I guess you are talking about sms...

Yes I was. I've just committed some changes that will get mode 2 (https://github.com/z88dk/z88dk/wiki/Classic-TMS9918#mode-2-text-32x24-graphics-256x192) working in gencon. 

Mode 0/1 display, but have a broken scroll for some reason.

suborb, 2020-10-05:
> @suborb, I followed your instructions and it works now.

Great stuff, I'll close this issue then.

> @suborb I guess you are talking about sms...

I've spent a little time and now SMS has both gencon and a `joystick()` interface, so it should have enough to support cross-lib without using custom code.

closed 2020-10-05 by suborb

Fabrizio-Caruso, 2020-10-06:
Does the sub-target GameGear also get support with standard Z88DK APIs?

suborb, 2020-10-06:
It shares the same library code so should work - you'll need to offset obviously so that things end up being displayed.

Fabrizio-Caruso, 2020-10-06:
@suborb it may not support the same lib
```
/cygdrive/c/z88dk/bin/zcc.exe +sms  -lgamegear -clib=sdcc_iy -startup=17 \
...
-Igames/chase/generated_assets -Igames/chase/ \
-pragma-include:cross_lib/cfg//z88dk/zpragma_sms.inc \
...
-o ../build/WIP_TINY_gamegear_buffered.bin \
...
games/chase/main.c:743: warning 126: unreachable code
Error: cannot read file 'gamegear.lib'
```

suborb, 2020-10-06:
Oh. `-clib=sdcc_iy -startup=17` means that you're using newlib rather than classic.