z88dk/z88dk #1585: [mz2500] Missing kbhit()?

Fabrizio-Caruso, 2020-10-05:
I am compiling my game for the mz2500 with
```
mz2500:
	$(Z88DK_PATH)$(MYZ88DK) +mz2500 $(SCCZ80_OPTS) $(INCLUDE_OPTS) -vn -lndos \
	-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
	-pragma-redirect:getk=getk_inkey  \
	-D__MZ2500__ -DFORCE_XSIZE=40 -DFORXE_YSIZE=24  \
	-DCONIO -DNO_COLOR -DNO_INIT_GRAPHICS \
	-create-app -o $(BUILD_PATH)/FULL_mz2500.prg \
	$(FULL_FILES)
	rm $(BUILD_PATH)/FULL_mz2500.prg
```

but I get:
```
cross_lib/input/input_macros.c:293:24: warning: Implicit definition of function
'kbhit' it will return an int. Prototype it explicitly if this is not what you w
ant. [-Wimplicit-function-definition]
cross_lib/input/input_macros.c:294:30: warning: Implicit definition of function
'getch' it will return an int. Prototype it explicitly if this is not what you w
ant. [-Wimplicit-function-definition]
Error at file 'C:/cygwin/tmp/zcc000039E844D618/zcc_opt.def' line 14: symbol 'get
k_inkey' not defined
Errors in source file c:/z88dk/\lib\target\mz2500\classic\mz2500_crt0.asm:
Error at file 'C:/cygwin/tmp/zcc000039E844D618/zcc_opt.def' line 14: symbol 'get
k_inkey' not defined
                   ^ ----       defc getk = getk_inkey
make: *** [makefiles.chase/targets/makefiles_z88dk/Makefile_z88dk_l-p:441: mz2500] Error 1
```

suborb, 2020-10-05:
I think you need to include `<conio.h>`to define it.

Fabrizio-Caruso, 2020-10-05:
@suborb Sorry for my stupid mistake.

closed 2020-10-05 by Fabrizio-Caruso