z88dk/z88dk #1586: [altair8800][bug?] Error: cannot read file 'altair8800_clib.lib'

Fabrizio-Caruso, 2020-10-05:
I am using the very last snapshot (Oct 4th, 2020).
I am trying to compile my game for the Altair8800 in STDIO mode (buffered turn-based mode that only uses stdio.h) with

```
altair8800_tiny_buffered:
	$(Z88DK_PATH)$(MYZ88DK) +altair8800 $(SCCZ80_OPTS) \
	$(INCLUDE_OPTS) \
	-D__ALTAIR8800__ \
	-vn \
...
	-lndos \
	$(TINY_FILES) \
	$(CROSS_LIB_PATH)/display/alt_print/memory_mapped_alt_print.c \
	-create-app -o altair8800
```

but I get:
```
Error: cannot read file 'altair8800_clib.lib'
make: *** [makefiles.chase/targets/makefiles_z88dk/Makefile_z88dk_a-k:271: altair8800_tiny_buffered] Error 1
```

suborb, 2020-10-05:
This one should be sorted in the next build.

closed 2020-10-05 by suborb

Fabrizio-Caruso, 2020-10-06:
It does work now. It produces a .rom file. Some emulators such as Altair32 (http://www.classiccmp.org/altair32/) expect .hex or .bin files.
What is the start address I should give to the emulator. The Z88DK doc only mentions variables and stack address.
Which emulator have you used to test the binaries?

suborb, 2020-10-06:
I just used this one: https://github.com/gorskima/altair8800 and overwrote the BASIC ROM with a bit of a hack.

The start address is 0x0000, the .ROM file can be just loaded as a .BIN file. Adding an ihx output will be easy if it's needed.

Fabrizio-Caruso, 2020-10-06:
Altair32 seems to be a recent emulator easy-to-use emulator.
It can load raw code in .bin and .hex (I guess the same as .ihx) formats.
I tried to pass the .rom as .bin but nothing happened. Moreover I am new to Altair8800. I guess I need to connect to some virtual teletype terminal to see output and insert some input.

suborb, 2020-10-06:
The altair port is pretty raw - all it's got is a raw serial port so you'd have to output raw VT100 codes to drive the terminal that you're using.

I'll look at altair32 and see what it can load.

Fabrizio-Caruso, 2020-10-06:
I guess you just output something with an encoding X to a terminal that expects and understand the control codes X. So you can use ADM3/A, VT52, VT100, etc.. depending on the terminal you use. I guess ADM3/A was used at that time but I may be wrong.

suborb, 2020-10-06:
I've got something working on altair32 but it's a bit weird.

Copy the binary into the files directory of the emulator:

[image]

And load it address 0:

[image]

You then need to turn on the altair using the toggle. And then try to click the run/stop button once.

Regarding the terminal, it looks like on windows you can open a console - that'll support VT100 so that's the set of codes you should pick.

Fabrizio-Caruso, 2020-10-06:
Thanks a lot! I got it to work thanks to your instructions!