z88dk/z88dk #1587: Issue using latest snap (06Oct20) and alias

shifters67, 2020-10-06:
When following the instructions on the snap page, trying to create

sudo snap alias z88dk.z88dk-appmake z88d-appmake

error: cannot perform the following tasks:
- Setup manual alias "z88d-appmake" => "z88dk-appmake" for snap "z88dk" (cannot enable alias "z88d-appmake" for "z88dk", target application "z88dk-appmake" does not exist)

Changing the obvious mistake to z88dk-appmake from z88d-appmake (on the snap page) does not fix the issue

Ubuntu 20.04 latest

suborb, 2020-10-06:
Oh, that's clever of me. I didn't expose z88dk-appmake as a command for some reason - I'll add that into the yaml file.

Thanks for the report.

suborb, 2020-10-06:
Fixed in snap 17009-4f87322df-20201006 onwards.

closed 2020-10-06 by suborb