z88dk/z88dk #1590: [rc2014][doc?] I cannot find explicit instructions on this target

Fabrizio-Caruso, 2020-10-07:
I am building my games also for the rc2014 target.
If I only use the standard library I get Z88DK to compile it.
Is it possible to test this on an emulator? Which one?

Is this the same target ar Rabbit Control Module? described in https://github.com/z88dk/z88dk/wiki/Platform---Rabbit ?
The doc has broken links (e.g., Makefile instructions are missing: https://raw.githubusercontent.com/z88dk/z88dk/master/support/rcmx000/)

suborb, 2020-10-07:
Docs are linked from the main readme: https://github.com/RC2014Z80/RC2014/wiki/Using-Z88DK

Fabrizio-Caruso, 2020-10-07:
The target RC2014 is not listed explicitly among the platforms (https://github.com/z88dk/z88dk/wiki/Platform)? Is it?
I only find Rabbit Control Module whose linked page has broken links.
Is RC2014 the same at Rabbit Control Module?

For the (probably similar) yaz80 target there is no reference and link.

feilipu, 2020-10-07:
Probably the main reason is that they are not classic devices, and are only represented in the new library.

This affects, [HBIOS](https://github.com/wwarthen/RomWBW) (like a CP/M BIOS), [RC2014](https://rc2014.co.uk/), SC180 (SC126, [SC130](https://smallcomputercentral.wordpress.com/sc130-z180-motherboard/), and SC131), and [YAZ180](https://github.com/feilipu/yaz180).

I'll add them at some time shortly. Sorry.

Fabrizio-Caruso, 2020-10-07:
Thanks @felipu. Could you please, when you have some time, provide some instructions on which emulators (if any) to use and how to run the binaries? I am more than willing to test these "new-retro" sort of targets with my programs.
As I wrote above, I can already compile my universal game with just stdio.h. I would like to test it on an emulator if there is one.

I am currently compiling without any sub-target option:
```
rc2014:
	$(Z88DK_PATH)$(MYZ88DK) +rc2014 $(SCCZ80_OPTS)  $(INCLUDE_OPTS) \
...
...
	-create-app -o$(BUILD_PATH)/FULL_rc2014.bin
```

feilipu, 2020-10-07:
If you're compiling without a subtype then you'll be collecting the BASIC subtype by default. This is with RAM origin at 0x8000, and the serial terminal on the RST interfaces.

I don't have an emulator, as such, but I think there are a few in use ([example](https://groups.google.com/g/rc2014-z80/c/fOW44dH2uS8/m/SncHv6YHDwAJ)). The RC2014 Group had some discussion on that, and IIRC Alan Cox also built an emulator so he could test FUZIX on it.

Fabrizio-Caruso, 2020-10-07:
@suborb, @feilipu I still don't understand it Rabbit Control Module is the same as RC2014.
Is it the same target? If not, where can I find instructions on how to build for the rabbit control module?

suborb, 2020-10-07:
The Rabbit board is a completely different board to the RC2014.

If you dig around on the forum there's some threads from someone who had a Rabbit board and how they got it working.

Fabrizio-Caruso, 2020-10-07:
@suborb Thanks for clarifying this point. I thought RC might stand for Rabbit Control...
The significant links in the Rabbit board page (https://github.com/z88dk/z88dk/wiki/Platform---Rabbit) are broken.
Which forum has the most up to date information?

suborb, 2020-10-07:
Maybe I'm confused, I thought it was mimarob on the forum who contributed the Rabbit support, but I can't find them (they might have got lost in the migration?)

The support directory is here: https://github.com/z88dk/z88dk/tree/master/support/rcmx000 which has some information.

Realistically, I don't think there's many Rabbit boards out there with enthusiasts - you might be able to pick one up cheap from ebay (I think I paid £15 for one that I've still not unpacked!).

It would be similar to the altair port - just a serial port. The +test target has `-clib=rabbit` so the code generation can be tested.

feilipu, 2020-10-07:
I looked for @mirarob too, but the account looks dead. #1580 questions which Rabbit should be supported? I guess emulation doesn't care though.

closed 2020-10-13 by feilipu