z88dk/z88dk #1591: [sms] Classic lib/ Conio Support

Fabrizio-Caruso, 2020-10-07:
I get a black screen if I compile my game for the SMS with no option (I guess classic mode):
Tested with Fusion emulator.
@suborb This may be for you.

```
sms_no_gfx:
	$(Z88DK_PATH)$(MYZ88DK) +sms $(SCCZ80_ROM_OPTS) $(INCLUDE_OPTS) -pragma-redirect:fputc_cons=fputc_cons_generic \
...
	-lndos \
	$(FULL_FILES) \
	-create-app -o$(BUILD_PATH)/FULL_sms_no_gfx.bin
...
```

I can get it to work if I use newlib only (`-clib=sdcc_iy -startup=17`).

Do I need to use a special clib option to get classic mode?

I have attached the broken built. 

[FULL_sms.zip](https://github.com/z88dk/z88dk/files/5342073/FULL_sms.zip)

suborb, 2020-10-07:
Obvious question, are you switching mode first of all?

https://github.com/z88dk/z88dk/wiki/Platform---SMS#generic-console

The SMS starts up with screen mode 4 which isn't supported by gencon so you need to switch to 0,1,2 as appropriate.

Fabrizio-Caruso, 2020-10-07:
Sorry, I am not. I will try the supported modes.
Can I change the font definitions as I do with all other targets? or do I need to provide a differently structured font file as for the gb target?

suborb, 2020-10-07:
The font is the same as  other targets - gencon treats the SMS as a bog standard TMS99x8 machine

Fabrizio-Caruso, 2020-10-07:
Thanks a lot @suborb !

closed 2020-10-07 by Fabrizio-Caruso