z88dk/z88dk #1592: [game_gear] Black screen on gamegear despite working sms version with the very same code

Fabrizio-Caruso, 2020-10-07:
I am compiling my game for both sms and gamegear with the very same code and settings except:
1. different number of columns/rows in my gamecode (20/18 for gamefear and 32/24 for sms)
2. -lgamegear in the gamegear

Both the sms and gamegear versions use screen 2.

The sms version works fine. The gamegear version produces a black screen.

I am compiling for the gamegear with:
```
gamegear_gfx: $(ASSETS_PATH)/z88dk_xchase.asm 
	$(Z88DK_PATH)$(MYZ88DK) +sms -lgamegear $(SCCZ80_ROM_OPTS) $(INCLUDE_OPTS) -pragma-redirect:fputc_cons=fputc_cons_generic \
	-DFORCE_XSIZE=20 \
	-DFORCE_YSIZE=16 \
...
	-lndos \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(FULL_FILES) \
	-create-app -o$(BUILD_PATH)/FULL_gamegear.bin
```

suborb, 2020-10-07:
I think on the Gamegear the palette isn't set so it defaults to black, I'll try to figure out a good way to do this automatically, but in the mean time just add this code:

```
static unsigned char pal[] = { 0x00, 0x00, 0x08, 0x0c, 0x10, 0x30, 0x01, 0x3c, 0x02, 0x03, 0x05, 0x0f, 0x04, 0x33, 0x15, 0x3f };

load_palette(&pal,0,16);
```

Fabrizio-Caruso, 2020-10-07:
If I use your code I get some warning:
```
cross_lib/display/init_graphics/z88dk/udg/udg_init_graphics.c:33:21: warning: Im
plicit definition of function 'load_palette' it will return an int. Prototype it
 explicitly if this is not what you want. [-Wimplicit-function-definition]
```

and the result is a corrupted image:

[image]

suborb, 2020-10-07:
Load_palette is defined in sms.h

I only did a quick check using emulicious and what I was doing displayed ok in mode 2. 

I’ll do more of a check tomorrow.

Fabrizio-Caruso, 2020-10-08:
If I include `sms.h` I get:
```
sms.h: line 64: macro 'JOY_UP' redefined unidentically
        included from cross_lib/display/init_graphics/z88dk/udg/udg_init_graphic
s.c:31
sms.h: line 65: macro 'JOY_DOWN' redefined unidentically
        included from cross_lib/display/init_graphics/z88dk/udg/udg_init_graphic
s.c:31
sms.h: line 66: macro 'JOY_LEFT' redefined unidentically
        included from cross_lib/display/init_graphics/z88dk/udg/udg_init_graphic
s.c:31
sms.h: line 67: macro 'JOY_RIGHT' redefined unidentically
        included from cross_lib/display/init_graphics/z88dk/udg/udg_init_graphic
s.c:31
make: *** [makefiles.chase/targets/makefiles_z88dk/Makefile_z88dk_q-z:330: gamegear_gfx] Error 1
```

Fabrizio-Caruso, 2020-10-08:
@suborb sms.h may be conflicting with the joystick definitions in classic lib.

suborb, 2020-10-08:
That is precisely the problem - I've got an outstanding commit to resolve that problem.

I've also (finally) got a solution for the palette issues, it's late so that'll be pushed tomorrow - your code won't have to do anything (apart from change the x,y offsets), it should "just work^{TM}" when using gencon.

Fabrizio-Caruso, 2020-10-08:
@suborb Thanks a lot! 
Indeed all your amazing work is making my work on CrossLib much easier on the Z80 and I8080 targets than on the other architectures where in some cases I have to build the lib myself. 
You and Stefano are the two people who have helped me the most.
Under which name should I cite you in the acknowledgements?

suborb, 2020-10-09:
The build tomorrow should resolve the Gamegear palette issues. Please read this section: https://github.com/z88dk/z88dk/wiki/Platform---SMS#gamegear-support - I've made some tweaks to the libraries and what you're currently doing won't link/work as you expect.

The key things are:

* Use -subtype=gamegear to produce a .gg file
* There's no -lgamegear
* A .sms file will load and run on a Gamegear correctly (i.e. the palette is now set)
* Colours in .gg mode look like other TMS99x8 machines, not like the SMS
* Coordinates no longer need to be offset, that's taken care of by the library (and `screensize()` reports the correct answer in .gg mode)

Fabrizio-Caruso, 2020-10-10:
@suborb I think there is a problem with gamegear because I get a non working binary:

[image]

suborb, 2020-10-10:
I'm using Emulicious, and I get this:

[image]

Admittedly there's something wrong with the input since it's controlling itself, but that's another issue.

Which emulator are you using?

Fabrizio-Caruso, 2020-10-10:
I am using Kega Fusion. The input controlling itself may be related to some emulator config.
With Kega Fusion I see a working newlib version (with just stdio).

suborb, 2020-10-10:
I'm sorry. I really don't think I can do anything with Fusion:

- There's no debugging tools in the emulator so I've no idea what's going on
- It's closed source so I can't see what's not implemented in the emulator
- It's not been updated in close to a decade

The cartridge displays correctly in Mame and Emulicious with timing constraints enabled so I think that shows that the code is correct.

Fabrizio-Caruso, 2020-10-10:
@suborb, you are perfectly right (as usual). The input problem was on some code I had not enabled. So everything looks perfect.
Thanks a lot as you!

closed 2020-10-10 by Fabrizio-Caruso