z88dk/z88dk #1593: [pencil2][bit_bang] Bit-bang sound is broken (never stops)

Fabrizio-Caruso, 2020-10-08:
Bit-bang sound is broken (never stops).

I have attached the .rom file (zipped).

Disregard display corruption (it is my code and I can fix it).

[FULL_pencil2.zip](https://github.com/z88dk/z88dk/files/5349616/FULL_pencil2.zip)

suborb, 2020-10-09:
Do you see this on any other SN76489 targets?

Fabrizio-Caruso, 2020-10-09:
Gonna test it on the SMS...

Fabrizio-Caruso, 2020-10-09:
No PSG nor bit-bang sound on the SMS (tested on Kega Fusion) unless I have done something wrong with the emulator or the code.

suborb, 2020-10-15:
I'm going to close this one based on #1612

closed 2020-10-15 by suborb

Fabrizio-Caruso, 2020-10-15:
You are right. I have made a big mess with PSG.

suborb, 2020-10-15:
No worries. Maybe hold off on any PSG work for the moment - I'm porting PSGlib (from the newlib +sms) to the SN76489 targets. This should give a framework to play music/effects on those machines.

For the AY819x based machines, there's WYZ (https://github.com/z88dk/z88dk/wiki/Classic-WYZ-Player) which can also handle effects.

The SN76489 will be well documented on the SMS development sites/forums if you want to just create adhoc effects as you're doing at the moment.