z88dk/z88dk #1594: What is the status of curses ?

jayacotton, 2020-10-08:
Not a bug, but a question.

Wondering if curses have had any additional work in the last year.

suborb, 2020-10-09:
If I remember correctly you're targeting +rc2014? If that's true then I'll hand over to @feilipu

jayacotton, 2020-10-09:
Yes rc2014

feilipu, 2020-10-10:
@jayacotton AFAIK, I don't think that newlib has curses support. Certainly, there's no `curses.h` file in the [appropriate place](https://github.com/z88dk/z88dk/tree/master/include/_DEVELOPMENT/proto). So at first pass, I'd say not in any of the newlib targets. But, if this is wrong, I'd stand corrected.

feilipu, 2020-10-10:
Though, at a pinch, you can use `conio.h` using the `+cpm` target and classic library.

jayacotton, 2020-10-10:
I'll try that.

feilipu, 2020-10-11:
Putting `curses.h` and the implementation on the list.

closed 2022-02-17 by feilipu