z88dk/z88dk #1595: [bic][cpm] What is the .dump binary produced for the bit subtype?

Fabrizio-Caruso, 2020-10-08:
I cannot find any documentation on the CP/M BIC subtype. I guess it is for the BIC/A5105 system.
I use JKCEMU. I can understand German but I don't know how to load it.
I select the BIC target and click on Laden (Load). Then I have to select the type of file and possibly the start and end address.

suborb, 2020-10-09:
It's a disc image (I really don't know why anadisk uses .dump as a suffix) - so you'll need to load CP/M and insert the .dump as the secondary drive.

Fabrizio-Caruso, 2020-10-09:
I guess I need a BIC-specific CP/M boot disk...

suborb, 2020-10-09:
They're built into the emulator!

* Click on disc icon
* Laufwerk 1
* Eject icon
* Andere Diskette einlegen
* BIC A5105 system disk

Fabrizio-Caruso, 2020-10-09:
Thanks a lot!

closed 2020-10-09 by Fabrizio-Caruso