z88dk/z88dk #1596: [kramer-mc][question] What is the binary format? How to load it?

Fabrizio-Caruso, 2020-10-09:
I cannot find any information about the produced binary and how to load it.

I have used JKCEMU. I understand German but I do not have enough information on how to load the .kcc.
Is the .kcc a raw binary? a BASIC file?

I have tried with the JKCEME Monitor (default).
I have loaded it in JKCEMU (Laden) at the default address $1000 but nothing happens with the Monitor.
If I use E1000, nothing happens.

Should I use the BASIC interpreter?

suborb, 2020-10-09:
Click "Starten" instead of "Laden"!

Fabrizio-Caruso, 2020-10-09:
Thanks a lot! And sorry for not seeing it.
I am going to close this issue but I may open one of getk which seems not work on this target.

closed 2020-10-09 by Fabrizio-Caruso

suborb, 2020-10-09:
See the thread here: https://z88dk.org/forum/viewtopic.php?f=3&t=11124&p=18132&hilit=Kramermc#p18132